gnuradio/volk #234: Can't compile on mac

alexbirkett, 2019-01-27:
Steps to reproduce:

From Mac terminal (Late 2013 running 10.14.2)

```
git clone https://github.com/gnuradio/volk.git
cd volk/
cmake ..
make -j4
```

Compiler error is attached

[error.txt](https://github.com/gnuradio/volk/files/2800550/error.txt)

alexbirkett, 2019-01-28:
Yes sorry. I missed the 
```
mkdir build
cd build
``` 
steps from the description.

The output to standard out is just 

```
[ 68%] Built target volk_obj
[ 71%] Built target volk
[ 73%] Building CXX object lib/CMakeFiles/volk_test_all.dir/testqa.cc.o
```

I'm guessing that's not what you're after. How do provide the build log?

alexbirkett, 2019-01-28:
So it turns out that I had built volk in 2017. I had old version of the headers in `/usr/local/include/volk`. This is a bug IMHO because the tests should not be built against old version of the headers.

alexbirkett, 2019-01-28:
I tried to change some system includes to user includes to fix the issue but could not make it work. Here are the old headers that were placed in `/usr/local/include/volk`
[headers.zip](https://github.com/gnuradio/volk/files/2802194/headers.zip)

michaelld, 2019-01-28:
If you comment out the 2 lines here < https://github.com/EttusResearch/uhd/blob/master/host/CMakeLists.txt#L343 >, clear out the build directory, then rebuild from scratch, everything should work. be curious if it does for you.

michaelld, 2019-01-28:
I am continually trying to keep the includes ordered so-as internal-to-build come first, then system ones (on all of the projects I work on, which includes Volk). I recently changed a patch that I use for my local building of Volk, which fixes the ordering so that the build works. I'll have to revisit that change to verify it is now correct. Volk was recently migrated to "modern CMake" target specification, which might have broken includes ordering.

michaelld, 2019-04-26:
I changed the name of this issue to reflect the actual problem. This shouldn't matter whether OSX or Linux or whatever: already-installed Volk headers should be ignored!

n-west, 2019-04-28:
This seems like a complicating factor due to the whole "volk module" capability which does need to depend on installed headers

michaelld, 2019-11-14:
Update: In a recent test, I still see this issue. I'll try to find some time to look into it.

michaelld, 2019-12-08:
A brief update: Fixing this has not been as simple as I had hoped.

The overarching goal here is that the include directories are ordered such that those internal to the build are searched for headers before those external to the build. The problem of this issue is that this goal isn't being met for critical include directories such as `$CMAKE_INSTALL_PREFIX/include`, and hence any already-installed Volk header will be found before those internal to the build -- leading to build errors when the Volk API changes because the old API doesn't provide what is required to build the new API code (as is the case for #312 ).

When building Volk we use CMake targets to define certain steps of the build, such as the primary Volk library. A target contains source files to be compiled, and generally some of those sources -- but not all -- require certain Boost headers. For Volk, none of the library source requires Boost, but some of the QA / testing source does.

Each CMake target has properties such as the `include_directories` ( https://cmake.org/cmake/help/latest/command/target_include_directories.html  
 and   https://cmake.org/cmake/help/latest/prop_tgt/INCLUDE_DIRECTORIES.html ). These include directories can also be added globally via `include_directories` ( https://cmake.org/cmake/help/latest/command/include_directories.html ), or as source properties ( https://cmake.org/cmake/help/latest/prop_sf/INCLUDE_DIRECTORIES.html ).

Clearly there are a variety of ways to set include directories, and which one is used is critical to the ordering determined by CMake. Broadly speaking, the ordering seems to be (1) source ; (2) global ; (3) target.

For many sources, Boost is the only external dependency [ORC is an optional one, but it's headers are by default found in `$PREFIX/include/orc` and hence don't provide access to other project headers.].

The issue for current Volk is that -all- of the include directories except Boost are provided by the internal targets ... so, given the ordering above, how does one supply the Boost include directory in a way that it comes last? We can always add it into one of the targets even though it's not required for building ... but that seems unnecessary and like there should be a better way to do what we need to do!

michaelld, 2019-12-08:
After some poking through the Volk source code ... I think we can remove the Boost stuff from `lib`, which would solve this issue! The usage in `app` already has the include directories correctly ordered, and the recent C++17 `std::filesystem` would entirely remove Boost usage there. It's just 1 fie, I believe, where Boost is being used now!

The only reference I find to "boost" is here: https://github.com/gnuradio/volk/blob/master/lib/qa_utils.h#L146-L152 , and this macro `VOLK_RUN_TESTS` was the -old- way of doing tests before the `boost::filesystem` way was introduced here https://github.com/gnuradio/volk/commit/9a5e290cd4b8565b24506a879b55fd25c88717ed  ... and, hence I think it can safely be removed. I disabled it and all references to Boost in a trial build and it built just fine.

I will put together a PR with these changes to review.

closed 2019-12-13 by michaelld