gnuradio/volk #236: lib/qa_utils.cc#L628 aligned vs unaligned kernel selection

hcab14, 2019-02-04:
https://github.com/gnuradio/volk/blob/dd5e293c85975d4db6718aef685b1a15de88a74f/lib/qa_utils.cc#L623
https://github.com/gnuradio/volk/blob/dd5e293c85975d4db6718aef685b1a15de88a74f/lib/qa_utils.cc#L628

Shouldn't  `&& desc.impl_alignment[i] != 0` be added to the condition in L628 in order to ensure that only aligned kernels are selected in this if branch?

With the current code it can and does happen that an unaligned kernel is selected instead of an aligned one, _i.e._, I sometimes get the line `volk_16u_byteswap u_sse2 u_sse2` in `~/.volk/volk_config` which makes my application crash in `volk_16u_byteswap`.

n-west, 2019-04-28:
That appears possibly true and maybe the logic is backwards. You can accept an aligned version in the unaligned case if it profiles fastest but you cannot accept unaligned as the fastest for aligned versions.

Can you come up with a QA test that fails on bad currently bad results and another commit that fixes it?

hcab14, 2019-05-23:
Hi @n-west,

for some reason I cannot reproduce the behavior anymore, so as far as I am concerned this issue can be closed.

closed 2019-05-23 by hcab14