gnuradio/volk #242: Volk build fails using Boost 1.70.0

michaelld, 2019-04-26:
See < https://github.com/gnuradio/gnuradio/issues/2432 > for where this started.

michaelld, 2019-04-26:
Here's the error log:
```
/usr/bin/ld: CMakeFiles/volk_profile.dir/volk_profile.cc.o: in function write_results(std::vector<volk_test_results_t, std::allocator<volk_test_results_t> > const*, bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)': volk_profile.cc:(.text+0x31a): undefined reference to boost::filesystem::path::parent_path() const'
/usr/bin/ld: volk_profile.cc:(.text+0x329): undefined reference to boost::filesystem::detail::status(boost::filesystem::path const&, boost::system::error_code*)' /usr/bin/ld: volk_profile.cc:(.text+0x36d): undefined reference to boost::filesystem::path::parent_path() const'
/usr/bin/ld: volk_profile.cc:(.text+0x4ed): undefined reference to boost::filesystem::path::parent_path() const' /usr/bin/ld: volk_profile.cc:(.text+0x4f7): undefined reference to boost::filesystem::detail::create_directories(boost::filesystem::path const&, boost::system::error_code*)'
collect2: error: ld returned 1 exit status
make[2]: *** [apps/CMakeFiles/volk_profile.dir/build.make:116: apps/volk_profile] Error 1
make[1]: *** [CMakeFiles/Makefile2:327: apps/CMakeFiles/volk_profile.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
```

michaelld, 2019-04-26:
The "right thing to do" here is to move from Boost:: to std:: , which I think is what #174 is proposing ... though requiring c++17 might be pushing it too far just yet. Hence, just fixing this issue by adding in other Boost library linkage or whatever seems the most prudent in the moment.

n-west, 2019-04-28:
Yes, #174 would switch to c++17 filesystem. That library seems to have funny support in compilers, so it seems like one of those things that requires waiting for support to percolate. Did the new version of boost remove their filesystem library?

michaelld, 2019-04-30:
So the issue here I think is that Boost was not built with the same C++ std (e.g., 11, 14, 17) as Volk. In my testing, Volk GIT master builds cleanly using Boost 1.70.0, (so long as we don't use the Boost-provided CMake scripts for `find_package(Boost ...)`. Unless others can show that this is not the case, then there's really nothing more to be done here. I'm closing this ticket awaiting more info from others if/as it comes up.

closed 2019-04-30 by michaelld