gnuradio/volk #243: Missing NEON implementations

ast, 2019-04-27:
Here's a list of kernels missing NEON implementations. Some of these are easy, some are hard, some would not provide any benefit. I'm determined to write as many as I can, so I'm creating this list to keep track.

- [ ] volk_32fc_s32f_x2_power_spectral_density_32f.h
- [ ] volk_32f_tan_32f.h
- [ ] volk_32f_s32f_stddev_32f.h
- [ ] volk_32f_s32f_convert_16i.h
- [ ] volk_32f_cos_32f.h
- [ ] volk_32f_s32f_convert_8i.h
- [ ] volk_32fc_index_max_16u.h
- [ ] volk_32fc_s32fc_x2_rotator_32fc.h
- [ ] volk_8ic_deinterleave_real_16i.h
- [ ] volk_32f_null_32f.h
- [ ] volk_8ic_s32f_deinterleave_32f_x2.h
- [ ] volk_32f_x2_pow_32f.h
- [ ] volk_32u_popcntpuppet_32u.h
- [ ] volk_16ic_s32f_magnitude_32f.h
- [ ] volk_16i_permute_and_scalar_add.h
- [ ] volk_32f_atan_32f.h
- [ ] volk_32fc_deinterleave_64f_x2.h
- [ ] volk_32f_64f_multiply_64f.h
- [ ] volk_8u_conv_k7_r2puppet_8u.h
- [ ] volk_32f_s32f_normalize.h
- [ ] volk_8ic_s32f_deinterleave_real_32f.h
- [ ] volk_32fc_s32f_atan2_32f.h
- [ ] volk_16ic_s32f_deinterleave_real_32f.h
- [ ] volk_32f_64f_add_64f.h
- [ ] volk_32f_s32f_convert_32i.h
- [ ] volk_32f_x2_s32f_interleave_16ic.h
- [ ] volk_16ic_deinterleave_real_16i.h
- [ ] volk_32f_x2_fm_detectpuppet_32f.h
- [ ] volk_32f_sin_32f.h
- [ ] volk_8u_x3_encodepolarpuppet_8u.h
- [ ] volk_64f_convert_32f.h
- [ ] volk_8ic_deinterleave_16i_x2.h
- [ ] volk_32f_convert_64f.h
- [ ] volk_32fc_s32f_magnitude_16i.h
- [ ] volk_16ic_magnitude_16i.h
- [ ] volk_64f_x2_max_64f.h
- [ ] volk_32f_x2_dot_prod_16i.h
- [ ] volk_8u_x2_encodeframepolar_8u.h
- [ ] volk_32f_expfast_32f.h
- [ ] volk_8ic_x2_s32f_multiply_conjugate_32fc.h
- [ ] volk_32f_s32f_s32f_mod_range_32f.h
- [ ] volk_16i_branch_4_state_8.h
- [ ] volk_32f_binary_slicer_32i.h
- [ ] volk_64f_x2_add_64f.h
- [ ] volk_32f_s32f_power_32f.h
- [ ] volk_64f_x2_multiply_64f.h
- [ ] volk_32i_s32f_convert_32f.h
- [ ] volk_8u_x4_conv_k7_r2_8u.h
- [ ] volk_64f_x2_min_64f.h
- [ ] volk_32f_accumulator_s32f.h
- [ ] volk_32f_s32f_mod_rangepuppet_32f.h
- [ ] volk_32f_index_max_16u.h
- [ ] volk_32fc_s32f_power_spectrum_32f.h
- [ ] volk_8u_x3_encodepolar_8u_x2.h
- [ ] volk_32fc_s32f_power_32fc.h
- [ ] volk_32fc_index_max_32u.h
- [ ] volk_32f_s32f_32f_fm_detect_32f.h
- [ ] volk_32f_acos_32f.h
- [ ] volk_32f_asin_32f.h
- [ ] volk_32u_popcnt.h
- [ ] volk_16ic_deinterleave_16i_x2.h
- [ ] volk_32fc_s32fc_rotatorpuppet_32fc.h
- [ ] volk_32fc_x2_s32f_square_dist_scalar_mult_32f.h
- [ ] volk_32f_s32f_calc_spectral_noise_floor_32f.h
- [ ] volk_32f_tanh_32f.h
- [ ] volk_32f_8u_polarbutterfly_32f.h
- [ ] volk_8ic_x2_multiply_conjugate_16ic.h
- [ ] volk_32f_8u_polarbutterflypuppet_32f.h
- [ ] volk_32f_stddev_and_mean_32f_x2.h
- [ ] volk_32fc_deinterleave_real_64f.h
- [ ] volk_32fc_s32f_deinterleave_real_16i.h

n-west, 2019-04-28:
Excellent! If you don't mind me asking what's your motivation? It usually helps to prioritize kernels that you care about for your application of interest unless you're just trying to learn more about NEON/SIMD

ast, 2019-04-29:
We're building an SDR "hat" for the raspberry pi, tujasdr.com. I already have some private implementations so might as well try to get them into volk. The others I will probably try to implement as I go along.

Most are quite easy to do because the avx implementations are often very similar.

ast, 2019-05-02:
@n-west are you the maintainer now?

vielmetti, 2019-05-02:
Perhaps of interest to @lemire ?

lemire, 2019-05-02:
Yes. Following.

ast, 2019-05-03:
@vielmetti @lemire thanks for the support!

bhilburn, 2019-05-07:
@ast - Andrej (@noc0lour) is nominally maintaining volk right now. This is really cool work - thanks so much for sharing it and keeping us posted on your progress!

noc0lour, 2019-05-09:
I'm the only one merging PRs but everyone of @gnuradio/gr-officers is entitled to merge things.

noc0lour, 2019-05-17:
@ast do get your work merged once you are done please contact bhilburn@gnuradio.org for getting a CLA in place, thanks (:

dmiralles2009, 2019-08-24:
Is this list updated?. Planning to add some additional NEON support but not sure what has been completed?

ast, 2019-08-26:
@dmiralles2009 @bhilburn yes it's updated!

I would suggest you start at the bottom. Also I had some trouble with atan2 and the functions based on atan2, that might be interesting for you? Ping me if you have any questions!

ast, 2019-08-26:
Great! atan works fine but I lose precision in the division required in atan2... Not sure how to solve that.

ast, 2019-08-30:
@dmiralles2009 Did you a chance to have a look at it?

dmiralles2009, 2019-08-30:
Hey @ast ..Yes, I have been looking at it. I think I will be able to put some code during the weekend. I will keep you posted :)

ast, 2019-08-30:
No stress, just tell me if  you need some pointers.

I implemented these today:
* volk_32fc_s32fc_rotate_up_halfpi_32fc
* volk_32fc_s32f_rotate_pi_32fc
* volk_32fc_substract_real_imag_32f
* volk_32fc_add_real_imag_32f

dmiralles2009, 2019-08-30:
@ast Jaja, oh man you are on fire!!. I like the friendly pressure :).
This week is crazy with work but I think the weekend will be good. BTW, since you offered, do you cross-compile for the RPI right? or do you compile natively on the board?. 
Any info on how to set up the platform will help me speed things up. Thanks

ast, 2019-08-30:
Volk I compile native on the raspberry pi 4.
Just install the normal build tools.
Remember you need to use the correct toolchain file (included with Volk):
`cmake -DCMAKE_TOOLCHAIN_FILE=../cmake/Toolchains/arm_cortex_a72_hardfp_native.cmake ..`

I normally mount the device using sshfs so I can edit on my laptop.

I have rootfs on an SSD on USB3, SD-cards are painfully slow.

igorauad, 2019-08-31:
@dmiralles2009 maybe this is helpful https://wiki.gnuradio.org/index.php/Cross_compile_for_Raspberry_Pi

I wrote this (@ast helped too), but still need to link it into a top-level page in the wiki.

This write-up came from [this gist](https://gist.github.com/igorauad/2ddb62a1341d5c87beb3b4e38104e32f)

dmiralles2009, 2019-09-03:
Thanks @igorauad and @ast . Got 4 protokernels done from the list, I should upload to code later today. Running late for work ....:)

ast, 2019-09-03:
Well done!! Fork volk and push your changes so we can review!

dmiralles2009, 2019-09-05:
Hey @ast, so just got a new RPI4 to further develop some volks and now I am unable to properly compile. This is odd because with RPI3+ was working. Any clues?
```
pi@raspberrypi:~/Documents/pi-volk/build $ cmake -DCMAKE_TOOLCHAIN_FILE=../cmake/Toolchains/arm_cortex_a72_hardfp_native.cmake ..

-- Architecture is not x86 or x86_64, Overruled arch 3dnow
-- Architecture is not x86 or x86_64, Overruled arch mmx
-- Architecture is not x86 or x86_64, Overruled arch sse
-- Architecture is not x86 or x86_64, Overruled arch sse2
-- Architecture is not x86 or x86_64, Overruled arch sse3
-- Architecture is not x86 or x86_64, Overruled arch ssse3
-- Architecture is not x86 or x86_64, Overruled arch sse4_a
-- Architecture is not x86 or x86_64, Overruled arch sse4_1
-- Architecture is not x86 or x86_64, Overruled arch sse4_2
-- Architecture is not x86 or x86_64, Overruled arch avx
-- Architecture is not x86 or x86_64, Overruled arch avx512f
-- Architecture is not x86 or x86_64, Overruled arch avx512cd
-- Performing Test neon_compile_result
-- Performing Test neon_compile_result - Success
-- Performing Test have_neonv7_result
-- Performing Test have_neonv7_result - Success
-- Performing Test have_neonv8_result
-- Performing Test have_neonv8_result - Failed
-- CPU is armv7, Overruled arch neonv8
-- ORC support not found, Overruled arch orc
-- Available architectures: generic;hardfp;neon;neonv7;norc
-- Available machines: generic;neon;neonv7_hardfp
-- BUILD TYPE = RELEASE
-- Base cflags = -O3 -DNDEBUG -march=armv8-a -mtune=cortex-a72 -mfpu=neon-fp-armv8 -mfloat-abi=hard -Wall
-- BUILD INFO ::: generic ::: GNU ::: -O3 -DNDEBUG -march=armv8-a -mtune=cortex-a72 -mfpu=neon-fp-armv8 -mfloat-abi=hard -Wall 
-- BUILD INFO ::: neon ::: GNU ::: -O3 -DNDEBUG -march=armv8-a -mtune=cortex-a72 -mfpu=neon-fp-armv8 -mfloat-abi=hard -Wall -funsafe-math-optimizations
-- BUILD INFO ::: neonv7_hardfp ::: GNU ::: -O3 -DNDEBUG -march=armv8-a -mtune=cortex-a72 -mfpu=neon-fp-armv8 -mfloat-abi=hard -Wall -funsafe-math-optimizations -mfpu=neon -funsafe-math-optimizations -mfloat-abi=hard
-- Compiler Version: gcc (Raspbian 8.3.0-6+rpi1) 8.3.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
-- ---- Adding ASM files
-- -- Detected neon architecture; enabling ASM
-- Adding source file: /home/pi/Documents/pi-volk/kernels/volk/asm/neon/volk_16i_max_star_horizontal_16i.s
-- Adding source file: /home/pi/Documents/pi-volk/kernels/volk/asm/neon/volk_32f_s32f_multiply_32f_a_neonasm.s
-- Adding source file: /home/pi/Documents/pi-volk/kernels/volk/asm/neon/volk_32f_x2_add_32f_a_neonasm.s
-- Adding source file: /home/pi/Documents/pi-volk/kernels/volk/asm/neon/volk_32f_x2_add_32f_a_neonpipeline.s
-- Adding source file: /home/pi/Documents/pi-volk/kernels/volk/asm/neon/volk_32f_x2_dot_prod_32f_a_neonasm.s
-- Adding source file: /home/pi/Documents/pi-volk/kernels/volk/asm/neon/volk_32f_x2_dot_prod_32f_a_neonasm_opts.s
-- Adding source file: /home/pi/Documents/pi-volk/kernels/volk/asm/neon/volk_32fc_32f_dot_prod_32fc_a_neonasm.s
-- Adding source file: /home/pi/Documents/pi-volk/kernels/volk/asm/neon/volk_32fc_32f_dot_prod_32fc_a_neonasmvmla.s
-- Adding source file: /home/pi/Documents/pi-volk/kernels/volk/asm/neon/volk_32fc_32f_dot_prod_32fc_a_neonpipeline.s
-- Adding source file: /home/pi/Documents/pi-volk/kernels/volk/asm/neon/volk_32fc_32f_dot_prod_32fc_a_unrollasm.s
-- Adding source file: /home/pi/Documents/pi-volk/kernels/volk/asm/neon/volk_32fc_x2_dot_prod_32fc_a_neonasm.s
-- Adding source file: /home/pi/Documents/pi-volk/kernels/volk/asm/neon/volk_32fc_x2_dot_prod_32fc_a_neonasm_opttests.s
-- Adding source file: /home/pi/Documents/pi-volk/kernels/volk/asm/neon/volk_32fc_x2_multiply_32fc_a_neonasm.s
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/gcc
-- c flags: -march=armv8-a -mtune=cortex-a72 -mfpu=neon-fp-armv8 -mfloat-abi=hard -Wall;
-- asm flags:  -mfpu=neon -g
-- c flags: -march=armv8-a -mtune=cortex-a72 -mfpu=neon-fp-armv8 -mfloat-abi=hard -Wall;
-- asm flags:  -mfpu=neon -g -mfpu=neon -g
-- Did not find liborc and orcc, disabling orc support...
-- Loading version 2.0 into constants...
-- Using install prefix: /usr/local
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/Documents/pi-volk/build
pi@raspberrypi:~/Documents/pi-volk/build $ make
[  2%] Generating volk_machine_neonv7_hardfp.c
[  4%] Generating ../include/volk/volk.h
[  6%] Generating volk.c
[  8%] Generating ../include/volk/volk_typedefs.h
[ 10%] Generating ../include/volk/volk_cpu.h
[ 12%] Generating volk_cpu.c
[ 14%] Generating ../include/volk/volk_config_fixed.h
[ 16%] Generating volk_machines.h
[ 18%] Generating volk_machines.c
[ 20%] Generating volk_machine_generic.c
[ 22%] Generating volk_machine_neon.c
Scanning dependencies of target volk_obj
[ 25%] Building ASM object lib/CMakeFiles/volk_obj.dir/__/kernels/volk/asm/neon/volk_16i_max_star_horizontal_16i.s.o
[ 27%] Building ASM object lib/CMakeFiles/volk_obj.dir/__/kernels/volk/asm/neon/volk_32f_s32f_multiply_32f_a_neonasm.s.o
[ 29%] Building ASM object lib/CMakeFiles/volk_obj.dir/__/kernels/volk/asm/neon/volk_32f_x2_add_32f_a_neonasm.s.o
[ 31%] Building ASM object lib/CMakeFiles/volk_obj.dir/__/kernels/volk/asm/neon/volk_32f_x2_add_32f_a_neonpipeline.s.o
[... cut ...]
/home/pi/Documents/pi-volk/kernels/volk/asm/neon/volk_32f_x2_dot_prod_32f_a_neonasm_opts.s:46: Error: selected processor does not support `sbfx r11,r1,#2,#1' in ARM mode
make[2]: *** [lib/CMakeFiles/volk_obj.dir/build.make:1639: lib/CMakeFiles/volk_obj.dir/__/kernels/volk/asm/neon/volk_32f_x2_dot_prod_32f_a_neonasm_opts.s.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:159: lib/CMakeFiles/volk_obj.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
pi@raspberrypi:~/Documents/pi-volk/build $ ```

ast, 2019-09-06:
Yeah ok this is a known bug, thought I already had submitted a change for that..? My toolchain file looks like this. Notice `CMAKE_ASM_FLAGS` and ` -mthumb` which is needed to compile that particular asm file that is failing.

```
set(CMAKE_CXX_COMPILER g++)
set(CMAKE_C_COMPILER  gcc)
set(CMAKE_CXX_FLAGS "-ffast-math -march=armv8-a -mtune=cortex-a72 -mfpu=neon-fp-armv8 -mfloat-abi=hard" CACHE STRING "" FORCE)
set(CMAKE_C_FLAGS ${CMAKE_CXX_FLAGS} CACHE STRING "" FORCE) #same flags for C sources
set(CMAKE_ASM_FLAGS "${CMAKE_CXX_FLAGS} -mthumb -g" CACHE STRING "" FORCE) #same flags for asm sources
```

You might have to remove the line`set(ARCH_ASM_FLAGS "-mfpu=neon -g")` in `lib/CMakeLists.txt` also so the variable is not overwritten.

dmiralles2009, 2019-09-09:
Hi @ast , that was a good fix. I do not think the changes are available in volk (master). Maybe they are on your repo. I switched development to my RPI 3b+ board, I was unable to get a 64 bit OS functional in RPI 4. Have you been luckier in this regard?.
I did complete a couple of proto-kernels

1.  volk_8ic_s32f_deinterleave_real_32f.h
2. volk_8ic_x2_multiply_conjugate_16ic.h
3. volk_8ic_s32f_deinterleave_32f_x2.h
4. volk_32fc_deinterleave_64f_x2.h
5. volk_32fc_deinterleave_real_64f.h
6.  volk_32f_64f_add_64f.h

but I have been trying to run those in rpi4 to validate results. I should be pushing then soon.

ast, 2019-09-09:
I had Ubuntu aarch64 working very well on rbpi3 using the raspbian
bootloader (to correctly load the device tree). If I remember correctly I
also added some users/groups that raspberry pi software expects.

Just try to edit the Cmake and toolchain flags to add the -mthumb flag. OR
you can just rename/remove the assembly file that fails on Raspberry Pi 4,
this is the easiest fix, you don't need it anyway.

michaelld, 2019-11-14:
@ast Is this list up to date? If not, can you get it updated? Thanks for your work getting NEON kernels into Volk!

ast, 2019-11-15:
It's up to date! All the code is in my linked repo. I have limited time to create PRs so everyone is welcome to look at/improve upon my work and create PRs!

dmiralles2009, 2019-11-16:
I also have completed the proto kernel I listed but I am busy with paying
jobs. In a couple of days, I will submit a new PR.

@ast you ok if I submit your code changes on your behalf?. Having those on
the mainstream repo will be quite helpful.

Best,
Damian

ast, 2019-11-16:
@dmiralles2009 yes go ahead! That would be great! Please update the list if you do! Or ping me and I'll update it...

jdemel, 2021-02-21:
@ast do you still have kernels in your repo that are not yet merged? It would be great to see your work merged upstream.

I added a recent overview of [kernels per platform on libvolk.org](https://www.libvolk.org/platforms.html).

Ka-zam, 2021-04-14:
I can work on some of these as well now as I have recently got my aarch64 environment up and running.

Aang23, 2022-08-09:
This is just a suggestion, but for some (more complicated) kernels, a conversion to NEON from SSE using something like sse2neon works pretty decently. I wouldn't especially include the header in volk but perhaps simply the generated NEON code after preprocessing.

I did this on the convolutional decoding kernel obtaining a significant performance boost.
It's not as good as a kernel optimized specifically for NEON, but I'd consider it better than nothing in most cases, especially on complicated kernels like spiral which is machine-generated.

jdemel, 2022-08-09:
This sounds like an interesting idea. Do you have a link to the sse2neon tool?

playaspec, 2022-08-09:
It's a single header file. Project is on on github under the same name.

[https://github.com/DLTcollab/sse2neon](https://github.com/DLTcollab/sse2neon)

ast, 2022-08-09:
I think it's a pretty good idea. Just mark them as machine-generated in a comment so it's easy to know which to (maybe) try to improve. With more recent compiler versions and ARM architectures, I have noted diminishing returns in hand written NEON code. But it's always interesting to try!!

Aang23, 2022-08-09:
That sounds very good. Mark them as a conversion of the SSE implementation, for possible future rewrites or improvements.
I have a tool nearly finished to produce NEON code from SSE source (as to now include the actual header, and be able to modify the actual NEON code).

If this is all fine, I could PR my port of the Viterbi kernel to NEON (alongside tests results once I test it on a non-Android ARM platform for a proper performance check), and perhaps a few others kernels later.

Aang23, 2022-08-10:
Ok, after more testing around, it would quite a bit simpler to have sse2neon as a header in Volk. The license is MIT so should be no problem with LGPL as far as I am aware.

The last question that remains then, in case a SSE kernel is made NEON-compatible with this method, duplicating the kernel feels redundant. In the case of the convolutional decoder, I am just swapping out SSE headers with sse2neon if NEON support is detected. The actual kernel codes remains identical.

As spiral is the kernel arch name in this instance, not an issue, but what about a "sse" kernel? Should it conserve its SSE naming (which does not seem to affect Volk in any negative way), or be changed over to "neon" or perhaps sse_neon or something else?

Personally I'd think keeping the original SSE naming even on NEON would reflect the implementation being utilized is a "port" of the SSE kernel, and not something purpose-made for NEON. If a kernel ported this way is not too complicated to swap out SSE intrinsics to NEON's, it could perhaps become a "proper" NEON kernel?

Those are just some ideas I had thinking about how starting to go this route could affect Volk outside of already sligtly different kernels.