gnuradio/volk #247: Warnings when building from source on macOS

citizenrich, 2019-05-21:
I'm working on making gnuradio build from source on mac. When building just gnuradio-runtime and volk, one test fails that relate to volk. I'm tracing the build steps and I want to share some warnings to understand if this could be a source of issue for other tests for gnuradio-runtime that are also failing. Thoughts?

Failed test:
```
Test  #85: qa_volk_32fc_s32f_magnitude_16i ....................***Failed    0.07 sec
```

Make output warnings:
```
[ 28%] Building C object volk/lib/CMakeFiles/volk_obj.dir/volk_16ic_magnitude_16i_a_orc_impl.c.o
cd /Users/richard/src/github.com/gnuradio/gnuradio/build/volk/lib && /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc -DGR_PERFORMANCE_COUNTERS -DHAVE_CPUID_H -DHAVE_DLFCN_H -DHAVE_FENV_H -DHAVE_POSIX_MEMALIGN -DHAVE_XGETBV -D_GLIBCXX_USE_CXX11_ABI=1 -I/usr/local/Cellar/orc/0.4.28/include/orc-0.4 -I/Users/richard/src/github.com/gnuradio/gnuradio/build/volk/include -I/Users/richard/src/github.com/gnuradio/gnuradio/volk/include -I/Users/richard/src/github.com/gnuradio/gnuradio/volk/kernels -I/Users/richard/src/github.com/gnuradio/gnuradio/build/volk/lib -I/Users/richard/src/github.com/gnuradio/gnuradio/volk/lib  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -Wall -fvisibility=hidden -g -O2 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk   -fPIC -std=c11 -o CMakeFiles/volk_obj.dir/volk_16ic_magnitude_16i_a_orc_impl.c.o   -c /Users/richard/src/github.com/gnuradio/gnuradio/build/volk/lib/volk_16ic_magnitude_16i_a_orc_impl.c
/Users/richard/src/github.com/gnuradio/gnuradio/build/volk/lib/volk_16ic_magnitude_16i_a_orc_impl.c:374:16: warning: comparison of integers of different signs: 'int' and 'unsigned int' [-Wsign-compare]
       if (tmp == 0x80000000 && !(var52.i&0x80000000)) tmp = 0x7fffffff;
           ~~~ ^  ~~~~~~~~~~
1 warning generated.
...

[ 38%] Building C object volk/lib/CMakeFiles/volk_obj.dir/volk_32fc_s32f_magnitude_16i_a_orc_impl.c.o
cd /Users/richard/src/github.com/gnuradio/gnuradio/build/volk/lib && /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc -DGR_PERFORMANCE_COUNTERS -DHAVE_CPUID_H -DHAVE_DLFCN_H -DHAVE_FENV_H -DHAVE_POSIX_MEMALIGN -DHAVE_XGETBV -D_GLIBCXX_USE_CXX11_ABI=1 -I/usr/local/Cellar/orc/0.4.28/include/orc-0.4 -I/Users/richard/src/github.com/gnuradio/gnuradio/build/volk/include -I/Users/richard/src/github.com/gnuradio/gnuradio/volk/include -I/Users/richard/src/github.com/gnuradio/gnuradio/volk/kernels -I/Users/richard/src/github.com/gnuradio/gnuradio/build/volk/lib -I/Users/richard/src/github.com/gnuradio/gnuradio/volk/lib  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -Wall -fvisibility=hidden -g -O2 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk   -fPIC -std=c11 -o CMakeFiles/volk_obj.dir/volk_32fc_s32f_magnitude_16i_a_orc_impl.c.o   -c /Users/richard/src/github.com/gnuradio/gnuradio/build/volk/lib/volk_32fc_s32f_magnitude_16i_a_orc_impl.c
/Users/richard/src/github.com/gnuradio/gnuradio/build/volk/lib/volk_32fc_s32f_magnitude_16i_a_orc_impl.c:352:16: warning: comparison of integers of different signs: 'int' and 'unsigned int' [-Wsign-compare]
       if (tmp == 0x80000000 && !(var50.i&0x80000000)) tmp = 0x7fffffff;
           ~~~ ^  ~~~~~~~~~~
1 warning generated.
...

[ 51%] Building C object volk/lib/CMakeFiles/volk_obj.dir/volk_machine_avx_64_mmx_orc.c.o
cd /Users/richard/src/github.com/gnuradio/gnuradio/build/volk/lib && /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc -DGR_PERFORMANCE_COUNTERS -DHAVE_CPUID_H -DHAVE_DLFCN_H -DHAVE_FENV_H -DHAVE_POSIX_MEMALIGN -DHAVE_XGETBV -D_GLIBCXX_USE_CXX11_ABI=1 -I/usr/local/Cellar/orc/0.4.28/include/orc-0.4 -I/Users/richard/src/github.com/gnuradio/gnuradio/build/volk/include -I/Users/richard/src/github.com/gnuradio/gnuradio/volk/include -I/Users/richard/src/github.com/gnuradio/gnuradio/volk/kernels -I/Users/richard/src/github.com/gnuradio/gnuradio/build/volk/lib -I/Users/richard/src/github.com/gnuradio/gnuradio/volk/lib  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -Wall -fvisibility=hidden -g -O2 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk   -fPIC -std=c11 -m64 -mmmx -msse -msse2 -msse3 -mssse3 -msse4.1 -msse4.2 -mpopcnt -mavx -o CMakeFiles/volk_obj.dir/volk_machine_avx_64_mmx_orc.c.o   -c /Users/richard/src/github.com/gnuradio/gnuradio/build/volk/lib/volk_machine_avx_64_mmx_orc.c
In file included from /Users/richard/src/github.com/gnuradio/gnuradio/build/volk/lib/volk_machine_avx_64_mmx_orc.c:142:
/Users/richard/src/github.com/gnuradio/gnuradio/volk/kernels/volk/volk_32fc_x2_conjugate_dot_prod_32fc.h:131:38: warning: comparison of integers of different signs: 'int' and 'unsigned int' [-Wsign-compare]
  for (point=quarter_points*4; point < num_points; ++point) {
                               ~~~~~ ^ ~~~~~~~~~~
In file included from /Users/richard/src/github.com/gnuradio/gnuradio/build/volk/lib/volk_machine_avx_64_mmx_orc.c:147:
/Users/richard/src/github.com/gnuradio/gnuradio/volk/kernels/volk/volk_32fc_x2_s32f_square_dist_scalar_mult_32f.h:203:12: warning: comparison of integers of different signs: 'int' and 'unsigned int' [-Wsign-compare]
  for (; i < avx_work_size; i += work_size) {
         ~ ^ ~~~~~~~~~~~~~
/Users/richard/src/github.com/gnuradio/gnuradio/volk/kernels/volk/volk_32fc_x2_s32f_square_dist_scalar_mult_32f.h:239:12: warning: comparison of integers of different signs: 'int' and 'unsigned int' [-Wsign-compare]
  for (; i < num_points; ++i) {
         ~ ^ ~~~~~~~~~~
3 warnings generated.
...
```

michaelld, 2019-05-21:
Hi @citizenrich -- The test failure & warnings are normal right now. Can you execute the following:
{{{
ctest -V -R qa_volk_32fc_s32f_magnitude_16i
}}}
and post back the output? I'll look into the warnings & see about fixing them. - MLD

citizenrich, 2019-05-21:
Thanks. In the latest attempt I only built gnuradio-runtime and volk to try to isolate the issues so if I may need to do a proper fuller build to actually get the tests...

```
$ ctest -V -R qa_volk_32fc_s32f_magnitude_16i
UpdateCTestConfiguration  from :/Users/richard/src/github.com/gnuradio/gnuradio/build/DartConfiguration.tcl
UpdateCTestConfiguration  from :/Users/richard/src/github.com/gnuradio/gnuradio/build/DartConfiguration.tcl
Test project /Users/richard/src/github.com/gnuradio/gnuradio/build
Constructing a list of tests
Done constructing a list of tests
Updating test list for fixtures
Added 0 tests to meet fixture requirements
Checking test dependency graph...
Checking test dependency graph end
No tests were found!!!
```

michaelld, 2019-05-21:
Thinking the "ctest" command is for the Volk build; not sure how you did that one, but it's shouldn't be in the top-level GR build directory. Might be in a "Volk" subdirectory ...

Tests are in general added only if their component is active / built; and, otherwise not. I think there might be a couple stray CTRLPORT tests still that misbehave, but these are just tests; they don't impact runtime.

citizenrich, 2019-05-21:
Thanks for your patience.

```
$ ctest -V -R qa_volk_32fc_s32f_magnitude_16i
UpdateCTestConfiguration  from :/Users/richard/src/github.com/gnuradio/gnuradio/build/volk/DartConfiguration.tcl
UpdateCTestConfiguration  from :/Users/richard/src/github.com/gnuradio/gnuradio/build/volk/DartConfiguration.tcl
Test project /Users/richard/src/github.com/gnuradio/gnuradio/build/volk
Constructing a list of tests
Done constructing a list of tests
Updating test list for fixtures
Added 0 tests to meet fixture requirements
Checking test dependency graph...
Checking test dependency graph end
test 85
    Start 85: qa_volk_32fc_s32f_magnitude_16i

85: Test command: /bin/sh "/Users/richard/src/github.com/gnuradio/gnuradio/build/volk/lib/volk_32fc_s32f_magnitude_16i_test.sh" "/Users/richard/src/github.com/gnuradio/gnuradio/build/volk/lib"
85: Test timeout computed to be: 10000000
85: RUN_VOLK_TESTS: volk_32fc_s32f_magnitude_16i(131071,1)
85: a_sse3 completed in 0.268ms
85: a_sse completed in 0.291ms
85: generic completed in 0.383ms
85: u_orc completed in 1.609ms
85: offset 18022 in1: 318 in2: 319 tolerance was: 0
85: offset 74279 in1: 282 in2: 283 tolerance was: 0
85: volk_32fc_s32f_magnitude_16i: fail on arch u_orc
85: Best aligned arch: a_sse3
85: Best unaligned arch: generic
1/1 Test #85: qa_volk_32fc_s32f_magnitude_16i ...***Failed    0.03 sec

0% tests passed, 1 tests failed out of 1

Total Test time (real) =   0.04 sec

The following tests FAILED:
	 85 - qa_volk_32fc_s32f_magnitude_16i (Failed)
Errors while running CTest
```

michaelld, 2019-05-21:
NP & thx! That's what I thought it would come back with. I think we're just going to disable ORC for that kernel. It hasn't worked in a long time & shows no signs of being fixed, and the only platforms it's fastest on are the early OSX Intel for 10.5 and maybe 10.6 (mostly the 32-bit ones). As you can tell here, it is -much- slower than even the generic kernel, which is also true on 10.4 and 10.5 PPC. That ORC implementation just doesn't make the cut & since it isn't accurate, removing it seems like the best way to go.

citizenrich, 2019-05-21:
Yep, that gets past that error. gnuradio-runtime are still failing but that would be another issue. Thanks!

```
$ ctest -V -R qa_volk_32fc_s32f_magnitude_16i
UpdateCTestConfiguration  from :/Users/richard/src/github.com/gnuradio/gnuradio/build/volk/DartConfiguration.tcl
UpdateCTestConfiguration  from :/Users/richard/src/github.com/gnuradio/gnuradio/build/volk/DartConfiguration.tcl
Test project /Users/richard/src/github.com/gnuradio/gnuradio/build/volk
Constructing a list of tests
Done constructing a list of tests
Updating test list for fixtures
Added 0 tests to meet fixture requirements
Checking test dependency graph...
Checking test dependency graph end
test 85
    Start 85: qa_volk_32fc_s32f_magnitude_16i

85: Test command: /bin/sh "/Users/richard/src/github.com/gnuradio/gnuradio/build/volk/lib/volk_32fc_s32f_magnitude_16i_test.sh" "/Users/richard/src/github.com/gnuradio/gnuradio/build/volk/lib"
85: Test timeout computed to be: 10000000
85: RUN_VOLK_TESTS: volk_32fc_s32f_magnitude_16i(131071,1)
85: a_sse3 completed in 0.347ms
85: a_sse completed in 0.805ms
85: generic completed in 0.859ms
85: Best aligned arch: a_sse3
85: Best unaligned arch: generic
1/1 Test #85: qa_volk_32fc_s32f_magnitude_16i ...   Passed    0.03 sec

The following tests passed:
	qa_volk_32fc_s32f_magnitude_16i

100% tests passed, 0 tests failed out of 1
```

closed 2019-05-21 by citizenrich

michaelld, 2019-05-21:
Glad that worked! Thanks for your efforts & reporting back! I'll see about disabling just that test's use of ORC && fixing those compile warnings.

michaelld, 2019-05-22:
FYI: that kernel's orc implementation has been removed in GIT ... it'll be gone in the next release: https://github.com/gnuradio/volk/commit/ff0751705ac323333f1276900df077888af743da