gnuradio/volk #252: CMake does not respect explicitly set Python Interpreter

jdemel, 2019-05-24:
If I run `cmake -DPYTHON_EXECUTABLE=/usr/bin/python3`, CMake will finally reset this value and I end up with Py2k. I checked and this happens in `cmake/Modules/VolkPython.cmake`. Before line 31, the variable is still set correctly and after the following if/else block, it is set to Py2k again.
One can fix this by running cmake once, editing the cache and then rerun cmake.

hcab14, 2019-05-28:
Which version of CMake do you use? Using version 3.13.3 I can set the python executable to python3.

jdemel, 2019-05-29:
This happened in an Ubuntu 18.04 (CMake 3.10) and a 19.04 (CMake 3.13) environment.

jdemel, 2019-09-14:
The latest `master` does work with `cmake -DPYTHON_EXECUTABLE=/usr/bin/python3` as opposed to the version I used when I issued this. I suggest to close this issue.

jdemel, 2019-11-09:
Issue is resolved. I'm closing this issue.

closed 2019-11-09 by jdemel