gnuradio/volk #253: qa_volk_32fc_s32f_magnitude_16i fails on SSE,SSE3,AVX2

noc0lour, 2019-05-28:
Something is still funky here: https://travis-ci.org/gnuradio/volk/jobs/537817549

noc0lour, 2019-07-20:
```
85: Test command: /bin/sh "/home/travis/build/gnuradio/volk/build/lib/volk_32fc_s32f_magnitude_16i_test.sh" "/home/travis/build/gnuradio/volk/build/lib"
85: Test timeout computed to be: 9.99988e+06
85: RUN_VOLK_TESTS: volk_32fc_s32f_magnitude_16i(131071,1)
85: a_avx2 completed in 0.129919 ms
85: a_sse3 completed in 0.552591 ms
85: a_sse completed in 0.509194 ms
85: generic completed in 0.556202 ms
85: u_avx2 completed in 0.111349 ms
85: offset 64578 in1: 323 in2: 322 tolerance was: 0
85: volk_32fc_s32f_magnitude_16i: fail on arch a_avx2
85: offset 64578 in1: 323 in2: 322 tolerance was: 0
85: volk_32fc_s32f_magnitude_16i: fail on arch a_sse3
85: offset 64578 in1: 323 in2: 322 tolerance was: 0
85: volk_32fc_s32f_magnitude_16i: fail on arch u_avx2
85: Best aligned arch: u_avx2
85: Best unaligned arch: u_avx2
 85/130 Test  #85: qa_volk_32fc_s32f_magnitude_16i ....................***Failed    0.02 sec
test 86
        Start  86: qa_volk_32fc_s32f_power_32fc
```
https://travis-ci.org/gnuradio/volk/jobs/558651966

michaelld, 2019-10-27:
Guessing Closed by #285 ... so closing & please reopen if the issue persists.

closed 2019-10-27 by michaelld