gnuradio/volk #254: qa_volk_32f_x3_sum_of_poly_32f

noc0lour, 2019-05-28:
https://travis-ci.org/gnuradio/volk/jobs/537817550#L1883
```
        Start  68: qa_volk_32f_x3_sum_of_poly_32f
68: Test command: /bin/sh "/home/travis/build/gnuradio/volk/build/lib/volk_32f_x3_sum_of_poly_32f_test.sh" "/home/travis/build/gnuradio/volk/build/lib"
68: Test timeout computed to be: 9.99988e+06
68: RUN_VOLK_TESTS: volk_32f_x3_sum_of_poly_32f(131071,1)
68: a_sse3 completed in 0.086484 ms
68: a_avx2_fma completed in 0.105624 ms
68: a_avx completed in 0.062776 ms
68: generic completed in 0.379732 ms
68: u_avx_fma completed in 0.059967 ms
68: u_avx completed in 0.064478 ms
68: offset 0 in1: -186.28 in2: -181.125 tolerance was: 0.01
68: volk_32f_x3_sum_of_poly_32f: fail on arch a_sse3
68: Best aligned arch: u_avx_fma
68: Best unaligned arch: u_avx_fma
 68/130 Test  #68: qa_volk_32f_x3_sum_of_poly_32f .....................***Failed    0.02 sec
```

hcab14, 2019-05-28:
This happens intermittently. For me locally this test fails in about ~2% of tests.

These failures are related to the fact hat the sum is done differently in the generic version and in the SIMD versions. As a consequence, the numerical errors add up differently, see, _e.g._, 
https://stackoverflow.com/questions/32098385/an-accumulated-computing-error-in-sse-version-of-algorithm-of-the-sum-of-squared

When modifying the generic version to follow the avx version no failures occur for avx anymore (for ~3000 test runs). However the sum is done differently for sse3 and for the other SIMD kernels, so without specifying a summation order for all kernels the problem cannot be solved.

This problem does also show up in other kernels, _e.g._, in `volk_32f_stddev_and_mean_32f_x2` (3 failures in 3854 tests)

```
53: RUN_VOLK_TESTS: volk_32f_stddev_and_mean_32f_x2(131071,1)
53: a_avx completed in 0.071 ms
53: u_avx completed in 0.08 ms
53: a_sse4_1 completed in 0.077 ms
53: a_sse completed in 0.064 ms
53: generic completed in 0.157 ms
53: offset 0 in1: -4.75153e-07 in2: -4.64941e-07 tolerance was: 0.01
53: volk_32f_stddev_and_mean_32f_x2: fail on arch a_sse4_1
53: offset 0 in1: -4.75153e-07 in2: -4.64867e-07 tolerance was: 0.01
53: volk_32f_stddev_and_mean_32f_x2: fail on arch a_sse
53: Best aligned arch: a_sse
53: Best unaligned arch: u_avx
1/1 Test #53: qa_volk_32f_stddev_and_mean_32f_x2 ...***Failed    0.11 sec
```

noc0lour, 2019-05-29:
Hmmm, thanks for analyzing this. So is there a preferable way of adding up numerical errors? (Maybe even avoiding errors with a small performance penalty.) I did not look into either kernel due to lack of enough spare time.  Will do so in the next couple of days, maybe I'll have an idea.

michaelld, 2019-10-26:
Closed by #287

closed 2019-10-26 by michaelld

ValZapod, 2020-02-10:
https://travis-ci.org/gnuradio/volk/builds/647859794?utm_source=github_status&utm_medium=notification