gnuradio/volk #257: gcc (Raspbian 8.3.0-6+rpi1) 8.3.0 does not properly detect NEON

ast, 2019-06-30:
```
cmake
-- Compiler doesn't support NEON, Overruled arch neon
-- Compiler doesn't support NEON, Overruled arch neonv7
-- Compiler doesn't support NEON, Overruled arch neonv8
```
Manually adding `-mfpu=neon-fp-armv8` did the trick.

Not sure how to add this to the cmake files in a good way.

hcab14, 2019-06-30:
Hi @ast,

did you use one of the toolchains in volk/cmake/Toolchains/, _i.e._, something like  
`cmake -DCMAKE_TOOLCHAIN_FILE=... ../` ?

ast, 2019-06-30:
No, just `cmake ..`, is adding a toolchain file the default way of building now?

ast, 2019-06-30:
Ok created a a toolchain file for raspberry pi 4 now. It's working, thanks.

```
########################################################################
# Toolchain file for building native on a ARM Cortex A72 w/ NEON
# Usage: cmake -DCMAKE_TOOLCHAIN_FILE=<this file> <source directory>
########################################################################
set(CMAKE_CXX_COMPILER g++)
set(CMAKE_C_COMPILER  gcc)
set(CMAKE_CXX_FLAGS "-mtune=cortex-a72 -mfpu=neon-fp-armv8 -mfloat-abi=hard" CACHE STRING "" FORCE)
set(CMAKE_C_FLAGS ${CMAKE_CXX_FLAGS} CACHE STRING "" FORCE) #same flags for C sources
```

hcab14, 2019-06-30:
As I understand, using toolchain files is a way of handling the complexity of different compiler options for different compilers on different CPUs. It might be useful to commit your toolchain file.

ast, 2019-06-30:
It's a good idea. Detection is complicated and error prone. However, this should be reflected in the README.md.

I will make a PR for my file. I also have A LOT of NEON optimizations to PR.

ast, 2019-08-11:
Resolved by
https://github.com/gnuradio/volk/pull/271
https://github.com/gnuradio/volk/pull/270

closed 2019-08-11 by ast