gnuradio/volk #259: gracefully handle if $HOME and $APPDATA is empty and /etc/volk/volk_config is nonexistent

noc0lour, 2019-07-13:
Currently if both environment variables and no system-wide config are empty `volk_profile` tries to use an empty path to save results. 
In any case `volk_profile` should also not save into the system-wide config.

jdemel, 2019-11-07:
Should this function have a negative return value instead? Or should a callee handle this situation? 

```
 $ grep -irnI volk_get_config_path .
./include/volk/volk_prefs.h:22:VOLK_API void volk_get_config_path(char *, bool);
./apps/volk_profile.cc:28:#include <volk/volk_prefs.h>                 // for volk_get_config_path
./apps/volk_profile.cc:159:    volk_get_config_path(path, true);
./apps/volk_profile.cc:216:    volk_get_config_path(path, false);
./lib/volk_prefs.c:14:void volk_get_config_path(char *path, bool read)
./lib/volk_prefs.c:73:    volk_get_config_path(path, true);
```

`volk_prefs.c:73` handles `path[0] == 0`. Those calls in `volk_profile` don't handle this case. Is this a bug in `volk_get_config_path` or in `volk_profile`? We could throw an exception if  `path[0] == 0`.

michaelld, 2019-11-08:
Was this perhaps fixed by @noc0lour in 59a19ed35b7c65b6d1011f5e2f6d9f9c7ba1c6b4 on July 14?

michaelld, 2019-11-08:
Answering my query: no it wasn't.

The issue is [here](https://github.com/gnuradio/volk/blob/master/apps/volk_profile.cc#L221) ... in the function `write_results`. We should check whether the path[0] == 0 & if so then print something about not being able to write the results & return. Or something like that.

michaelld, 2019-11-08:
Even better would be to add in the check (here)[https://github.com/gnuradio/volk/blob/master/apps/volk_profile.cc#L217] and just no call `write_results( results, update_result, std::string(path))` if there is no path returned.

jdemel, 2019-11-08:
@michaelld  Thanks for clarifying this. It's exactly what I tried to express.

noc0lour, 2019-11-08:
Need more input from me or did you figure it out yet?

michaelld, 2019-11-08:
@noc0lour I think we deciphered it ...

PR #305 (2019-11-09): configpath: Abort read/write if path not found
This commit fixes #259.
After a call to `volk_get_config_path` check if `path[0] == 0`. If yes,
return because we could not find a volk config path.
commit d1965e62 (2019-11-09): configpath: Abort read/write if path not found

This commit fixes #259.
After a call to `volk_get_config_path` check if `path[0] == 0`. If yes,
return because we could not find a volk config path.
closed 2019-11-09 by michaelld