gnuradio/volk #261: 32f_log2_32f: flaky kernel

marcusmueller, 2019-07-18:
During CI run of https://github.com/gnuradio/gnuradio/pull/2640, 
https://ci.gnuradio.org/buildbot/#/builders/46/builds/1060/steps/7/logs/stdio
failed: 

```
 38/362 Test  #38: qa_volk_32f_log2_32f ...............................***Failed    0.02 sec
RUN_VOLK_TESTS: volk_32f_log2_32f(131071,1)
a_avx2_fma completed in 0.103887 ms
a_avx2 completed in 0.083858 ms
generic completed in 1.18807 ms
a_sse4_1 completed in 0.188755 ms
u_generic completed in 1.27004 ms
u_sse4_1 completed in 0.138355 ms
u_avx2_fma completed in 0.089172 ms
u_avx2 completed in 0.089164 ms
offset 90473 in1: -inf in2: -127 tolerance was: 1e-05
volk_32f_log2_32f: fail on arch a_avx2_fma
offset 90473 in1: -inf in2: -127 tolerance was: 1e-05
volk_32f_log2_32f: fail on arch a_avx2
offset 90473 in1: -inf in2: -127 tolerance was: 1e-05
volk_32f_log2_32f: fail on arch a_sse4_1
offset 90473 in1: -inf in2: -127 tolerance was: 1e-05
volk_32f_log2_32f: fail on arch u_sse4_1
offset 90473 in1: -inf in2: -127 tolerance was: 1e-05
volk_32f_log2_32f: fail on arch u_avx2_fma
offset 90473 in1: -inf in2: -127 tolerance was: 1e-05
volk_32f_log2_32f: fail on arch u_avx2
Best aligned arch: a_avx2
Best unaligned arch: u_avx2
```

marcusmueller, 2019-07-18:
Since multiple archs fail, chances are the generic impl is –once again– to blame.

Math is hard.

marcusmueller, 2019-07-20:
huh, I'm almost certain I didn't mean to add the [ARM] tag

noc0lour, 2019-07-23:
`-inf` as a result looks weird.

jdemel, 2019-09-22:
There's a fix in #286 . 

So far, the reason for this bug seems to be differing behavior for invalid input. Negative values are invalid input. Besides trying to make all kernels behave equally for invalid input, the solution would be to restrict the random test input values to a valid range.

hcab14, 2019-10-28:
AFAIC this issue can be closed .

closed 2019-10-28 by michaelld