gnuradio/volk #262: 32f_x3_sum_of_poly_32f: flaky kernel

marcusmueller, 2019-07-21:
``` 68/362 Test  #68: qa_volk_32f_x3_sum_of_poly_32f .....................***Failed    0.02 sec
RUN_VOLK_TESTS: volk_32f_x3_sum_of_poly_32f(131071,1)
a_sse3 completed in 0.115966 ms
a_avx2_fma completed in 0.084056 ms
a_avx completed in 0.058172 ms
generic completed in 0.15752 ms
u_avx_fma completed in 0.058003 ms
u_avx completed in 0.062948 ms
offset 0 in1: -116.748 in2: -113.156 tolerance was: 0.01
volk_32f_x3_sum_of_poly_32f: fail on arch a_sse3
offset 0 in1: -116.748 in2: -114.197 tolerance was: 0.01
volk_32f_x3_sum_of_poly_32f: fail on arch a_avx2_fma
offset 0 in1: -116.748 in2: -114.197 tolerance was: 0.01
volk_32f_x3_sum_of_poly_32f: fail on arch a_avx
offset 0 in1: -116.748 in2: -114.197 tolerance was: 0.01
volk_32f_x3_sum_of_poly_32f: fail on arch u_avx_fma
offset 0 in1: -116.748 in2: -114.197 tolerance was: 0.01
volk_32f_x3_sum_of_poly_32f: fail on arch u_avx
Best aligned arch: u_avx_fma
Best unaligned arch: u_avx_fma
```

Again, due to the large number of failing architectures, the generic impl is the suspect.

marcusmueller, 2019-07-21:
oh no, here in2 differ – so there's discrepancy between the SIMD implementations :(

michaelld, 2019-10-26:
Closed by #287

closed 2019-10-26 by michaelld