gnuradio/volk #266: volk_8u_conv_k7_r2puppet_8u : flaky kernel on OSX 10.14

michaelld, 2019-07-31:
Here's the output from `volk_profile`
```
RUN_VOLK_TESTS: volk_8u_conv_k7_r2puppet_8u(131071,1987)
spiral completed in 1008.89 ms
generic completed in 32251.3 ms
offset 0 in1: 1 in2: 0 tolerance was: 0
offset 1 in1: 1 in2: 0 tolerance was: 0
offset 2 in1: 1 in2: 0 tolerance was: 0
offset 4 in1: 1 in2: 0 tolerance was: 0
offset 5 in1: 1 in2: 0 tolerance was: 0
offset 6 in1: 1 in2: 0 tolerance was: 0
offset 8 in1: 1 in2: 0 tolerance was: 0
volk_8u_conv_k7_r2puppet_8u: fail on arch spiral
Best aligned arch: generic
Best unaligned arch: generic
```

jdemel, 2019-09-22:
Is this issue related to #117 ?

hcab14, 2019-10-28:
AFAIC this issue can be closed .

michaelld, 2019-10-28:
Yes guessing so. I'll reopen if when I update Volk I still see this issue.

closed 2019-10-28 by michaelld