gnuradio/volk #279: volk_32fc_s32fc_x2_rotator_32fc AVX FMA - invalid data at end of buffer

proto17, 2019-08-20:
The `volk_32fc_s32fc_x2_rotator_32fc` method for AVX FMA is producing invalid data at the end of the buffer.  My `~/.volk/volk_config` file has the line `volk_32fc_s32fc_x2_rotator_32fc a_avx u_avx_fma` (generated with `volk_profile`).  If I run with `_u` then I get garbage at the end, but if I use `_a` the output is correct.  As a sanity check I also changed my `volk_config` file to have `volk_32fc_s32fc_x2_rotator_32fc a_avx u_avx` and both `_a` and `_u` worked properly. 

Computer info:
`Intel(R) Core(TM) i9-7960X CPU @ 2.80GHz`
```
pu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc art arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid dca sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault epb cat_l3 cdp_l3 invpcid_single pti mba ibrs ibpb stibp tpr_shadow vnmi flexpriority ept vpid ept_ad fsgsbase tsc_adjust bmi1 hle avx2 smep bmi2 erms invpcid rtm cqm mpx rdt_a avx512f avx512dq rdseed adx smap clflushopt clwb intel_pt avx512cd avx512bw avx512vl xsaveopt xsavec xgetbv1 xsaves cqm_llc cqm_occup_llc cqm_mbm_total cqm_mbm_local dtherm ida arat pln pts
```

VOLK Build Info: I did not enable any special flags.  Just built VOLK with `-DCMAKE_BUILD_TYPE=Debug`.

Compiler info:
```
Using built-in specs.
COLLECT_GCC=g++
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-pc-linux-gnu/9.1.0/lto-wrapper
Target: x86_64-pc-linux-gnu
Configured with: /build/gcc/src/gcc/configure --prefix=/usr --libdir=/usr/lib --libexecdir=/usr/lib --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=https://bugs.archlinux.org/ --enable-languages=c,c++,ada,fortran,go,lto,objc,obj-c++ --enable-shared --enable-threads=posix --with-system-zlib --with-isl --enable-__cxa_atexit --disable-libunwind-exceptions --enable-clocale=gnu --disable-libstdcxx-pch --disable-libssp --enable-gnu-unique-object --enable-linker-build-id --enable-lto --enable-plugin --enable-install-libiberty --with-linker-hash-style=gnu --enable-gnu-indirect-function --enable-multilib --disable-werror --enable-checking=release --enable-default-pie --enable-default-ssp --enable-cet=auto
Thread model: posix
gcc version 9.1.0 (GCC) 
```

See attached plot from Matlab for the invalid samples.
Here is my example code (built with `g++ volk_test.cc -o volk_test -lvolk`):

```c++
#include <vector>
#include <complex>
#include <volk/volk.h>

using sample_t = float;
using complex_t = std::complex<sample_t>;
using complex_vec_t = std::vector<complex_t>;

template <typename T>
void write(const T * const samples, const size_t sample_count, const char * path){
    auto * file_handle = fopen(path, "wb");
    if(! file_handle){
        fprintf(stderr, "Failed to open output file handle\n");
        exit(1);
    }
    fwrite(samples, sizeof(T), sample_count, file_handle);
    fclose(file_handle);
}

template <typename T>
void write(const std::vector<std::complex<T>> & samples, const char * path){
    write(&samples[0], samples.size(), path);
}

int main(){
    const auto sample_rate = 1e6;
    const auto tone_freq = 10e3;
    const auto sample_count = static_cast<uint64_t>(8 * 500);

    // Create the buffer the VOLK way
    auto in_buffer = (complex_t * const)volk_malloc(sizeof(complex_t) * sample_count, volk_get_alignment());
    // Fill the input buffer with 1,0 otherwise the output will have zero amplitude (nothing to rotate)
    std::fill(&in_buffer[0], &in_buffer[sample_count], complex_t(1, 0));

    // Allocate memory for the output buffer
    auto out_buffer = (complex_t * const)volk_malloc(sizeof(complex_t) * sample_count, volk_get_alignment());

    // Calculate the phase increment per sample
    const auto phase_inc = 2 * M_PI * (tone_freq / sample_rate);

    // Doesn't matter what this value is.  Starting it at 1,0 prevents having a single sample that isn't
    // at the expected amplitude of 1.0
    auto starting_phase = complex_t(1, 0);

    // NOTE: Change the 32fc_u to 32fc_a and the output will be correct
    volk_32fc_s32fc_x2_rotator_32fc_u(&out_buffer[0],
                                    &in_buffer[0],
                                    std::polar<sample_t>(1.0, static_cast<sample_t>(phase_inc)),
                                    &starting_phase,
                                    static_cast<uint32_t>(sample_count)
    );

    volk_free(in_buffer);
    volk_free(out_buffer);

    write(in_buffer, sample_count, "/tmp/in_buffer");
    write(out_buffer, sample_count, "/tmp/out_buffer");

    return 0;
}
```

proto17, 2019-08-20:
Also, rather interestingly, the tail at the end is always the same shape.  But, the number of invalid samples changes based on the number of input samples.

argilo, 2019-08-20:
I encountered this problem as well: https://github.com/gnuradio/gnuradio/issues/2748

I already tracked down the bug and opened a pull request: https://github.com/gnuradio/volk/pull/278

proto17, 2019-08-20:
Doh!  I didn't see that in my quick glance through existing bugs.  Thanks!

closed 2019-08-20 by proto17