gnuradio/volk #281: Clarify where to use C and C++

jdemel, 2019-09-14:
This is an issue I stumbled upon for some time. While the VOLK 'infrastructure' does use C++ and Python in some places, the actual library is written in C. I suggest to clarify this structure in the README for new users.

michaelld, 2019-11-14:
+1 from me for someone to do this. Make it so, I say!

jdemel, 2019-12-07:
Where should this be documented? Maybe in the project README?

michaelld, 2019-12-07:
Yes I'd say the top-level project README is the best place. Shouldn't be a long section I'd think!

jdemel, 2020-01-26:
Our current structure looks like this 	
- apps 
     - Python and C++
- cmake 	
     - CMake stuff obviously
- docs
     - Doxygen
- gen
     - Python: Build system
- include/volk
     - Headers to include VOLK in C and C++
     - C++ specific headers for `volk::vector`
- kernels 	
     - C kernels, should only be C but C++ is present as well.
- lib
     - C and C++ for tests. 	
- python
     - Python: volk_modtool
- scripts
- tmpl
     - Templates: Build system
- tools
     - Tools for releases etc.

Basically, I'd suggest we stick with C in kernels. As well as everything that is part of libvolk.so. Everything else are tools that may be C/C++/Python/Bash.
Also, scripts and tools may be merged into one folder.

michaelld, 2020-05-18:
Trying to clean out old stuff & came upon this! Still relevant, I think!

I'd say if you combine scripts and tools as noted, then your list here is good to go. I'd say to add it to the top-level README. It is handy!

jdemel, 2020-05-19:
Yeah. I'll put together a new PR.
@michaelld Are you in favor of merging scripts and tools into on dir?

michaelld, 2020-05-19:
@jdemel yes! they are just related scripts & tools to do stuff that Volk needs but unrelated to the build. 1 directory is easier than 2 here!