gnuradio/volk #290: volk_32f_stddev_and_mean_32f_x2 inaccurate kernel

Ka-zam, 2019-10-02:
When given input with E[X]^2 ~= E[X^2], which is not particularly uncommon, the stddev calculation fails with a -nan result for both
1) volk_32f_stddev_and_mean_32f_x2
2) volk_32f_s32f_stddev_32f

Upon inspection of the code this happens because the following identity is used:
Var(X)  = E[(X−mu)^2] = E[X^2]−E[X]^2

which according to [https://dl.acm.org/citation.cfm?id=3221269.3223036](url) should never be used in practice as it suffers from catastrophic cancellation.

Test run where naive_f is the two-pass approach of first calculating the mean and then estimating the variance give the following output
~~~~
in[0]: 1.000000000
in[1]: 0.999998987
in[2]: 1.000002027
in[3]: 0.999997020
in[4]: 1.000004053
in[5]: 0.999994993
in[6]: 1.000005960
in[7]: 0.999993026
in[8]: 1.000007987
in[9]: 0.999991000

naive_f
mu:
0.999999523
omega:
0.000005311

volk_std_and_mean
mu:
0.999999523
omega:
-nan
~~~~

Also of note might be the fact that we should probably compute the sample variance and not the population variance. 
[https://en.wikipedia.org/wiki/Variance#Sample_variance](url)

Ka-zam, 2019-12-19:
I'm implementing Welford's algorithm now which is the recommended way of calculating these values.

jdemel, 2019-12-19:
Do you refer to [Welford's algorithm here](https://en.wikipedia.org/wiki/Algorithms_for_calculating_variance)?

Ka-zam, 2019-12-19:
Yes. I implemented it in generic C and it works fine without the accuracy problems of the so-called textbook approach that's used now. 

So far so good but it's decidedly not easy to make a SIMD implementation out of this.

Ka-zam, 2019-12-20:
Progress. I made a divide and conquer approach that lends itself well to SIMD implementation.

michaelld, 2019-12-20:
Looking forward to seeing this code in a PR! Thx!

jdemel, 2020-01-14:
This issue should be fixed with #322 .

michaelld, 2020-01-24:
#332 is merged. Closing this issue as fixed. If it's not, please reopen and explain what's going on so it can be addressed.

jdemel, 2020-01-25:
@michaelld I think you mixed up #332 and #322 . 
#322 fixes this issue and we are waiting for a CLA. I just reopen this issue.

jdemel, 2021-04-13:
We just merged #457 which is a rebased version of #322 . Thus, this issue should be fixed and can be closed. Thanks @Ka-zam !

closed 2021-04-13 by jdemel