gnuradio/volk #292: RUN_VOLK_TESTS: volk_8u_x3_encodepolarpuppet_8u segmentation fault

cbr600, 2019-10-24:
macOS 10.14.6 Mac mini (2018) 3.2 GHz Intel Core i7 after running volk_profile. The profile process always results in segmentation fault (last portion):

RUN_VOLK_TESTS: volk_8u_x3_encodepolarpuppet_8u(131071,1987)
generic completed in 912.276 ms
u_ssse3 completed in 233.32 ms
Segmentation fault: 11

michaelld, 2019-10-24:
I see the same on my 10.14.6 2019 MBP 8-core i9. Haven't looked into it further.

hcab14, 2019-10-25:
see https://github.com/gnuradio/volk/issues/229

hcab14, 2019-10-25:
.. and the PR https://github.com/gnuradio/volk/pull/288

michaelld, 2019-10-27:
Guessing Closed by #288 ... so closing this one. Please reopen if it isn't fixed.

closed 2019-10-27 by michaelld

cbr600, 2019-10-27:
Yup! that fixed it. I do have to run it from the build directory since the installation location is not the same as the Mac is using for python packages. But I know that is another story...

Thanks for the fast fix!

michaelld, 2019-10-28:
@cbr600 YW! Glad the timing worked out so well!