gnuradio/volk #310: VOLK Error when Running GNURADIO

mwk088, 2019-11-20:
VOLK: Error allocating memory (posix_memalign: error 12: Cannot allocate memory)
terminate called after throwing an instance of 'std::runtime_error'
what():  bad alloc for d_vp->decisions!

OS:  Ubuntu 18.04
GNU:  main-3.8
Volk:  2.0
Used malloc implementation: posix_memalign
Python:  3.6

I am watching htop as my GNU flowgraph runs, and see the allocated memory slowly creep up until it hits 106Gb of the available 756Gb, and then I get the error.

Any ideas as to what is going on?

Thank you
Mark

jdemel, 2019-11-20:
What exactly do you run? I assume a GNU Radio flowgraph.
This sounds like some kind of memory leak. Though, I've seen such behavior because I forgot to disable a 'Message Debug' block and it stored messages for half a day.
We won't be able to make solve this issue without more info on how to reproduce this behavior.

michaelld, 2019-11-20:
I'm guessing it's some block that uses Volk and isn't deallocating correctly. A stack traceback from the allocation error would be really useful!

mwk088, 2019-11-21:
I am running a GNU flowgraph with a USRP collecting IQ data.

How do I produce a stack traceback from the allocation error?  I will provide the output once I am able to get the traceback.

mwk088, 2019-11-21:
I ran my application with gdb and only received the output below, which is much help to me.  I will see if I can gather more information.

Thread 117 "python3" received signal SIGABRT, Aborted.
[Switching to Thread 0x7ff7ebfff700 (LWP 599)]
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
51	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.

jdemel, 2019-11-21:
So the error pops up in `libwifi_ac-decoder.so`. I'd recommend to recompile your module with debug flags. Maybe that'll give you more info on where you receive this error. So far, I'd guess the bug is somewhere in `wifi_ac`. If this is the name of your module.

mwk088, 2019-11-21:
Thank you.  I will look into this right away, and that is the name of one of my modules.

michaelld, 2019-11-21:
start Python running in GDB. the continue execution running the FG Python script.

michaelld, 2019-11-21:
there's probably a way to do those 2 together ... I've never figured it out & so I just do it in 2 steps.

michaelld, 2019-11-21:
Sorry about those 2 comments ... somehow GitHub didn't update this page & I didn't see anything above ... now that I see it, I see y'all have handled this already ... :+1:

mwk088, 2019-11-22:
I reran my flowgraph with the modules (I think) built with debug symbols (using cmake).  I admit I am not very good at interpreting the output, so I don't see much difference from what I pasted previously.  Any insight or help would be greatly appreciated.  Thank you
```
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x00007ffff7a24801 in __GI_abort () at abort.c:79
#2  0x00007ffff4a37957 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007ffff4a3dab6 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007ffff4a3daf1 in std::terminate() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007ffff4a3dd24 in __cxa_throw () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007fff6d26b102 in gr::fec::code::cc_decoder_impl::cc_decoder_impl (this=0x7fce5fd09e10, frame_size=4000000, k=7, rate=2, polys=std::vector of length 2, capacity 2 = {...}, start_state=0, 
    end_state=-1, mode=CC_TERMINATED, padded=false, __in_chrg=<optimized out>, __vtt_parm=<optimized out>) at /opt/truearrow/6.3/src/gnuradio/gr-fec/lib/cc_decoder_impl.cc:133
#7  0x00007fff6d269d76 in gr::fec::code::cc_decoder::make (frame_size=4000000, k=7, rate=2, polys=std::vector of length 2, capacity 2 = {...}, start_state=0, end_state=-1, mode=CC_TERMINATED, 
    padded=false) at /opt/truearrow/6.3/src/gnuradio/gr-fec/lib/cc_decoder_impl.cc:49
#8  0x00007fff6d272eab in Framework_Decoder::Framework_Decoder (this=0x7ff8467fb540) at /workspaces/ta/wifi-ac/lib/Shim/Framework_Decoder.cc:26
#9  0x00007fff6d71f9aa in CPhy_Decode_Ofdm_impl::CPhy_Decode_Ofdm_impl (this=0x7ff8467fb540) at /workspaces/ta/wifi-ac/lib/Modem/Phy/Ofdm/CPhy_Decode_Ofdm_impl.cc:22
#10 0x00007fff6d7ba46c in CState_Data_Ofdm_impl::Process (this=0x7ff8467fb958, nInputSamplesPerNrx_Fs=580, ppInputSamplesAllNrx_Fs=0x7ff8467fb9d0, pcPsduDataBytes=0x7ff8467fba40, 
    pPsduDataBytes=0x7d2a6de84920 "", pAmpduInfo=std::vector of length 0, capacity 0, pBurstStateParams=0x7ffbb1ac5010) at /workspaces/ta/wifi-ac/lib/Modem/State/Ofdm/CState_Data_Ofdm_impl.cc:115
#11 0x00007fff6d7c4ca8 in CStateHandler_Ht_Train_Data_Ofdm_impl::Process (this=0x16c3360, nStartingSamplePerNrx_Fs=6855759267, nInputSamplesPerNrx_Fs=580, ppInputSamplesAllNrx_Fs=0x7ff8467fb9d0, 
    pcPsduDataBytes=0x7ff8467fba40, pPsduDataBytes=0x7d2a6de84920 "", pAmpduInfo=std::vector of length 0, capacity 0, pBurstStateParams=0x7ffbb1ac5010)
    at /workspaces/ta/wifi-ac/lib/Modem/State/Ofdm/CStateHandler_Ht_Train_Data_Ofdm_impl.cc:201
#12 0x00007fff6dca98e9 in gr::wifi_ac::CGRHandler_Ofdm_Ht_Train_Data_impl::handler (this=0x16c1520, msg=..., buf=std::vector of length 2, capacity 2 = {...})
    at /workspaces/ta/wifi-ac/lib/GnuRadio/CGRHandler_Ofdm_Ht_Train_Data_impl.cc:358
#13 0x00007fff6e41ed39 in es_handler::handler_helper (this=0x16c1528, msg=...) at /workspaces/ta/gr-eventstream/lib/es_handler.cc:47
#14 0x00007fff6e4128fc in es_eh_pair::run (this=0x7ff7c4072fc0) at /workspaces/ta/gr-eventstream/lib/es_eh_pair.cc:39
#15 0x00007fff6e4153d7 in es_event_loop_thread::do_work (this=0x7ff8700049f0) at /workspaces/ta/gr-eventstream/lib/es_event_loop_thread.cc:91
#16 0x00007fff6e41ba41 in boost::_mfi::mf0<void, es_event_loop_thread>::operator() (this=0x7ff870004c28, p=0x7ff8700049f0) at /usr/include/boost/bind/mem_fn_template.hpp:49
#17 0x00007fff6e41b9a3 in boost::_bi::list1<boost::_bi::value<es_event_loop_thread*> >::operator()<boost::_mfi::mf0<void, es_event_loop_thread>, boost::_bi::list0> (this=0x7ff870004c38, f=..., a=...)
    at /usr/include/boost/bind/bind.hpp:259
#18 0x00007fff6e41b6e6 in boost::_bi::bind_t<void, boost::_mfi::mf0<void, es_event_loop_thread>, boost::_bi::list1<boost::_bi::value<es_event_loop_thread*> > >::operator() (this=0x7ff870004c28)
    at /usr/include/boost/bind/bind.hpp:1294
#19 0x00007fff6e41b36e in boost::detail::thread_data<boost::_bi::bind_t<void, boost::_mfi::mf0<void, es_event_loop_thread>, boost::_bi::list1<boost::_bi::value<es_event_loop_thread*> > > >::run (
    this=0x7ff870004a70) at /usr/include/boost/thread/detail/thread.hpp:116
#20 0x00007ffff40d0bcd in ?? () from /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.65.1
#21 0x00007ffff77cc6db in start_thread (arg=0x7ff8467fc700) at pthread_create.c:463
#22 0x00007ffff7b0588f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95

(gdb) info threads
43   Thread 0x7ff8b4c8c700 (LWP 285) "python3" 0x00007ffff7af8bf9 in __GI___poll (fds=0x7ff8b4c8bbb0, nfds=2, timeout=-1) at ../sysdeps/unix/sysv/linux/poll.c:29
  44   Thread 0x7ff8affff700 (LWP 286) "python3" 0x00007ffff77d29f3 in futex_wait_cancelable (private=<optimized out>, expected=0, futex_word=0x7fff727a4aa8 <log_rs()::log_rs+584>)
    at ../sysdeps/unix/sysv/linux/futex-internal.h:88
  45   Thread 0x7ff8af7fe700 (LWP 287) "python3" 0x00007ffff77d29f3 in futex_wait_cancelable (private=<optimized out>, expected=0, futex_word=0x7fff727a4964 <log_rs()::log_rs+260>)
    at ../sysdeps/unix/sysv/linux/futex-internal.h:88
  46   Thread 0x7ff8aeffd700 (LWP 288) "python3" 0x00007ffff7ac89d0 in __GI___nanosleep (requested_time=0x7ff8aeffcc40, remaining=0x7ff8aeffcc40) at ../sysdeps/unix/sysv/linux/nanosleep.c:28
  47   Thread 0x7ff8ae7fc700 (LWP 289) "python3" 0x00007ffff7af8bf9 in __GI___poll (fds=0x7ff8ae7fb7e0, nfds=1, timeout=100) at ../sysdeps/unix/sysv/linux/poll.c:29
[... cut ...]
  90   Thread 0x7ff83efed700 (LWP 332) "python3" 0x00007fff6dca9889 in gr::wifi_ac::CGRHandler_Ofdm_Ht_Train_Data_impl::handler (this=0x16c1520, msg=..., 
    buf=std::vector of length 2, capacity 2 = {...}) at /workspaces/ta/wifi-ac/lib/GnuRadio/CGRHandler_Ofdm_Ht_Train_Data_impl.cc:355
  91   Thread 0x7ff83e7ec700 (LWP 333) "python3" 0x00007fff6d726b4e in CPhy_Demap_Ofdm_impl::Phy_Demap_Ofdm_ac (this=0x7ff83e7eb718, pBurstStateParams=0x7fff1f2b9010)
    at /workspaces/ta/wifi-ac/lib/Modem/Phy/Ofdm/CPhy_Demap_Ofdm_impl.cc:200
```

jdemel, 2019-11-22:
@mwk088 Your program terminates because it can't construct an object from class `cc_decoder_impl`. It fails in the constructor at line 133. This is intended because `volk_malloc` could not allocate a very large array. You try to instantiate a Decoder with a frame length of 4,000,000. This is really large. Then the decoder tries to allocate an array of size `4e6 * 64 = 256,000,000` and fails. i.e. an array that's a quarter gigabyte large. Your OS does probably not allow you to allocate such large chunks.
Besides a CC code with that frame size is unrealistic. Since your module is called 'wifi_ac' I assume you try to implement something according to that standard. I'm pretty sure the maximum frame size is way lower than what you're trying to use.

mwk088, 2019-11-22:
Thank you for the feedback.  I would like to add a bit more background to my issue.

The wifi_ac module, which is a large part of the flowgraph, was originally developed with CentOS 7.5/ubuntu 1804, GNU v3.7.13.4, Volk 1.4, and python 2.7.  During the development and testing, no memory leak like this was detected.  I am now attempting to upgrade to GNU v3.8, Volk 2.0 and python 3.6, and just now I am seeing this memory leak/allocation error.  Could any of the upgrades be contributing to the memory leak problem or is it embedded in the wifi_ac module?

Thank you.

jdemel, 2019-11-22:
From `gr-fec/lib/cc_decoder_impl.cc:130`
```
    d_vp->decisions = (unsigned char*)volk_malloc(
        sizeof(unsigned char) * d_veclen * d_decision_t_size, volk_get_alignment());
    if (d_vp->decisions == NULL) {
        throw std::runtime_error("bad alloc for d_vp->decisions!\n");
    }
```
This is not a memory leak but you can't allocate this much memory. This might be a configuration issue or something completely different. In any case, I'd recommend to use either a shorter code or another code entirely since this is not standard compliant. I'd say this is an issue with your machine or your `wifi_ac` module.

mwk088, 2019-11-22:
I still don't get it.  I changed the max frame size orders of magnitude smaller, but I still see the memory being eaten up, just much more slowly, which will eventually lead to the volk memory error.

jdemel, 2019-11-24:
You need to make sure, all the memory you allocate is deallocated properly as well. Since you're in C++ land, you could use `volk::vector` with latest VOLK.

jdemel, 2019-12-07:
I suggest to close this issue. Apparently, the bug is not caused by VOLK. @michaelld what do you think?

michaelld, 2019-12-07:
I'm in agreement @jdemel : this looks like a usage-of-Volk issue, not a issue with Volk itself.

@mwk088 If you feel otherwise, please reopen and keep the discussion going with more debug logs -- or, even better: an example C / C++ code that shows the issue itself!

closed 2019-12-07 by michaelld