gnuradio/volk #312: use of undeclared identifier 'volk_32fc_x2_s32fc_multiply_conjugate_add_32fc_get_func_desc'

ast, 2019-12-08:
```
git clone https://github.com/gnuradio/volk.git volk2
cd volk2 && mkdir build && cmake ..
make
```

```
In file included from /Users/albin/src/volk2/lib/testqa.cc:32:
/Users/albin/src/volk2/lib/kernel_tests.h:144:8: error: use of undeclared identifier 'volk_32fc_x2_s32fc_multiply_conjugate_add_32fc_get_func_desc'; did you
      mean 'volk_8ic_x2_s32f_multiply_conjugate_32fc_get_func_desc'?
    QA(VOLK_INIT_TEST(volk_32fc_x2_s32fc_multiply_conjugate_add_32fc, test_params))
       ^
/Users/albin/src/volk2/lib/kernel_tests.h:16:22: note: expanded from macro 'VOLK_INIT_TEST'
    volk_test_case_t(func##_get_func_desc(), (void(*)())func##_manual, std::string(#func),\
                     ^
<scratch space>:10:1: note: expanded from here
volk_32fc_x2_s32fc_multiply_conjugate_add_32fc_get_func_desc
^
/usr/local/include/volk/volk.h:1952:34: note: 'volk_8ic_x2_s32f_multiply_conjugate_32fc_get_func_desc' declared here
extern VOLK_API volk_func_desc_t volk_8ic_x2_s32f_multiply_conjugate_32fc_get_func_desc(void);
                                 ^
In file included from /Users/albin/src/volk2/lib/testqa.cc:32:
/Users/albin/src/volk2/lib/kernel_tests.h:144:8: error: use of undeclared identifier 'volk_32fc_x2_s32fc_multiply_conjugate_add_32fc_manual'
    QA(VOLK_INIT_TEST(volk_32fc_x2_s32fc_multiply_conjugate_add_32fc, test_params))
       ^
/Users/albin/src/volk2/lib/kernel_tests.h:16:57: note: expanded from macro 'VOLK_INIT_TEST'
    volk_test_case_t(func##_get_func_desc(), (void(*)())func##_manual, std::string(#func),\
                                                        ^
<scratch space>:11:1: note: expanded from here
volk_32fc_x2_s32fc_multiply_conjugate_add_32fc_manual
^
2 errors generated.
make[2]: *** [lib/CMakeFiles/volk_test_all.dir/testqa.cc.o] Error 1
make[1]: *** [lib/CMakeFiles/volk_test_all.dir/all] Error 2
make: *** [all] Error 2
```

michaelld, 2019-12-08:
Yes indeed ... this is what happens when (1) Volk is already installed in the $CMAKE_INSTALL_PREFIX and/or a standard system search path prefix; and (2) the Volk API changes. In other words, this is a duplicate of https://github.com/gnuradio/volk/issues/234 .

michaelld, 2019-12-08:
Note the inclusion of `/usr/local/include/volk/volk.h`.

The simple solution for most folks is to uninstall the old Volk before trying to build the new Volk.

michaelld, 2019-12-08:
I'm closing this issue in favor of the other one. I'll add some notes from recent efforts there.

closed 2019-12-08 by michaelld