gnuradio/volk #316: Patch release to pick up AVX FMA rotator fix

vsergeev, 2019-12-12:
I use VOLK as a plugin for a number of accelerations in LuaRadio, including the `volk_32fc_s32fc_x2_rotator_32fc` kernel for complex mixing. I'm about to make a new release, but had issues with unit tests passing with v2.0.0, as this version still has the AVX FMA rotator bug that was fixed in f04a46f18308771b5ebf46292bf0fe6918578b35 and #278. I'm not sure if VOLK is presently or soon in a good state for a release, but a patch release to pick up this fix would be awesome for downstream library consumers.

jdemel, 2019-12-12:
@vsergeev thanks for your feedback. We'll have a new release soon.

jdemel, 2019-12-22:
We just released v2.1.0 where this bug is fixed. I suppose this fixes this issue. Thus, I close this issue.

closed 2019-12-22 by jdemel

vsergeev, 2019-12-22:
Thanks! This version passes all my tests.