gnuradio/volk #317: Kernel for folding vector label:Enhancement

ast, 2019-12-12:
There doesn't seem to be a kernel for folding a vector by a divisor. This is essential for decimation in the frequency domain.

```matlab
% Example, Matlab notation
Fx_alias = (Fx(1:256) + Fx(257:512) + Fx(513:768) + Fx(769:1024)) / 4;
```

jdemel, 2019-12-12:
Could you rewrite your example in Python?

Do you suggest a kernel that looks smth like this 
```
void volk_32fc_fold_32fc(result, src, integer_fold, num_points)
```

Essentially a combination of `volk_32f_x2_add_32f` and `volk_32f_s32f_multiply_32f`. Complex values are handled by reinterpreting complex pointers as float pointers with `2 * num_points`.

ast, 2019-12-12:
```
void fold(const float complex* in, float complex *out, const float scale, int fold, int len) {
    const int N = len/fold;
    for (int offset = 0; offset < len; offset+=N) {
        volk_32fc_x2_add_32fc(out, out, &in[offset], N);
    }
    volk_32f_s32f_multiply_32f((float*)out, (float*)out, scale, 2*N);
}
```

jdemel, 2019-12-13:
So the function signature should look like this?

```
void volk_32fc_fold_32fc(float complex* out, const float complex *in, 
                         const float scale, const int fold, 
                         const unsigned int num_points)
```
1. Should we fix to `const float scale = 1. / fold;`?
2. Is `fold` a good name for this kernel?

ast, 2019-12-13:
Everything is up for discussion really, maybe this kernel doesn't even make sense? But this is a common operation when you decimate in the frequency domain if you want it to be equivalent to decimation in the time domain.

michaelld, 2019-12-13:
"fold" seems to be more general than this suggested kernel -- which I like the idea of, BTW. MATLAB "fold" recursively applies a function to a set of data, like what "reduce" in Octave does. Hence I'd suggest we use some other name ... it can include "fold", but it should be more precise.