gnuradio/volk #323: Build is broken for armv7 architecture

drmpeg, 2020-01-03:
I've dissected this problem to commit https://github.com/gnuradio/volk/commit/4589103155a27f9e0805232fcb43d1b0a067c013

It seems that the new policies in cmake 3.8 breaks something.

Good cmake:
```
-- Performing Test neon_compile_result
-- Performing Test neon_compile_result - Success
-- Performing Test have_neonv7_result
-- Performing Test have_neonv7_result - Success
-- Performing Test have_neonv8_result
-- Performing Test have_neonv8_result - Failed
-- CPU is armv7, Overruled arch neonv8
-- Available architectures: generic;hardfp;neon;neonv7;orc;norc
-- Available machines: generic_orc;neon_orc;neonv7_hardfp_orc
-- BUILD TYPE = RELEASE
```

Bad cmake:
```
-- Performing Test neon_compile_result
-- Performing Test neon_compile_result - Success
-- Performing Test have_neonv7_result
-- Performing Test have_neonv7_result - Failed
-- Performing Test have_neonv8_result
-- Performing Test have_neonv8_result - Failed
-- Available architectures: generic;hardfp;neon;neonv7;neonv8;orc;norc
-- Available machines: generic_orc;neon_orc;neonv7_hardfp_orc;neonv8
-- BUILD TYPE = RELEASE
```

The build fails when volk_machine_neonv8.c is attempted to be compiled.

A workaround for this is to change line 21 in the top level CMakeLists.txt from:
cmake_minimum_required(VERSION 3.8)
to:
cmake_minimum_required(VERSION 3.7)

michaelld, 2020-01-03:
Thanks for the report & that's very interesting! I believe we're already using a CMake 3.8 feature or 2, so reverting to 3.7 probably isn't wise. I wonder what policy changed that's impacting the compile of that file?! Do you have access to the build log to show the difference? I really need to get an ARM build going locally ...

drmpeg, 2020-01-03:
Digging a little deeper, here's the web page that shows CMake policy changes by release.

https://cmake.org/cmake/help/v3.8/manual/cmake-policies.7.html

Policy CMP0067 (Honer language standard in try_compile() source-file signature) was added for version 3.8. If I add the lines:
```
if(POLICY CMP0067)
    cmake_policy(SET CMP0067 OLD)
endif()
```
to the top level CMakelists.txt, it works fine.

ARM architectures are detected by compiling snippets of code with try_compile(), so this makes total sense. I'll see if I can fix the specific try_compile() error, which would be the best solution.

drmpeg, 2020-01-03:
The check_c_source_compiles() can be fixed by using __asm__ instead of asm. This should probably be tested with gcc, clang and MSVC.

https://github.com/gnuradio/volk/blob/7a60cd66d74b0c5206e2e1cb9ea5198e2153af83/lib/CMakeLists.txt#L257-L261

drmpeg, 2020-01-03:
There's already a pull request for this. https://github.com/gnuradio/volk/pull/319

closed 2020-01-13 by jdemel