gnuradio/volk #327: Code cleanup

ValZapod, 2020-01-13:
@michaelld https://github.com/gnuradio/volk/blob/001e60a7f33cdc5430892201af88a332b67f5872/lib/qa_utils.cc#L624
The 'A = !fail' expression is equivalent to the 'A = false' expression. (We are in if (fail); )
https://github.com/gnuradio/volk/blob/001e60a7f33cdc5430892201af88a332b67f5872/lib/qa_utils.cc#L144
Throw what?
https://github.com/gnuradio/volk/blob/001e60a7f33cdc5430892201af88a332b67f5872/lib/volk_prefs.c#L81
No check for null. And consider assigning realloc() to a temporary pointer.

https://github.com/gnuradio/volk/blob/001e60a7f33cdc5430892201af88a332b67f5872/tmpl/volk_cpu.tmpl.c#L114
Why it is equivalent to get_avx_enabled? Should not the check be different?

https://github.com/gnuradio/volk/blob/001e60a7f33cdc5430892201af88a332b67f5872/apps/volk_profile.cc#L199-L201
Change that to `std::size_t found = config_str.find(' ');`

https://github.com/gnuradio/volk/blob/001e60a7f33cdc5430892201af88a332b67f5872/apps/volk_profile.cc#L169
return 0;

```
./volk/kernels/volk/volk_32f_x2_pow_32f.h	348	note	V624 The constant 0.6931471805 is being utilized. The resulting value could be inaccurate. Consider using the M_LN2 constant from <math.h>.
./volk/kernels/volk/volk_32f_x2_pow_32f.h	496	note	V624 The constant 0.6931471805 is being utilized. The resulting value could be inaccurate. Consider using the M_LN2 constant from <math.h>.
./volk/kernels/volk/volk_32f_x2_pow_32f.h	111	note	V624 The constant 0.6931471805 is being utilized. The resulting value could be inaccurate. Consider using the M_LN2 constant from <math.h>.
./volk/kernels/volk/volk_32f_x2_pow_32f.h	229	note	V624 The constant 0.6931471805 is being utilized. The resulting value could be inaccurate. Consider using the M_LN2 constant from <math.h>.
./volk/kernels/volk/volk_32f_x2_pow_32f.h	614	note	V624 The constant 0.6931471805 is being utilized. The resulting value could be inaccurate. Consider using the M_LN2 constant from <math.h>.
./volk/kernels/volk/volk_32f_x2_pow_32f.h	733	note	V624 The constant 0.6931471805 is being utilized. The resulting value could be inaccurate. Consider using the M_LN2 constant from <math.h>.
./volk/kernels/volk/volk_8u_conv_k7_r2puppet_8u.h	304	err	V547 Expression 'd_polys[i] < 0' is always false.
./volk/kernels/volk/volk_8u_conv_k7_r2puppet_8u.h	151	err	V547 Expression 'd_polys[i] < 0' is always false.
./volk/kernels/volk/volk_8u_conv_k7_r2puppet_8u.h	227	err	V547 Expression 'd_polys[i] < 0' is always false.

```

https://github.com/gnuradio/volk/blob/001e60a7f33cdc5430892201af88a332b67f5872/kernels/volk/volk_16i_max_star_horizontal_16i.h#L131
Can we change it to if(intermediate) or even if((num_bytes >> 4) & 1) And there is a crazy amount of those:
```
./volk/kernels/volk/volk_32fc_x2_conjugate_dot_prod_32fc.h	95	err	V1008 Consider inspecting the 'for' operator. No more than one iteration of the loop will be performed.
./volk/kernels/volk/volk_32fc_x2_conjugate_dot_prod_32fc.h	421	err	V1008 Consider inspecting the 'for' operator. No more than one iteration of the loop will be performed.
./volk/kernels/volk/volk_32fc_x2_dot_prod_32fc.h	96	err	V1008 Consider inspecting the 'for' operator. No more than one iteration of the loop will be performed.
./volk/kernels/volk/volk_32fc_x2_dot_prod_32fc.h	531	err	V1008 Consider inspecting the 'for' operator. No more than one iteration of the loop will be performed.
./volk/kernels/volk/volk_16ic_convert_32fc.h	118	err	V1008 Consider inspecting the 'for' operator. No more than one iteration of the loop will be performed.
./volk/kernels/volk/volk_16ic_convert_32fc.h	251	err	V1008 Consider inspecting the 'for' operator. No more than one iteration of the loop will be performed.
./volk/kernels/volk/volk_16i_branch_4_state_8.h	101	note	V1008 Consider inspecting the 'for' operator. No more than one iteration of the loop will be performed.
./volk/kernels/volk/volk_16i_max_star_horizontal_16i.h	131	err	V1008 Consider inspecting the 'for' operator. No more than one iteration of the loop will be performed.
./volk/kernels/volk/volk_32fc_index_max_16u.h	314	err	V1008 Consider inspecting the 'for' operator. No more than one iteration of the loop will be performed.
./volk/kernels/volk/volk_32fc_index_max_16u.h	342	err	V1008 Consider inspecting the 'for' operator. No more than one iteration of the loop will be performed.
./volk/kernels/volk/volk_32fc_index_max_32u.h	272	err	V1008 Consider inspecting the 'for' operator. No more than one iteration of the loop will be performed.
./volk/kernels/volk/volk_32fc_index_max_32u.h	300	err	V1008 Consider inspecting the 'for' operator. No more than one iteration of the loop will be performed.
./volk/kernels/volk/volk_32fc_x2_s32f_square_dist_scalar_mult_32f.h	314	err	V1008 Consider inspecting the 'for' operator. No more than one iteration of the loop will be performed.
./volk/kernels/volk/volk_32fc_x2_s32f_square_dist_scalar_mult_32f.h	332	err	V1008 Consider inspecting the 'for' operator. No more than one iteration of the loop will be performed.
./volk/kernels/volk/volk_32fc_x2_square_dist_32f.h	235	err	V1008 Consider inspecting the 'for' operator. No more than one iteration of the loop will be performed.
./volk/kernels/volk/volk_32fc_x2_square_dist_32f.h	252	err	V1008 Consider inspecting the 'for' operator. No more than one iteration of the loop will be performed.
./volk/kernels/volk/volk_8u_x4_conv_k7_r2_8u.h	591	err	V1008 Consider inspecting the 'for' operator. No more than one iteration of the loop will be performed.
./volk/kernels/volk/volk_32fc_s32fc_x2_rotator_32fc.h	302	err	V1008 Consider inspecting the 'for' operator. No more than one iteration of the loop will be performed.
./volk/kernels/volk/volk_32fc_s32fc_x2_rotator_32fc.h	407	err	V1008 Consider inspecting the 'for' operator. No more than one iteration of the loop will be performed.
./volk/kernels/volk/volk_32fc_index_max_16u.h	147	err	V1008 Consider inspecting the 'for' operator. No more than one iteration of the loop will be performed.
./volk/kernels/volk/volk_32fc_index_max_16u.h	172	err	V1008 Consider inspecting the 'for' operator. No more than one iteration of the loop will be performed.
./volk/kernels/volk/volk_32fc_index_max_16u.h	502	err	V1008 Consider inspecting the 'for' operator. No more than one iteration of the loop will be performed.
./volk/kernels/volk/volk_32fc_index_max_16u.h	527	err	V1008 Consider inspecting the 'for' operator. No more than one iteration of the loop will be performed.
./volk/kernels/volk/volk_32fc_index_max_32u.h	136	err	V1008 Consider inspecting the 'for' operator. No more than one iteration of the loop will be performed.
./volk/kernels/volk/volk_32fc_index_max_32u.h	160	err	V1008 Consider inspecting the 'for' operator. No more than one iteration of the loop will be performed.
./volk/kernels/volk/volk_32fc_index_max_32u.h	453	err	V1008 Consider inspecting the 'for' operator. No more than one iteration of the loop will be performed.
./volk/kernels/volk/volk_32fc_index_max_32u.h	477	err	V1008 Consider inspecting the 'for' operator. No more than one iteration of the loop will be performed.
./volk/kernels/volk/volk_32fc_x2_s32f_square_dist_scalar_mult_32f.h	141	err	V1008 Consider inspecting the 'for' operator. No more than one iteration of the loop will be performed.
./volk/kernels/volk/volk_32fc_x2_s32f_square_dist_scalar_mult_32f.h	161	err	V1008 Consider inspecting the 'for' operator. No more than one iteration of the loop will be performed.
./volk/kernels/volk/volk_32fc_x2_s32f_square_dist_scalar_mult_32f.h	179	err	V1008 Consider inspecting the 'for' operator. No more than one iteration of the loop will be performed.
./volk/kernels/volk/volk_32fc_x2_s32f_square_dist_scalar_mult_32f.h	427	err	V1008 Consider inspecting the 'for' operator. No more than one iteration of the loop will be performed.
./volk/kernels/volk/volk_32fc_x2_square_dist_32f.h	132	err	V1008 Consider inspecting the 'for' operator. No more than one iteration of the loop will be performed.
./volk/kernels/volk/volk_32fc_x2_square_dist_32f.h	151	err	V1008 Consider inspecting the 'for' operator. No more than one iteration of the loop will be performed.
./volk/kernels/volk/volk_32fc_x2_square_dist_32f.h	167	err	V1008 Consider inspecting the 'for' operator. No more than one iteration of the loop will be performed.
./volk/kernels/volk/volk_32fc_x2_square_dist_32f.h	376	err	V1008 Consider inspecting the 'for' operator. No more than one iteration of the loop will be performed.
[... cut ...]
./volk/kernels/volk/volk_32f_asin_32f.h	363	note	V537 Consider reviewing the correctness of 'z' item's usage.
./volk/kernels/volk/volk_32f_atan_32f.h	169	note	V537 Consider reviewing the correctness of 'z' item's usage.
./volk/kernels/volk/volk_32f_atan_32f.h	358	note	V537 Consider reviewing the correctness of 'z' item's usage.```

Do we need this https://github.com/gnuradio/volk/blob/001e60a7f33cdc5430892201af88a332b67f5872/kernels/volk/volk_32f_x2_pow_32f.h#L777 (Also lines 400, 548, 162, 280, 666.)

michaelld, 2020-01-13:
https://github.com/gnuradio/volk/blob/001e60a7f33cdc5430892201af88a332b67f5872/lib/qa_utils.cc#L624
> 
> The 'A = !fail' expression is equivalent to the 'A = false' expression. (We are in if (fail); )

Yup. Good catch. Should just be `result->pass = false;`

> https://github.com/gnuradio/volk/blob/001e60a7f33cdc5430892201af88a332b67f5872/lib/qa_utils.cc#L144
> 
> Throw what?

Indeed! It's a "runtime type error" or something like that.

> https://github.com/gnuradio/volk/blob/001e60a7f33cdc5430892201af88a332b67f5872/lib/volk_prefs.c#L81
> 
> No check for null. And consider assigning realloc() to a temporary pointer.

These seems like reasonable suggestions. Something like:
```
...
void* new_prefs = realloc(prefs, (n_arch_prefs+1) * sizeof(*prefs));
if (!new_prefs) {
  printf ("volk_load_preferences: bad malloc\n");
  break;
}
prefs = (volk_arch_pref_t *) new_prefs;
...
```

> https://github.com/gnuradio/volk/blob/001e60a7f33cdc5430892201af88a332b67f5872/tmpl/volk_cpu.tmpl.c#L114
> 
> Why it is equivalent to get_avx_enabled? Should not the check be different?

Good question! We should look up the difference in determining AVX versus AVX2 & fix this if/as necessary. Keeping these separate seems like the right thing to do, since there's probably something we need to fix here for determining the proper support.

> https://github.com/gnuradio/volk/blob/001e60a7f33cdc5430892201af88a332b67f5872/apps/volk_profile.cc#L199-L201
> 
> Change that to `std::size_t found = config_str.find(' ');`

Yes!

> https://github.com/gnuradio/volk/blob/001e60a7f33cdc5430892201af88a332b67f5872/apps/volk_profile.cc#L169
> 
> return 0;

Yes!

> ```
> ./volk/kernels/volk/volk_32f_x2_pow_32f.h	348	note	V624 The constant 0.6931471805 is being utilized. The resulting value could be inaccurate. Consider using the M_LN2 constant from <math.h>. [...]

Yes: Constants should be set to some globally provided value.

> ./volk/kernels/volk/volk_8u_conv_k7_r2puppet_8u.h	304	err	V547 Expression 'd_polys[i] < 0' is always false.
> ./volk/kernels/volk/volk_8u_conv_k7_r2puppet_8u.h	151	err	V547 Expression 'd_polys[i] < 0' [...]

Indeed! Guessing someone pulled this code from elsewhere & didn't bother fixing it up. Since `d_polys` is a constant, and both entries are always positive for the values, I think this code would be just fine as:
```
        Branchtab[i*d_numstates/2+state] = parity((2*state) & d_polys[i], Partab) ? 255 : 0;
```

> https://github.com/gnuradio/volk/blob/001e60a7f33cdc5430892201af88a332b67f5872/kernels/volk/volk_16i_max_star_horizontal_16i.h#L131
> 
> Can we change it to if(intermediate) or even if((num_bytes >> 4) & 1) And there is a crazy amount of those:

Yes! I think moving to `if` is the way to go.

> https://github.com/gnuradio/volk/blob/001e60a7f33cdc5430892201af88a332b67f5872/kernels/volk/volk_32f_acos_32f.h#L175
> 
> Why do we need this line? The same on 374, and

Seems like that line is unnecessary. Let's get it removed & see if the kernel still works.
> 
> Do we need this https://github.com/gnuradio/volk/blob/001e60a7f33cdc5430892201af88a332b67f5872/kernels/volk/volk_32f_x2_pow_32f.h#L777 (Also lines 400, 548, 162, 280, 666.)
> ```

The command does the following according to < https://software.intel.com/en-us/cpp-compiler-developer-guide-and-reference-mm256-cvtepi32-ps >:
```
Converts eight packed signed doubleword (32-bit) integers  in the source vector into eight packed single-precision (32-bit) floating-point values.
```
So 256 bits are converted into 256 bits, and hence any value in the variable is overwritten ... so zeroing out the variable before overwriting it seems unnecessary. Thus, yes, it seems like that command isn't necessary.

michaelld, 2020-01-13:
Good catches! Do you have time to put together a PR for these?

ValZapod, 2020-01-13:
@michaelld I am now checking all gnuradio. So would be nice if you pulled those, I mean for you it is so much simpler)) And I do not want to sign CLA) ;)

michaelld, 2020-01-13:
@ValZapod fair enough! Thanks for taking the time to track down these issues & note them here!

ValZapod, 2020-01-13:
@michaelld Here is how you check for avx2 https://github.com/webmproject/libvpx/blob/e2ade251b1231f59f3de3cd79ffbc01b2b2fec51/vpx_ports/x86.h#L213

It indeed looks like if `((xgetbv() & 0x6) == 0x6` should be done but you also check for ebx after cpuid 
And also look here, avx512 support is right or not... https://github.com/webmproject/libvpx/commit/602d25ec4344f96031ce1043c15490e79438c1bc

ValZapod, 2020-01-24:
@michaelld Should I give a pull request?

michaelld, 2020-01-24:
> @michaelld Should I give a pull request?

Sure! I started working on this but got side-tracked by other duties. If you have anything then please go ahead with it in one or more PRs!

ValZapod, 2020-01-27:
@michaelld BTW, did you know? That you cannot throw nothing? Haha https://stackoverflow.com/questions/43675700/is-is-possible-in-c-to-throw-nothing/43675980#43675980

ValZapod, 2020-01-27:
I will add @jessica-iwamoto to comment on `tmp = _mm256_setzero_ps(); `

ValZapod, 2020-01-28:
@jdemel Can you comment on AVX/2 things and https://github.com/gnuradio/volk/blob/001e60a7f33cdc5430892201af88a332b67f5872/kernels/volk/volk_32f_acos_32f.h#L175
now I cannot understand why this is invalid.

Also do you think it is safe to do 

> ./volk/kernels/volk/volk_32f_x2_pow_32f.h	348	note	V624 The constant 0.6931471805 is being utilized. The resulting value could be inaccurate. Consider using the M_LN2 constant from <math.h>.

jdemel, 2020-02-08:
I have no idea, why L175 does what it does. Seems unnecessary.

michaelld, 2020-02-08:
Here's a way to simplify that code:
{{{
    z = _mm256_sub_ps(z, _mm256_and_ps(_mm256_mul_ps(z, ftwos), condition));
    condition = _mm256_cmp_ps(z, fones, _CMP_LT_OS);
    x = _mm256_add_ps(z, _mm256_and_ps(_mm256_sub_ps(_mm256_div_ps(fones, z), z), condition));
}}}
(1) remove the `x = z;`
(2) replace the `add_ps(x` with `add_ps(z`

`z` is used further down in the code, so we can't totally get rid of `x`. But we can get rid of the `x = z` assignment by using `z` in the first spot where `x` is used.

I've tested this & it compiles and passes test.

ValZapod, 2020-02-08:
Okay, will do that...

michaelld, 2020-02-08:
There are 6 of those in the current `acos` code; ditto for `asin` and `atan` ... fun times!

ValZapod, 2020-02-08:
@michaelld Well, pushed... But are you familiar with those intrinsics? Because well, maybe in assembly there will be no difference. Can you check with godbolt? `_mm256_setzero_ps()` was much more obvious...

PR #339 (2020-02-14): Code cleanup
Partially fix #327
closed 2020-02-14 by michaelld