gnuradio/volk #329: volk_32fc_s32fc_x2_rotator_32fc 10 dB noise in results

ghostop14, 2020-01-14:
When testing the rotator block with a straight signal source I noticed that the volk implementation volk_32fc_s32fc_x2_rotator_32fc has almost 10 dB of noise that builds between -133 and -123 dB that is not present in the CPU-only version.  Testing Flowgraph: signal source -> throttle -> rotator -> Frequency Sink, samp_rate=2.4e6, frequency rotation 100e3.  Running on an i7-7700HQ CPU on Ubuntu 18 64-bit.  You'll see it start okay, then build, reach a peak, then "break", then start the cycle over again.

To see the difference, compile the gr-blocks/rotator_cc_impl.cc with the "#if 0" and with "#if 1" in the work function (test code looks like it's already there to flip back and forth between CPU and volk).

Just a suspicion but I saw similar behavior on GPUs with floats where the CPU is maintaining double precision accuracy but the GPU was using true IEEE spec 9 decimal point floats.  The difference in accuracy was enough to introduce similar noise.

jdemel, 2020-01-25:
Is this related to #331 ?
-123 dB noise is still way below our usual target. We compromise speed for accuracy sometimes. But usually with below `10^-7` deviation from our reference. -123dB would translate into smth like `10^-12`.
Also, -123dB is way above the dynamic range of a 16bit ADC and the high speed ADCs we see in hardware are around 12 bit. Just assuming a 16bit DAC, we would probably see a noise floor at -100db and on the receiver side, we would need an SNR in the range of 100dB to even get close to a situation where this inaccuracy would effect our system.
Correct me if I'm wrong but this is an issue we can only see in theory. Maybe my whole train of thought regarding ADCs etc. is wrong. Also, I've personally never done any simulation that is even close to 100dB SNR and I can't picture a situation where this is currently useful.

ghostop14, 2020-01-25:
Hi @jdemel , not sure if it's related to 331 or not.  It could be.  I've tested on both GR 3.9 and GR 3.8 versions of volk (I think 3.9 uses a newer version) just to confirm.  It's not just theoretical, if you create a new flowgraph, set samp_rate to 2.4M, then just connect a signal source to a throttle to a rotator (freq shift -100e3 in my test), then a GUI Frequency Sink (without adjusting the dB range so you see 0-> -140), you'll see it.  It'll start fine, then the whole signal will get an oscillation that builds, peaks, then appears to "break" or snap back to normal then cycle again.

jdemel, 2020-01-26:
As a reference, the GR rotator is implemented in `gr-blocks/include/gnuradio/blocks/rotator.h`.

So, we need to figure out if this effect is caused by SSE/AVX because it works with lower precision than the CPU floating point unit or if this is an actual discrepancy in our code.
If this effect is due to those AVX units, I'd say we can't fix hardware and this is a wontfix. Otherwise, we should discuss how to proceed.

Further, I'd like to figure out if this effect is also observable with the generic kernel.
Besides, can you rerun your test with `export VOLK_GENERIC=1`?  That should basically be the same as GR compiled to not use VOLK. The generic kernel code and the one in GR is basically the same.
If this effect doesn't appear with the generic kernel, we build some confidence that our reference is good.

A bit on the relevance of this issue. In your test, at which dB level is your signal peak?
My argument is basically that an SNR loss is acceptable as long as your SNR is high enough (~90dB) anyways because other impairments add so much more noise that this issue doesn't impact practical flowgraphs. I could be mistaken here. I'd like to figure this out too.

ghostop14, 2020-01-26:
Hi @jdemel,  just tested with export VOLK_GENERIC=1 and it didn't produce the same effect.  If you run the attached flowgraph both ways, you'll see the impact.  It could easily have to do with precision, it's just odd the way it builds and breaks, then cycles.  Do you have a fix for 331?  I can always test that and see if it corrects it.

[test_rotator.grc.txt](https://github.com/gnuradio/volk/files/4113973/test_rotator.grc.txt)

jdemel, 2020-01-27:
Since you don't observe this effect with the generic kernel, I assume it is not related to #331 because this is an issue with the generic kernel.

I'll test this flowgraph. But I wonder what we could do to improve precision. And more so if we would make the kernel slower and how to avoid this. @ghostop14 do you have a suggestion how to fix this?

ghostop14, 2020-01-27:
@jdemel, looking through the code, I don't know for sure but I have a thought.... in the pure GR rotator, there's a "stateful" d_counter that's used for the iterator %512 calculation.  However, in the volk kernels, the calculation is always based on starting at i=0 with every call so the rolling 512 gets lost on each subsequent call.  I wonder if that's causing the artifact. (See the single rotate() function in gr-blocks/include/gnuradio/blocks/rotator.h).  May need to pass that in/out as a ref/pointer parameter so it doesn't get lost (I know that would change the function signature).  May be worth using the GR rotator output as "golden data" and see how the volk output compares?

jdemel, 2020-01-27:
So with `VOLK_GENERIC=1`, the effect is gone?
Based on that observation I draw the conclusion that the generic VOLK kernel produces the same result as the one in GR.
All the AVX/SSE/Neon kernels normalize phase at the end of each call. The generic version does so only if more 512 or more elements are processed. It should do so at the end of each call as well. That's exactly #331 .

ghostop14, 2020-01-27:
Also realized the flowgraph I attached used a new convenience wrapper.  This one is the same thing but should run on the current 3.8/3.9.
[test_rotator.grc.txt](https://github.com/gnuradio/volk/files/4118132/test_rotator.grc.txt)

ghostop14, 2020-01-27:
Also, an update was pushed in gr-blocks/lib/rotator_cc_impl.cc in the work function to temporarily bypass folk so make sure the "if 1" is changed to "if 0" to make sure the volk function is called.

ghostop14, 2020-01-27:
I had a chance to save out 3 scenarios: c++, volk generic, and volk (on my system it was using avx as the optimized kernel).  After 2048 samples you can start to see the diversion:

sample	CPP Real	Volk AVX Real	volk generic real	CPP Imag	Volk AVX Imag	volk generic imag
2048	0.928807735443	0.928745627403	0.928807735443	0.370554447174	0.370553135872	0.370554447174
2049	0.799686908722	0.799626410007	0.799686908722	0.600419163704	0.60040140152	0.600419163704
2050	0.614973664284	0.614920854568	0.614973664284	0.788544356823	0.788511991501	0.788544356823

Looks like it starts deviating at about the 5th decimal.  IEEE float single precision should take it out a bit further (9 I think).  Not sure if that helps or not.

jdemel, 2020-01-28:
Actually, we decided to set the tolerance for this kernel to `1e-3`. 

https://github.com/gnuradio/volk/blob/d3e9614cb9b0f5b2f65a7f9b77ede7421557c530/lib/kernel_tests.h#L32

Thus, our CI does not catch that behavior. Thus, this is known behavior.

Eventually, the question is:
Do we want and need that precision?

ghostop14, 2020-01-28:
What's the performance impact of adjusting the accuracy to say 1e-7 or 1e-6? (and could you suggest what change to make in the kernel to adjust the precision?  I could test how far it would have to go to minimize it, it may or may not be off by a decimal or two).

ghostop14, 2020-02-01:
Hi @jdemel , just following up on this.  Happy to do some additional testing.  After giving it some thought, I think the current accuracy is at 1e-5.  Going to 1e-6 I think should do the trick.  Is this something I can adjust in my local source and test?

jdemel, 2020-02-02:
In our test code, we set it to 1^-3. that's probably for a reason. 
Anyways, if you could add a PR for this, that would be a good start to discuss a specific solution.

PR #366 (2020-04-02): Fixed Broken Rotator
Fixes #329 

The non-generic rotator blocks were using incorrect loop maximums
for the outer, inner, and remainder loops that were causing
calculation issues.  This commit corrects those issues and restores
the rotator accuracy to the default 1e-6 precision.
closed 2020-04-02 by jdemel