gnuradio/volk #331: generic volk_32fc_s32fc_x2_rotator_32fc does not normalize rotator on short sequences

michaelld, 2020-01-21:
Moved from https://github.com/gnuradio/gnuradio/issues/1019 .

michaelld, 2020-01-21:
mbr0wn commented on Sep 17, 2016
In gnuradio/volk/kernels/volk/volk_32fc_s32fc_x2_rotator_32fc.h, all special functions are normalizing rotator to unit vector at the end:

    if (i) { 
        tmp1 = _mm_mul_ps(phase_Val, phase_Val);
        tmp2 = _mm_hadd_ps(tmp1, tmp1);
        tmp1 = _mm_shuffle_ps(tmp2, tmp2, 0xD8);
        tmp2 = _mm_sqrt_ps(tmp1);
        phase_Val = _mm_div_ps(phase_Val, tmp2);
    }

However, the generic implementation normalizes it only every ROTATOR_RELOAD (currently 512) samples. This means that repeated rotations of less than 512 samples may lead to walking into infinity or zero.

Code to reproduce:

int main(int argc, char **argv) {

  int N = 100;
  unsigned int alignment = volk_get_alignment();
  lv_32fc_t* in  = (lv_32fc_t*)volk_malloc(sizeof(lv_32fc_t)*N, alignment);
  lv_32fc_t* out = (lv_32fc_t*)volk_malloc(sizeof(lv_32fc_t)*N, alignment);
  float frequency = 0.1f;
  lv_32fc_t phase_increment = lv_cmake(cos(frequency), sin(frequency));
  lv_32fc_t phase = lv_cmake(1.f, 0.0f);

  for(int i = 0; i<1000*1000*1000; i++) {
    volk_32fc_s32fc_x2_rotator_32fc(out, in, phase_increment, &phase, N);
    if(i%1000000 == 0) {
      printf("at %i vector magnitude: %f\n", i, hypotf(lv_creal(phase), lv_cimag(phase)));
    }
  }

}
AVX impl:

at 0 vector magnitude: 1.000000
at 1000000 vector magnitude: 1.000000
at 2000000 vector magnitude: 1.000000
at 3000000 vector magnitude: 1.000000
at 4000000 vector magnitude: 1.000000
at 5000000 vector magnitude: 1.000000
at 6000000 vector magnitude: 1.000000
at 7000000 vector magnitude: 1.000000
at 8000000 vector magnitude: 1.000000
^C
generic impl:

at 0 vector magnitude: 1.000001
at 1000000 vector magnitude: 3.445564
at 2000000 vector magnitude: 11.871789
at 3000000 vector magnitude: 40.909409
at 4000000 vector magnitude: 140.926544
at 5000000 vector magnitude: 485.528656
at 6000000 vector magnitude: 1673.781860
at 7000000 vector magnitude: 5765.379395
at 8000000 vector magnitude: 19869.707031
^C
I suggest the atatched patch.

@mbr0wn
 
Member
Author
mbr0wn commented on Sep 17, 2016
Author: Jan Hrach
Date: Fri, 29 July 2016 22:32:19

The attachements (and previews) unfortunately don't seem to work. The patch inserted inline:

diff --git a/kernels/volk/volk_32fc_s32fc_x2_rotator_32fc.h b/kernels/volk/volk_32fc_s32fc_x2_rotator_32fc.h
index 70bc153..76ed730 100644
--- a/kernels/volk/volk_32fc_s32fc_x2_rotator_32fc.h
+++ b/kernels/volk/volk_32fc_s32fc_x2_rotator_32fc.h
@@ -109,6 +109,14 @@ static inline void volk_32fc_s32fc_x2_rotator_32fc_generic(lv_32fc_t* outVector,
_outVector++ = *inVector++ * (_phase);
(*phase) *= phase_inc;
}

if(i) {
+#ifdef __cplusplus
   (_phase) /= std::abs((_phase));
+#else
   (_phase) /= hypotf(lv_creal(_phase), lv_cimag(*phase));
+#endif
+
}
}

jdemel, 2020-01-23:
So this issue is a remainder of #49 .

`export VOLK_GENERIC=1` forces generic kernel use and `unset VOLK_GENERIC` gets you back to normal.

With GCC
`gcc -Wall -I[INSTALL_PREFIX]/include -L[INSTALL_PREFIX]/lib test_volk_32fc_s32fc_x2_rotator_32gc_issue_1019.c -lvolk -lm`

I suggest to always use `hypotf` instead of this ifdef because the change is in a file in `kernels/volk` where we try to use C. A quick `grep -irnI "std::" kernels/volk` proves me wrong though. That's actually related to #281 . And another good reason to tackle this issue.

PR #344 (2020-02-14): rotator: Refactor AVX kernels
Fix #331

This started as some work to track down the root cause of #329 . As a first result, it fixes #331 .
Apparently, these two issues are not as closely related as previously suspected.

Besides a fix for the generic kernel, this PR includes quite a bit of refactoring for better readibility of the AVX u/a rotator kernels.
commit 2bc699d1 (2020-02-14): rotator: Refactor AVX kernels

Fix #331

This is ongoing work to track down the root cause of #329.

1. Make sure all input scalars are normalized -> no improvement
2. Trigger renormalization after same num_points -> no improvement

Lower tolerance values move the first point that breaks this bound
further to the front. It is not obvious why this would happen so far.

Since SSE kernels tend to break later, approx. factor 2, maybe the
generic kernels accumulates errors faster. That's just a wild guess
though.
closed 2020-02-14 by michaelld