gnuradio/volk #335: Apply clang-format

jdemel, 2020-01-26:
I suggest we take the clang format file from GR and apply it to all our C/C++ files. Further, we should set up a GitHub Action to apply clang-format to all PRs.

michaelld, 2020-01-26:
Yes, agreed! We'll get there, sooner or later ...

jdemel, 2020-01-28:
I just tested this:
`find . -regex '.*\.\(c\|hh\|cc\|cxx\)' -exec clang-format -style=file -i {} \;`
with [GR .clang-format](https://github.com/gnuradio/gnuradio/blob/master/.clang-format).
It works pretty well, except that we need to exclude `tmpl`. If we adopt this style, we can set up a GitHub Action to auto-format code. That'll make it easier to review code.
Though, we need to agree on coding style. Should we just pull in GNU Radios clang-format?

michaelld, 2020-01-28:
Yeah I'm thinking pulling in the GR clang-format is good enough for our needs. We might not even need all of the settings.

mbr0wn, 2020-01-28:
You should be able to re-use the GH Action from GNU Radio with very little modifications. However, it assumes the codebase is in a good shape already, it's not smart enough to only check the diff of a given PR.

closed 2020-03-18 by michaelld