gnuradio/volk #336: Update CI infrastructure

jdemel, 2020-01-26:
Currently we have 3 CI systems that cover our codebase partially. And a fourth one to consider.
I have a list of suggestions for updates and issues that we should address. Additions and comments are welcome.

- Travis is probably the one we rely on the most. It needs to be updated! 
     - `distro: bionic` should be the baseline instead of `trusty`
     - All Jobs should have names to make them easier to distinguish
     - All jobs should be run on amd64 and aarch64 and maybe aarch32. That would probably imply to merge some jobs into one.
     - Use Python 3 only (remove six)
     - Include Boost only for old systems e.g. old enough GCC/Clang which do not provide `std::filesystem` or `experimental`
- AppVeyor to check if MSVC is broken
     - Needs an update to use Python3. 
     - Remove obsolete packages (e.g. Cheetah)
- BuildBot, the GR worker
     - Needs to run reliably.
     - Needs an update to use newer Ubuntu versions etc.
- GitHub Actions
     - Since we are on GitHub, we should use it as well to test at least on build

michaelld, 2020-01-26:
good list! Lots to do here ... never enough time in the day!

ValZapod, 2020-01-28:
It is broken now as python3.8 brokes everything... Sigh.

jdemel, 2020-01-28:
@ValZapod what exactly is broken?

ValZapod, 2020-01-28:
@jdemel Look into my pull request (bb/* builds)
```
/usr/lib/python3.8/site-packages/buildbot/process/buildstep.py:567 in startStep
566                    self._running = True
567                    self.results = yield self.run()
568                finally:
```

jdemel, 2020-01-28:
@ValZapod So BuildBot/GR is broken. Yeah, if we continue to use this system, we need to update it.

ValZapod, 2020-01-28:
@jdemel It just updated to 3.8 version of python. Automatically.

jdemel, 2020-02-15:
Lately, we had a couple of occasions where MacOS was a PITA. 
That's a really good reason to integrate CI for MacOS. GitHub Actions as well as TravisCI support MacOS now, so this should actually be possible.
In essence, we'll have CI for Linux (GCC, Clang), Windows and MacOS then.
Since some GR users report bugs on BSDs, it'd be nice to have CI feature but I haven't seen a CI service for that yet.

PR #376 (2020-04-19): ci: Add Github CI Action
This commit adds CI tests for GitHub Actions for Linux, Windows and MacOS.

Fixes #336 

Since we use GitHub for code hosting, it is worthwhile to have CI tests that run in every VOLK repo on push. 

Be aware that Windows and MacOS CI does only run generic kernels at the moment because it cannot find `xgetbv`.
closed 2020-04-19 by michaelld