gnuradio/volk #340: qa_volk_32f_8u_polarbutterflypuppet_32f failing with new gcc/g++

dkozel, 2020-01-30:
This issue was reported by @sirjansel0t on https://github.com/gnuradio/gnuradio/issues/3116

The QA test passes on gcc/g++ 8.3 and fails with gcc/g++ 9.2.1

jdemel, 2020-02-08:
Could you provide more details on this error @sirjansel0t ?
Which CPU did you use? ARM, Intel, AMD? Which OS/distribution?

jdemel, 2020-02-09:
I just tested this issue with this Dockerfile
```
FROM ubuntu:20.04

RUN apt-get update && apt-get clean && apt-get autoremove && apt-get install -y apt-utils
RUN apt-get install -y \
        build-essential \
        cmake \
        git \
        pkg-config \
        python3-mako \
        python3-numpy \
        python3-distutils \
        gcc-9 g++-9 gdb
```
I have this Dockerfile in my VOLK top level directory. Then I run:
`docker build -t volkbuilder .`
Afterwards:
`docker run -v $(pwd):/opt/volk -it volkbuilder bash`
Finally, in my container, I go to `/opt/volk` and create a new build folder and run `cmake`, `make - j8` and `ctest`. All tests pass. This is on an `Intel Core i7-3630QM`.

Just to double check, this is the compiler config that cmake picks up:
```
-- The C compiler identification is GNU 9.2.1
-- The CXX compiler identification is GNU 9.2.1
[...]
Compiler Version: cc (Ubuntu 9.2.1-28ubuntu1) 9.2.1 20200203
```
So this test runs on GCC 9.21
We need more info on that issue because it is not reproducible so far.

nckx, 2020-03-11:
Not sure if it's the same thing, but I encounter a similar error on [Guix](https://guix.gnu.org) [fe60ef998](https://git.savannah.gnu.org/cgit/guix.git/commit/?id=fe60ef998f537e0e71bbb3377d5886ad297821d0), which uses GCC 7.4.0.  So not that recent.

It happens on the following CPU models (and more):
- Intel(R) Core(TM) i5-6402P CPU @ 2.80GHz
- Intel(R) Xeon(R) CPU E5-2695 v4 @ 2.10GHz

Please let me know if I can provide or test anything!

<details><summary>Here's the log.</summary>

```
starting phase `set-SOURCE-DATE-EPOCH'
phase `set-SOURCE-DATE-EPOCH' succeeded after 0.0 seconds
starting phase `set-paths'
environment variable `PATH' set to `/gnu/store/iz9500ssxcqlyr74hg1jq10ycrh42yq1-cmake-minimal-3.15.1/bin:/gnu/store/iaqmwj2290z5nnrk69bss6r0d9lpr8cs-python2-2.7.16/bin:/gnu/store/y5nxvx3gq3a9g98hdmwn8lynqardgmv6-python2-cheetah-3.1.0/bin:/gnu/store/cnqpra8vr2l5fz00rr4yj4bp3hr00cfw-tar-1.32/bin:/gnu/store/py3k9zla9fj3z7430v4crqj5pyrsd3qj-gzip-1.10/bin:/gnu/store/l86azr7r3p5631wj3kk329jl1y1mpjgy-bzip2-1.0.6/bin:/gnu/store/lbip9isk25isymvnb159l115xnacb5j8-xz-5.2.4/bin:/gnu/store/6jdshxwdrad9mlhcqc9k0g24yw45rqf1-file-5.33/bin:/gnu/store/58sq8iabw3jkv0fvf95hd7sq2g4xcsnz-diffutils-3.7/bin:/gnu/store/v76scv4n63ip08g119rczh2mrw31zwpd-patch-2.7.6/bin:/gnu/store/g9d3wv1d68iflx57yp3mcp3k3sv8spsl-findutils-4.6.0/bin:/gnu/store/2z9hsww76aag37p40671l9niq5pvvasx-gawk-5.0.1/bin:/gnu/store/afmvfw1yhfal48n1kjq6bk6kcw8sc3db-sed-4.7/bin:/gnu/store/7iyvxhp2g3v3655zqwr6biz2h0lqv7pr-grep-3.3/bin:/gnu/store/9kzrrccpzl6i1sfwb0drb00gi2gwk0x0-coreutils-8.31/bin:/gnu/store/b5vpfzkr59bpgcsg1k9vvad9h5rwvpgk-make-4.2.1/bin:/gnu/store/29jhbbg1hf557x8j53f9sxd9imlmf02a-bash-minimal-5.0.7/bin:/gnu/store/nc5vlidpxbvngalng30nif8nb3j7gfy2-ld-wrapper-0/bin:/gnu/store/3hkdiscs4910r75njbrql10znxxn7chk-binutils-2.32/bin:/gnu/store/x3jx25cd3q363mr7nbgzrhrv1vza6cf7-gcc-7.4.0/bin:/gnu/store/ahqgl4h89xqj695lgqvsaf6zh2nhy4pj-glibc-2.29/bin:/gnu/store/ahqgl4h89xqj695lgqvsaf6zh2nhy4pj-glibc-2.29/sbin:/gnu/store/6k4gcgbav9vz018x0x8v149l0f5pn5db-python2-markdown-3.1.1/bin'
environment variable `CMAKE_PREFIX_PATH' set to `/gnu/store/iz9500ssxcqlyr74hg1jq10ycrh42yq1-cmake-minimal-3.15.1/:/gnu/store/iaqmwj2290z5nnrk69bss6r0d9lpr8cs-python2-2.7.16/:/gnu/store/y5nxvx3gq3a9g98hdmwn8lynqardgmv6-python2-cheetah-3.1.0/:/gnu/store/cnqpra8vr2l5fz00rr4yj4bp3hr00cfw-tar-1.32/:/gnu/store/py3k9zla9fj3z7430v4crqj5pyrsd3qj-gzip-1.10/:/gnu/store/l86azr7r3p5631wj3kk329jl1y1mpjgy-bzip2-1.0.6/:/gnu/store/lbip9isk25isymvnb159l115xnacb5j8-xz-5.2.4/:/gnu/store/6jdshxwdrad9mlhcqc9k0g24yw45rqf1-file-5.33/:/gnu/store/58sq8iabw3jkv0fvf95hd7sq2g4xcsnz-diffutils-3.7/:/gnu/store/v76scv4n63ip08g119rczh2mrw31zwpd-patch-2.7.6/:/gnu/store/g9d3wv1d68iflx57yp3mcp3k3sv8spsl-findutils-4.6.0/:/gnu/store/2z9hsww76aag37p40671l9niq5pvvasx-gawk-5.0.1/:/gnu/store/afmvfw1yhfal48n1kjq6bk6kcw8sc3db-sed-4.7/:/gnu/store/7iyvxhp2g3v3655zqwr6biz2h0lqv7pr-grep-3.3/:/gnu/store/9kzrrccpzl6i1sfwb0drb00gi2gwk0x0-coreutils-8.31/:/gnu/store/b5vpfzkr59bpgcsg1k9vvad9h5rwvpgk-make-4.2.1/:/gnu/store/29jhbbg1hf557x8j53f9sxd9imlmf02a-bash-minimal-5.0.7/:/gnu/store/nc5vlidpxbvngalng30nif8nb3j7gfy2-ld-wrapper-0/:/gnu/store/3hkdiscs4910r75njbrql10znxxn7chk-binutils-2.32/:/gnu/store/x3jx25cd3q363mr7nbgzrhrv1vza6cf7-gcc-7.4.0/:/gnu/store/ahqgl4h89xqj695lgqvsaf6zh2nhy4pj-glibc-2.29/:/gnu/store/qky1x5bb2jygy58bn6y95ygfsmpakf52-glibc-2.29-static/:/gnu/store/mmqp1xqffn6qw6v88i627c2bpbq36fcy-glibc-utf8-locales-2.29/:/gnu/store/f5g4av3mwn7zr81yqr1gn9hpb5d2c4m4-boost-1.70.0/:/gnu/store/6k4gcgbav9vz018x0x8v149l0f5pn5db-python2-markdown-3.1.1/:/gnu/store/7czrqpi0kwazras6pgyx0bhdn89pg0jl-linux-libre-headers-4.19.56/'
environment variable `PYTHONPATH' set to `/gnu/store/iaqmwj2290z5nnrk69bss6r0d9lpr8cs-python2-2.7.16/lib/python2.7/site-packages:/gnu/store/y5nxvx3gq3a9g98hdmwn8lynqardgmv6-python2-cheetah-3.1.0/lib/python2.7/site-packages:/gnu/store/6k4gcgbav9vz018x0x8v149l0f5pn5db-python2-markdown-3.1.1/lib/python2.7/site-packages'
environment variable `BASH_LOADABLES_PATH' unset
environment variable `CPATH' set to `/gnu/store/iaqmwj2290z5nnrk69bss6r0d9lpr8cs-python2-2.7.16/include:/gnu/store/l86azr7r3p5631wj3kk329jl1y1mpjgy-bzip2-1.0.6/include:/gnu/store/lbip9isk25isymvnb159l115xnacb5j8-xz-5.2.4/include:/gnu/store/6jdshxwdrad9mlhcqc9k0g24yw45rqf1-file-5.33/include:/gnu/store/2z9hsww76aag37p40671l9niq5pvvasx-gawk-5.0.1/include:/gnu/store/b5vpfzkr59bpgcsg1k9vvad9h5rwvpgk-make-4.2.1/include:/gnu/store/3hkdiscs4910r75njbrql10znxxn7chk-binutils-2.32/include:/gnu/store/x3jx25cd3q363mr7nbgzrhrv1vza6cf7-gcc-7.4.0/include:/gnu/store/ahqgl4h89xqj695lgqvsaf6zh2nhy4pj-glibc-2.29/include:/gnu/store/f5g4av3mwn7zr81yqr1gn9hpb5d2c4m4-boost-1.70.0/include:/gnu/store/7czrqpi0kwazras6pgyx0bhdn89pg0jl-linux-libre-headers-4.19.56/include'
environment variable `LIBRARY_PATH' set to `/gnu/store/iaqmwj2290z5nnrk69bss6r0d9lpr8cs-python2-2.7.16/lib:/gnu/store/y5nxvx3gq3a9g98hdmwn8lynqardgmv6-python2-cheetah-3.1.0/lib:/gnu/store/l86azr7r3p5631wj3kk329jl1y1mpjgy-bzip2-1.0.6/lib:/gnu/store/lbip9isk25isymvnb159l115xnacb5j8-xz-5.2.4/lib:/gnu/store/6jdshxwdrad9mlhcqc9k0g24yw45rqf1-file-5.33/lib:/gnu/store/2z9hsww76aag37p40671l9niq5pvvasx-gawk-5.0.1/lib:/gnu/store/3hkdiscs4910r75njbrql10znxxn7chk-binutils-2.32/lib:/gnu/store/ahqgl4h89xqj695lgqvsaf6zh2nhy4pj-glibc-2.29/lib:/gnu/store/qky1x5bb2jygy58bn6y95ygfsmpakf52-glibc-2.29-static/lib:/gnu/store/mmqp1xqffn6qw6v88i627c2bpbq36fcy-glibc-utf8-locales-2.29/lib:/gnu/store/f5g4av3mwn7zr81yqr1gn9hpb5d2c4m4-boo [... cut ...]
Best aligned arch: generic
Best unaligned arch: generic
Failure on volk_32f_8u_polarbutterflypuppet_32f
Kernel QA finished: 1 failures out of 101 tests.
The following kernels failed QA:
    volk_32f_8u_polarbutterflypuppet_32f

0% tests passed, 1 tests failed out of 1

Total Test time (real) =   0.65 sec

The following tests FAILED:
	  1 - qa_test_all (Failed)
Errors while running CTest
make: *** [Makefile:89: test] Error 8

Test suite failed, dumping logs.
command "make" "test" failed with status 2
```
</details>

jdemel, 2020-06-14:
I missed the update in this issue. 

Can we reproduce this error on a recent VOLK version?  VOLK 1.3 was released July 2016 and I suppose this issue got fixed since then. VOLK 2.3 should be compatible to 1.3.

jdemel, 2020-10-21:
I assume this issue got resolved by updating to a newer VOLK release.
I close this issue now. Feel free to reopen it in case the issue persists.

closed 2020-10-21 by jdemel

LukasKrieg, 2021-01-09:
Got the Same Issue as @sirjansel0t with a more recent libvolk version.

Processor is an Intel Xeon E3-1231v3
GCC: gcc (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
OS: Ubuntu 20.04.1 LTS
VOLK:
libvolk2-bin/focal,now 2.2.1-1 amd64 
libvolk2-dev/focal,now 2.2.1-1 amd64 
libvolk2-doc/focal,focal 2.2.1-1 all
libvolk2.2/focal,now 2.2.1-1 amd64  

Compile Gnuradio v3.8.2.0 and ran 

`make test`

```
99% tests passed, 1 tests failed out of 161

Total Test time (real) =   6.35 sec

The following tests FAILED:
	 25 - qa_volk_32f_8u_polarbutterflypuppet_32f (Failed)
Errors while running CTest
make: *** [Makefile:130: test] Fehler 8
```

`ctest -V -R qa_volk_32f_8u_polarbutterflypuppet_32f`

```

> UpdateCTestConfiguration  from :/home/study/workarea/gnuradio/build/DartConfiguration.tcl
> UpdateCTestConfiguration  from :/home/study/workarea/gnuradio/build/DartConfiguration.tcl
> Test project /home/study/workarea/gnuradio/build
> Constructing a list of tests
> Done constructing a list of tests
> Updating test list for fixtures
> Added 0 tests to meet fixture requirements
> Checking test dependency graph...
> Checking test dependency graph end
> test 25
>     Start 25: qa_volk_32f_8u_polarbutterflypuppet_32f
> 
> 25: Test command: /usr/bin/sh "/home/study/workarea/gnuradio/build/volk/lib/volk_32f_8u_polarbutterflypuppet_32f_test.sh" "/home/study/workarea/gnuradio/build/volk/lib"
> 25: Test timeout computed to be: 10000000
> 25: RUN_VOLK_TESTS: volk_32f_8u_polarbutterflypuppet_32f(131071,1)
> 25: generic completed in 4.7462 ms
> 25: u_avx completed in 0.706781 ms
> 25: u_avx2 completed in 0.641404 ms
> 25: offset 8 in1: -2 in2: -0.498039 tolerance was: 1e-06
> 25: offset 9 in1: -4 in2: -2.49804 tolerance was: 1e-06
> 25: offset 10 in1: -4 in2: -0.996078 tolerance was: 1e-06
> 25: offset 11 in1: 8 in2: 4.99608 tolerance was: 1e-06
> 25: offset 12 in1: -4 in2: -0.996078 tolerance was: 1e-06
> 25: offset 13 in1: 8 in2: 4.99608 tolerance was: 1e-06
> 25: offset 14 in1: -8 in2: -2.99608 tolerance was: 1e-06
> 25: offset 15 in1: -16 in2: -10.9961 tolerance was: 1e-06
> 25: offset 16 in1: 2 in2: 0.498039 tolerance was: 1e-06
> 25: offset 17 in1: 4 in2: 2 tolerance was: 1e-06
> 25: volk_32f_8u_polarbutterflypuppet_32f: fail on arch u_avx2
> 25: offset 26 in1: 0 in2: 1 tolerance was: 0
> 25: offset 27 in1: 0 in2: 1 tolerance was: 0
> 25: offset 30 in1: 1 in2: 0 tolerance was: 0
> 25: offset 31 in1: 0 in2: 1 tolerance was: 0
> 25: offset 32 in1: 0 in2: 1 tolerance was: 0
> 25: offset 36 in1: 0 in2: 1 tolerance was: 0
> 25: offset 48 in1: 0 in2: 1 tolerance was: 0
> 25: offset 52 in1: 0 in2: 1 tolerance was: 0
> 25: offset 56 in1: 0 in2: 1 tolerance was: 0
> 25: offset 59 in1: 0 in2: 1 tolerance was: 0
> 25: volk_32f_8u_polarbutterflypuppet_32f: fail on arch u_avx2
> 25: Best aligned arch: u_avx
> 25: Best unaligned arch: u_avx
> 1/1 Test #25: qa_volk_32f_8u_polarbutterflypuppet_32f ...***Failed    0.01 sec
> 
> 0% tests passed, 1 tests failed out of 1
> 
> Total Test time (real) =   0.02 sec
> 
> The following tests FAILED:
> 	 25 - qa_volk_32f_8u_polarbutterflypuppet_32f (Failed)
> Errors while running CTest

```

jdemel, 2021-01-11:
@LukasKrieg Did you build GNU Radio from source but installed a system VOLK version? Afaik, GNU Radio 3.8 does still use VOLK as a submodule. You might end up with 2 different versions.

LukasKrieg, 2021-01-11:
i did not install VOLK specifically. I followed this Guide

https://kb.ettus.com/Building_and_Installing_the_USRP_Open-Source_Toolchain_(UHD_and_GNU_Radio)_on_Linux

up until the required dependencies for Ubuntu had to be installed, since a lot of the packages were deprecated and were not present i followed the advice of

https://itectec.com/ubuntu/ubuntu-how-to-solve-the-error-unable-to-locate-package-libuhd003-for-ubuntu-20-04/

and installed libuhd-dev
Building older versions of UHD led to Errors, which led to some forum threads about deprecated dependencies
UHD v4.0.0.0 was able to build without errors and make test did not throw any errors.
Gnuradio v3.8.2.0 was also able to build without errors but make test showed this lonely VOLK Error

I think libvolk2.2 was installed as some dependencie.

I just ignored the Issue, Gnuradio-companion is working as far as i can tell and my USRP was also detected. 
I just wanted to let you know since the Error was not reproducible for you. If you need any more information, let me know.