gnuradio/volk #348: Make sure checks for AVX(2) and AVX-512 are correct

ValZapod, 2020-02-18:
Here is how you check for avx2 https://github.com/webmproject/libvpx/blob/e2ade251b1231f59f3de3cd79ffbc01b2b2fec51/vpx_ports/x86.h#L213
'''
if (max_cpuid_val >= 7) {
        /* Get the leaf 7 feature flags. Needed to check for AVX2 support */
        cpuid(7, 0, reg_eax, reg_ebx, reg_ecx, reg_edx);

        if (reg_ebx & BIT(5)) flags |= HAS_AVX2;
''''
besides 
'''
if ((reg_ecx & (BIT(27) | BIT(28))) == (BIT(27) | BIT(28))) {
    // Check for OS-support of YMM state. Necessary for AVX and AVX2.
    if ((xgetbv() & 0x6) == 0x6) {
      flags |= HAS_AVX;
'''
https://github.com/gnuradio/volk/blob/1ec627c153a9625b588eef394c70af42811c53ab/tmpl/volk_cpu.tmpl.c#L114
It indeed looks like if `((xgetbv() & 0x6) == 0x6` should be done but you also check for ebx after cpuid 
And also look here, avx512 support is right or not... https://github.com/webmproject/libvpx/commit/602d25ec4344f96031ce1043c15490e79438c1bc

_Originally posted by @ValZapod in https://github.com/gnuradio/volk/issues/327#issuecomment-573787182_

jdemel, 2020-02-23:
I'd assume that [libcpuid](https://github.com/anrieff/libcpuid) could be a good source to figure out how to detect CPU features for x86.

ValZapod, 2020-02-23:
@jdemel The main problem jere is that those two functions 
https://github.com/gnuradio/volk/blob/2cf27f3dcc048e34852ff1a9d3a7f21998ae9641/tmpl/volk_cpu.tmpl.c#L106-L120
are the same, which is well, strange.

jdemel, 2020-02-24:
That's indeed very strange. Also, AppVeyor tells us that MSVC doesn't find any archs. I thought these issues are somehow related. We need to investigate this.

Is there a good resource out there to learn about the details? That would be a useful resource to continue this discussion.

ValZapod, 2020-02-24:
@jdemel You mean besides Intel Manual? Well, you can look into wikipedia, obvioulsy. Then look into my link on google implentation in chromium (as most know, chrome/chromium code is best of the best in the world, hehe).
Also there is ffmpeg implemenation (with NASM things also, I like that one).
Also stackechange ;)
But unfortunately I am not very into this staff, assembly is okay, AVX not so...

ValZapod, 2020-02-25:
Oh, I forgot this one https://github.com/sekrit-twc/zimg/blob/master/src/zimg/common/x86/cpuinfo_x86.cpp
where I found a bug in one of AVX512 things..

jdemel, 2020-03-17:
@ValZapod would you be willing to submit a PR that fixes this issue?

ValZapod, 2020-03-17:
No. I am not very into this stuff. Just note that gnss-sdr also uses this two identical functions... copy paste, I suppose.

jdemel, 2020-05-03:
Would [cpu_features](https://github.com/google/cpu_features) be a viable path forward? [libcpuid](https://github.com/anrieff/libcpuid) is restricted to x86.

I'd like to avoid to reinvent the wheel. This library does at least promise to support all archs we use. Further, we can include it as a submodule and just use it. At least that's the recommended way to use it here.

It is licensed under Apache 2. Thus, I assume we can use it as a submodule without any worries.
However, I don't want to run into a situation where distro maintainers need to untangle this.

ValZapod, 2020-05-04:
So you came to a conclusion that AVX2 check is incorrect?

jdemel, 2020-05-04:
@ValZapod , I didn't doubt the checks are suspicious. They look like they're incorrect.

What I try to discuss in this thread, is a way forward. I've actually seen more such issues. Most notably, on ThreadripperV3, the kernel reports fma but VOLK does not detect it. On a XEON 62xx the fma check passes.

All in all, CPU feature detection is always fragile. That's what I learned during my investigation. As a result I lean towards let's depend on a project that does this task well rather than we reinvent the wheel and need to fix all those bugs again.

ValZapod, 2020-05-04:
> depend on a project that does

This is a good idea! But I think some things like https://github.com/google/cpu_features/issues/4 are not our level? Or you know all of it ;)

jdemel, 2020-05-04:
Well, we need CPU feature detection. I guess we benefit from a project that does continuously improve without requiring us to know all the details. We can focus on kernels.

ValZapod, 2020-08-02:
@carlesfernandez You may be also intrested in this as here https://github.com/gnss-sdr/gnss-sdr/blob/66a908fc42debcb7157acd773bccc056233bdc9a/src/algorithms/libs/volk_gnsssdr_module/volk_gnsssdr/tmpl/volk_gnsssdr_cpu.tmpl.c#L111 is the same probem.

jdemel, 2020-08-02:
Yeah `volk_gnsssdr` derived from VOLK with some special kernels. No surprises here. I really hope we can improve the situation with `cpu_features`. This effort is somewhat stalling though because we need people to look at the Windows specific issues. At the moment mostly: Why does it not compile on Windows with MSVC? Could you have a look at this PR?

PR #393 (2020-10-23): cpu_features: Switch to cpu_features
We used our own CPU feature detection methods. These were not totally
reliable and difficult to implement across compilers and OSes. We switch
to `cpu_features` as a subproject and build upon this expertise.

So far this is a work-in-progress. Known issues:
   - <s>code needs clean-up</s>
   - <s>remove unnecessary checks</s>
   - <s>make sure all NEON versions are still detected correctly</s> - CI indicates correct detection.
   - <s>incomplete `VOLK_CPU` feature checks:</s>
	- <s>`has_64`: check 64bit CPU.</s> We compile for one or the other. Not for both [... cut]
closed 2020-10-23 by michaelld