gnuradio/volk #354: volk_32fc_s32fc_x2_rotator_32fc AVX kernel issues

michaelld, 2020-02-23:
As reported on Slack #gnuradio ... On GNU Radio 3.8.1rc1, on Ubuntu 20.04: audio is badly distorted. The culprit appears to be the volk_32fc_s32fc_x2_rotator_32fc AVX kernel. Reverting https://github.com/gnuradio/volk/pull/344 fixes the issue.

michaelld, 2020-02-23:
So @jdemel we somehow messed the AVX kernel up in that PR. I'd prefer to fix it up rather than revert it ... so much good work in there! Do you have the bandwidth to look into this?

jdemel, 2020-02-23:
That's a bummer. Is this related to the issue #329 ? And there is a start to increase precision in #341 .
I'd assume this would also happen with SSE. Can the issue be fixed with `VOLK_GENERIC`?
That would actually imply that we need more precision with this kernel. Also, it would point us in the right direction. 

If I recall correctly, the GR audio sink just got VOLKized. That does actually imply that we need more precision for this kernel. Smth we didn't really care for before.

argilo, 2020-02-23:
The generic kernel works fine, as does the AVX FMA kernel.

argilo, 2020-02-23:
I tested those cases by placing `volk_32fc_s32fc_x2_rotator_32fc generic generic` and then `volk_32fc_s32fc_x2_rotator_32fc a_avx_fma u_avx_fma` in my `.volk/vol_config`.

argilo, 2020-02-23:
When i `volk_32fc_s32fc_x2_rotator_32fc a_avx u_avx` the strong audio distortion occurs.

argilo, 2020-02-23:
The problem can be reproduced like so:

[image]

[image]

Interestingly, when the Audio Sink is removed, the discontinuities no longer appear in the Time Sink.

Perhaps some state being stored in a global variable?

rear1019, 2020-02-23:
> Interestingly, when the Audio Sink is removed, the discontinuities no longer appear in the Time Sink.
> 
> Perhaps some state being stored in a global variable?

Try to replace the audio sink with a file sink/wav sink and play back the file by other means (i.e. do not use GNU Radio’s audio sink).

argilo, 2020-02-23:
Actually the glitches do occur without the Audio Sink also, it seems I just wasn't patient enough.

argilo, 2020-02-23:
A simpler flowgraph that can be used to reproduce it:

[image]

[image]

argilo, 2020-02-23:
I think I found the bug. `num_points` should be `fourthPoints` here:

https://github.com/gnuradio/volk/blob/2cf27f3dcc048e34852ff1a9d3a7f21998ae9641/kernels/volk/volk_32fc_s32fc_x2_rotator_32fc.h#L530

closed 2020-02-23 by jdemel

jdemel, 2020-02-24:
Thanks for hunting down this bug. I assume, we should put out a bugfix release for that. 2.2.1.

argilo, 2020-02-24:
Yes, I think this should go into a bug fix release. Hopefully it's not too late to make it in before Ubuntu 20.04's Debian import freeze later this week.