gnuradio/volk #356: Test with short, medium and large vectors

jdemel, 2020-02-23:
Since we sometimes see issues with kernels on very short vector sizes and sometimes on larger vector sizes, we should run all tests on ~3 vector sizes. I suggest 127, 8091, and 2^17-1 (current default). I hope that catches a few more bugs.

argilo, 2020-02-24:
The trailing bits of all those numbers are the same, so doing those additional tests wouldn't have caught https://github.com/gnuradio/volk/pull/358. Those kernels only failed when bit 2 of `num_points` differs from bit 1.

Sweeping over a range of vector lengths (e.g. 0-32) might be more useful. The tests run quickly when the vectors are short.

argilo, 2020-02-24:
Some other failures I've noticed:

* `volk_8u_conv_k7_r2puppet_8u` crashes when the vector length is less than 12.
* `volk_8u_x3_encodepolarpuppet_8u` fails for `u_ssse3`, `u_avx2`, `a_ssse3` and `a_avx2` for vector lengths up to 15. `u_avx2`, and `a_avx2` fail all the way up to 31.

argilo, 2020-02-24:
* `volk_32fc_index_max_16u` fails intermittently for `a_avx2` and `u_avx2`. (Tested with vector length 21.)

argilo, 2020-02-24:
* `volk_32fc_index_max_32u` also fails intermittently for `a_avx2` and `u_avx2`. (Again, tested with vector length 21.)

rear1019, 2020-02-24:
> Sweeping over a range of vector lengths (e.g. 0-32) might be more useful. The tests run quickly when the vectors are short.

+1. This covers the most common errors when working on a new kernel. As mentioned in [1], this also would have cought the error reported in #355.

> Some other failures I've noticed:
> 
>     * `volk_8u_conv_k7_r2puppet_8u` crashes when the vector length is less than 12.

Not too surprising considering the used convolutional code :wink:.

[1] https://github.com/gnuradio/volk/pull/355#issuecomment-590114980

jdemel, 2020-02-24:
I totally agree. We need to sweep over a range of vector inputs. This should be the default.

With vector size `<12`, the convolutional decoder segfaults. With vector size `<16`, the polarencoder fails. In both cases, it doesn't make sense to use these codes with smaller sizes.

I'd like to update https://github.com/gnuradio/volk/blob/5c61b7ea9742c66bf43c48601e31f061e14b5fe2/lib/testqa.cc#L38
In this function, we hard code 
https://github.com/gnuradio/volk/blob/5c61b7ea9742c66bf43c48601e31f061e14b5fe2/lib/testqa.cc#L45
We should make the vector length an optional argument of this function. Then we can run all tests with arbitrary sizes and just call it with a range of small vector sizes in our CI.

jdemel, 2021-05-26:
In order to add this to our test system, we need to put a lot of effort into:

https://github.com/gnuradio/volk/blob/c9d42a4bf31af53ec2d81bbd1dec8754feb7303f/lib/qa_utils.cc#L511

And the whole infrastructure there. Also, this refactoring/rewrite would be required for #203 .

argilo, 2023-10-13:
> `volk_8u_x3_encodepolarpuppet_8u` fails for `u_ssse3`, `u_avx2`, `a_ssse3` and `a_avx2` for vector lengths up to 15. `u_avx2`, and `a_avx2` fail all the way up to 31.

Fixed in #642.

argilo, 2023-10-13:
And I just found another one: `volk_32fc_x2_square_dist_32f_a_sse3` has a buffer overrun when the vector length is less than 4, because the "last" iteration of an unrolled loop runs unconditionally.