gnuradio/volk #359: AVX2 index max kernels fail when vector length is odd

argilo, 2020-02-24:
As noted in https://github.com/gnuradio/volk/issues/356#issuecomment-590150800 and https://github.com/gnuradio/volk/issues/356#issuecomment-590320253, the AVX2 index max kernels sometimes fail. I had a look at the code, and the reason is that they fail to process the last element of the vector when `num_points` is odd. There is special handling for trailing groups of 4 or 2 elements, but not 1.

argilo, 2020-02-24:
This probably went unnoticed because tests are performed with very large vectors. The chances of catching the bug would be 1 in n for a vector of length n.

jdemel, 2020-02-24:
I assume this issue can be closed. Fix #358

argilo, 2020-02-24:
Nope, that was a second, closely related bug.

argilo, 2020-02-24:
In fact, found this bug on the branch where I had already patched #358. So it's definitely still present.

jdemel, 2020-02-24:
It took me a while to make it fail. But it seems this is it. 
```
$ ./apps/volk_profile -n -R fc_index -i 100000000 -v 7

RUN_VOLK_TESTS: volk_32fc_index_max_16u(7,100000000)
a_avx2 completed in 1307.81 ms
u_avx2 completed in 2708.38 ms

RUN_VOLK_TESTS: volk_32fc_index_max_32u(7,100000000)
a_avx2 completed in 1236.02 ms
u_avx2 completed in 2176.83 ms
offset 0 in1: 0 in2: 5 tolerance was: 0
volk_32fc_index_max_32u: fail on arch a_avx2
offset 0 in1: 0 in2: 5 tolerance was: 0
volk_32fc_index_max_32u: fail on arch u_avx2
```

Only those two index tests have AVX2 kernels. And only `volk_32fc_index_max_32` fails in some cases. I couldn't make it fail for any other vector size except 7. The result seems to be non-deterministic. That makes it even harder to track down the root cause.

argilo, 2020-02-24:
I expect it will fail every time if you pass in a vector where the last element is the largest.

michaelld, 2020-02-24:
@argilo can you tell me/us how to reliable get these failures, so that we can try to track down the issue & fix it? Thx!

argilo, 2020-02-24:
```
$ for i in {1..20}; do apps/volk_profile -v 7 -i 1 -R fc_index | grep -B 1 fail; done
offset 0 in1: 2 in2: 4 tolerance was: 0
volk_32fc_index_max_32u: fail on arch a_avx2
offset 0 in1: 2 in2: 4 tolerance was: 0
volk_32fc_index_max_32u: fail on arch u_avx2
offset 0 in1: 6 in2: 4 tolerance was: 0
volk_32fc_index_max_16u: fail on arch a_avx2
offset 0 in1: 6 in2: 4 tolerance was: 0
volk_32fc_index_max_16u: fail on arch u_avx2
offset 0 in1: 1 in2: 5 tolerance was: 0
volk_32fc_index_max_32u: fail on arch a_avx2
offset 0 in1: 1 in2: 5 tolerance was: 0
volk_32fc_index_max_32u: fail on arch u_avx2
offset 0 in1: 6 in2: 4 tolerance was: 0
volk_32fc_index_max_32u: fail on arch a_avx2
offset 0 in1: 6 in2: 4 tolerance was: 0
volk_32fc_index_max_32u: fail on arch u_avx2
offset 0 in1: 3 in2: 5 tolerance was: 0
volk_32fc_index_max_32u: fail on arch a_avx2
offset 0 in1: 3 in2: 5 tolerance was: 0
volk_32fc_index_max_32u: fail on arch u_avx2
offset 0 in1: 6 in2: 4 tolerance was: 0
volk_32fc_index_max_32u: fail on arch a_avx2
offset 0 in1: 6 in2: 4 tolerance was: 0
volk_32fc_index_max_32u: fail on arch u_avx2
offset 0 in1: 2 in2: 4 tolerance was: 0
volk_32fc_index_max_32u: fail on arch a_avx2
offset 0 in1: 2 in2: 4 tolerance was: 0
volk_32fc_index_max_32u: fail on arch u_avx2
offset 0 in1: 2 in2: 5 tolerance was: 0
volk_32fc_index_max_32u: fail on arch a_avx2
offset 0 in1: 2 in2: 5 tolerance was: 0
volk_32fc_index_max_32u: fail on arch u_avx2
offset 0 in1: 2 in2: 4 tolerance was: 0
volk_32fc_index_max_32u: fail on arch a_avx2
offset 0 in1: 2 in2: 4 tolerance was: 0
volk_32fc_index_max_32u: fail on arch u_avx2
offset 0 in1: 6 in2: 4 tolerance was: 0
volk_32fc_index_max_32u: fail on arch a_avx2
offset 0 in1: 6 in2: 4 tolerance was: 0
volk_32fc_index_max_32u: fail on arch u_avx2
offset 0 in1: 6 in2: 4 tolerance was: 0
volk_32fc_index_max_32u: fail on arch a_avx2
offset 0 in1: 6 in2: 4 tolerance was: 0
volk_32fc_index_max_32u: fail on arch u_avx2
offset 0 in1: 0 in2: 5 tolerance was: 0
volk_32fc_index_max_32u: fail on arch a_avx2
offset 0 in1: 0 in2: 5 tolerance was: 0
volk_32fc_index_max_32u: fail on arch u_avx2
```

michaelld, 2020-02-24:
nice ... works like a charm!

argilo, 2020-02-24:
Here's an example of one of the broken kernels:

https://github.com/gnuradio/volk/blob/aae9fed1cd15f0f1695d5adad04c65cf8bbd3cc0/kernels/volk/volk_32fc_index_max_32u.h#L109-L182

It first reads in elements 8 at a time. After that, between 0 and 7 elements will be left over. If there are four or more, it reads those, leaving between 0 and 3. Then if there are two or more, it reads those, leaving 0 or 1. And that's where things stop.

argilo, 2020-02-24:
I suspect this bug crept in because AVX2 registers are twice as big as SSE3 registers, and the SSE3 code was adapted to AVX2 with minimal modification. SSE3 processes in chunks of 4, 2, then 1 and these doubled to 8, 4, 2 (but no 1!).

argilo, 2020-02-26:
Actually, there must be more to it than that, since failures occur for any vector length that's not a multiple of 8.

closed 2020-03-22 by jdemel