gnuradio/volk #369: Build results vary from build machine CPU

bmwiedemann, 2020-03-26:
Originally filed at https://bugzilla.opensuse.org/show_bug.cgi?id=1152001

While working on reproducible builds for openSUSE, I found that
our volk-2.2.1 package varied depending on build machine CPU type.

In our build environment, I can see that some files are only created on newer CPUs:

```diff
+++ /home/abuild/rpmbuild/BUILD/volk-2.2.1/build/CMakeFiles/CMakeRuleHashes.txt	2035-04-29 10:36:47.097658418 +0000
@@ -30,10 +30,6 @@
 10ab5f8cd40dc80f123094a691e3846e lib/volk_8i_convert_16i_a_orc_impl.c
 7fed82588c050353f4725686c4c4d55c lib/volk_8i_s32f_convert_32f_a_orc_impl.c
 ab188bfb1a0db04e8ac5dcdd16bd45ca lib/volk_cpu.c
-f8783a9360e4177fad6877c9981cf5b3 lib/volk_machine_avx2_64_mmx_orc.c
-0156eb25d341afbcd7da28bffb2211cf lib/volk_machine_avx512cd_64_mmx_orc.c
-010fd1209e2dead02ca1286493e4b93d lib/volk_machine_avx512f_64_mmx_orc.c
-35a4ba48334a08077b745897eee21beb lib/volk_machine_avx_64_mmx_orc.c
 48ce9369600175205bc1ae3d1398fd8e lib/volk_machine_generic_orc.c
 ea76d4812482f5afdf1b85fd0515cd3d lib/volk_machine_sse2_64_mmx_orc.c
```

Here is a partial diff between -cpu host and -cpu kvm64 builds
on a avx2 capable host

```diff
  -- Performing Test have_mavx512f - Success
  -- Performing Test have_mavx512cd
  -- Performing Test have_mavx512cd - Success
- -- CPU missing xgetbv, Overruled arch avx
- -- Available architectures: generic;64;3dnow;abm;popcount;mmx;fma;sse;sse2;orc;norc;sse3;ssse3;sse4_a;sse4_1;sse4_2;avx2;avx512f;avx512cd
+ -- Available architectures: generic;64;3dnow;abm;popcount;mmx;fma;sse;sse2;orc;norc;sse3;ssse3;sse4_a;sse4_1;sse4_2;avx;avx2;avx512f;avx512cd
+ -- Available machines: generic_orc;sse2_64_mmx_orc;sse3_64_mmx_orc;ssse3_64_mmx_orc;sse4_a_64_mmx_orc;sse4_1_64_mmx_orc;sse4_2_64_mmx_orc;avx_64_mmx_orc;avx2_64_mmx_orc;avx512f_64_mmx_orc;avx512cd_64_mmx_orc
+ -- BUILD INFO ::: avx_64_mmx_orc ::: GNU ::: -O2 -g -DNDEBUG -O2 -Wall -D_FORTIFY_SOURCE=2 -fstack-protector-strong -funwind-tables -fasynchronous-unwind-tables -fstack-clash-protection -Werror=return-type -flto=auto -DNDEBUG -Wall -m64 -mmmx -msse -msse2 -msse3 -mssse3 -msse4.1 -msse4.2 -mpopcnt -mavx
```

jdemel, 2020-03-27:
I assume this issue is related to #311 and #348 .

VOLK should build for all x86 versions and then just select SSE/AVX etc during runtime.

bmwiedemann, 2020-03-29:
It shoud, but then it does disable avx in
https://github.com/gnuradio/volk/blob/master/lib/CMakeLists.txt#L162
if the build machine CPU does not have avx.

michaelld, 2020-03-29:
We added this configure execution check a long time ago because Volk didn't have the runtime capable of failing on a compiled kernel ... I mean, why enable a kernel if it won't execute on the target CPU?

I agree that it would be better to allow any compilable kernel to be compiled, then do runtime detection and disabling of kernels based on the target CPU. I don't honestly know that Volk can do this yet.

PR #370 (2020-04-01): Drop compile-time CPU detection
Drop compile-time CPU detection
So that we can generate AVX binaries on non-AVX machines.

See https://reproducible-builds.org/ for why this matters.

Fixes #369

This PR was done while working on reproducible builds for openSUSE.
commit de93c273 (2020-04-01): Drop compile-time CPU detection

So that we can generate AVX binaries on non-AVX machines.

See https://reproducible-builds.org/ for why this matters.

Fixes #369
closed 2020-04-01 by michaelld