gnuradio/volk #375: CI fails on g++5 Ubuntu 16.04 missing isinf

jdemel, 2020-04-10:
I observed this for all PRs since the updated Travis CI was merged. When `log2f_non_ieee` was added, we introduced a bug for Ubuntu 16.04 with G++ 5. Apparently it builds for gcc-5 for the C part but things fail on the C++ side.

On the one hand this a bug specific to Ubuntu 16.04/g++5 on the other hand, this is an LTS release. Arguably Ubuntu 16.04 is very old and you shouldn't use old compilers/distros with new libraries but we need to make a decision for this one how to handle this error.

1. Just drop the Ubuntu 16.04/g++5 test
2. Change CI to use Ubuntu 18.04/g++5 OR Ubuntu 14.04/g++5 (previous mode)
3. Fix this as a conditional bug with all kinds of possible side-effects.

[A similar bug report](https://github.com/facebookarchive/caffe2/issues/1633)

```
Scanning dependencies of target volk_test_all
[ 89%] Building CXX object lib/CMakeFiles/volk_test_all.dir/testqa.cc.o
In file included from /home/travis/build/gnuradio/volk/build/include/volk/volk.h:31:0,
                 from /home/travis/build/gnuradio/volk/lib/qa_utils.h:5,
                 from /home/travis/build/gnuradio/volk/lib/kernel_tests.h:1,
                 from /home/travis/build/gnuradio/volk/lib/testqa.cc:31:

/home/travis/build/gnuradio/volk/include/volk/volk_common.h: In function ‘float log2f_non_ieee(float)’:
/home/travis/build/gnuradio/volk/include/volk/volk_common.h:153:24: error: ‘isinf’ was not declared in this scope
     return isinf(result) ? copysignf(127.0f, result) : result;
                        ^

/home/travis/build/gnuradio/volk/include/volk/volk_common.h:153:24: note: suggested alternative:
In file included from /usr/include/c++/5/complex:44:0,
                 from /home/travis/build/gnuradio/volk/include/volk/volk_complex.h:23,
                 from /home/travis/build/gnuradio/volk/build/include/volk/volk_typedefs.h:30,
                 from /home/travis/build/gnuradio/volk/build/include/volk/volk.h:29,
                 from /home/travis/build/gnuradio/volk/lib/qa_utils.h:5,
                 from /home/travis/build/gnuradio/volk/lib/kernel_tests.h:1,
                 from /home/travis/build/gnuradio/volk/lib/testqa.cc:31:

/usr/include/c++/5/cmath:626:5: note:   ‘std::isinf’
     isinf(_Tp __x)
     ^

lib/CMakeFiles/volk_test_all.dir/build.make:62: recipe for target 'lib/CMakeFiles/volk_test_all.dir/testqa.cc.o' failed
make[2]: *** [lib/CMakeFiles/volk_test_all.dir/testqa.cc.o] Error 1
CMakeFiles/Makefile2:200: recipe for target 'lib/CMakeFiles/volk_test_all.dir/all' failed
make[1]: *** [lib/CMakeFiles/volk_test_all.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
```

jdemel, 2020-04-18:
This bug reveals itself because of #364 . @ghostop14 Do you have a suggestion how to handle this?

ghostop14, 2020-04-18:
Hi @jdemel, the only change in 364 related to that function was it was moved out of volk_32f_log2_32f.h into volk_common.h.  I added the #include <math.h> there to cover what looked like its dependencies.  Possible on Ubuntu 16 it needs one of the other #includes that are in volk_32f_log2_32f.h.  Possibly "#include <inttypes.h>"?  I don't have an Ubuntu 16 here to test on.  One option may be to just move log2f_non_ieee from volk_common.h back to volk_32f_log2_32f.h.  Shouldn't need to roll back any of the rest of 364.

closed 2020-05-03 by michaelld