gnuradio/volk #377: VOLK VolkConfigVersion.cmake doesn't seem to export version properly

marcusmueller, 2020-04-12:
When doing a minimal CMakeLists.txt:

```cmake
cmake_minimum_required(VERSION 3.8)
find_package(Volk)
message(STATUS "  Found VOLK:")
message(STATUS "  * Version: ||${VOLK_VERSION}||")
```

you get 

```
-- Found VOLK: Volk::volk  
--   Found VOLK:
--   * Version: ||||
```

This of course causes a lot of trouble in the GNU Radio build process.

jdemel, 2020-04-13:
I wonder what could cause your issue. My first guess would be that your `VOLK_VERSION` variable should really be `Volk_VERSION` according to the [CMake find_package version docs](https://cmake.org/cmake/help/latest/command/find_package.html#version-selection) because variables are case-sensitive. Or did I miss anything in the docs that states `<PackageName>` is converted to uppercase for variables?

Could you provide a trace or debug output of your minimal example? 

[CMakeLists.txt:299ff](https://github.com/gnuradio/volk/blob/dca9988935a95aa7dda38a5c6407e15af520c5f0/CMakeLists.txt#L299) are the lines where we configure our CMake VOLK files.
[CMake 3.8 docs](https://cmake.org/cmake/help/v3.8/module/CMakePackageConfigHelpers.html) state that there are better functions for what we need. Specifically, we should use `configure_package_config_file` and `write_basic_package_version_file`. They are available with our minimum CMake version requirement (3.8), so these lines should be updated to make things more robust.

jdemel, 2020-04-24:
So, this is an updated CMake MWE
```
cmake_minimum_required(VERSION 3.8)
project(blablabla CXX C)

find_package(Volk REQUIRED)
message(STATUS "  Found VOLK:")
message(STATUS "  * Version: ||${Volk_VERSION}||")
message(STATUS "  * Volk_FOUND: |${Volk_FOUND}|")
message(STATUS "  * VOLK_FOUND: |${VOLK_FOUND}|")
```
with the expected output:
```
-- Found VOLK: Volk::volk  
--   Found VOLK:
--   * Version: ||2.2.1||
--   * Volk_FOUND: |1|
--   * VOLK_FOUND: |TRUE|
```
The difference is that `${Volk_VERSION}` is used instead of `${VOLK_VERSION}`. Thus, you can use all this info in your project but you need to be very careful with your variable names. CMake docs basically imply this behavior but it is really implicit.

@marcusmueller I assume this issue is solved and we can close it?

jdemel, 2020-05-09:
I assume the we could solve this issue. Thus, I close it now.

As a reminder: CMake Module names are case sensitive! So is e.g. `Volk_VERSION`.

closed 2020-05-09 by jdemel