gnuradio/volk #380: howintsalled volk

jn9531, 2020-05-09:


jdemel, 2020-05-09:
Welcome to VOLK!

VOLK 1.4 is a pretty old release. We added a bunch of new features since then. 

Since GNU Radio does check VOLK version, you want to install GNU Radio master branch. In that case, I recommend to install VOLK just like GNU Radio.

To help you with more specific advice, we need some more info.
- Which OS/distro do you use? (Ubuntu 20.04? Ubuntu 18.04? Fedora? Debian? MacOS?)
- How did you install VOLK 1.4? (via a package manager?)
- How do you try to install GNU Radio? (via PyBOMBS? GNU Radio PPA? ...)

jdemel, 2020-05-29:
@jn9531 I assume you could solve your issue. I close this issue now. If this issue persists, feel free to reopen it. Also, it would be super helpful if you'd provide more info in that case.

closed 2020-05-29 by jdemel