gnuradio/volk #382: 61 - qa_volk_32f_x2_dot_prod_32f (Failed)

ZeroChaos-, 2020-05-29:
```
ctest -j 12 --test-load 12
Test project /var/tmp/portage/sci-libs/volk-2.3.0/work/volk-2.3.0_build
        Start   1: qa_volk_16i_32fc_dot_prod_32fc
        Start   2: qa_volk_16i_branch_4_state_8
        Start   3: qa_volk_16i_convert_8i
        Start   4: qa_volk_16i_max_star_16i
        Start   5: qa_volk_16i_max_star_horizontal_16i
        Start   6: qa_volk_16i_permute_and_scalar_add
        Start   7: qa_volk_16i_s32f_convert_32f
  1/134 Test   #2: qa_volk_16i_branch_4_state_8 .........................   Passed    0.01 sec
        Start   8: qa_volk_16i_x4_quad_max_star_16i
        Start   9: qa_volk_16i_x5_add_quad_16i_x4
        Start  10: qa_volk_16ic_convert_32fc
        Start  11: qa_volk_16ic_deinterleave_16i_x2
        Start  12: qa_volk_16ic_deinterleave_real_16i
        Start  13: qa_volk_16ic_deinterleave_real_8i
  2/134 Test   #3: qa_volk_16i_convert_8i ...............................   Passed    0.03 sec
        Start  14: qa_volk_16ic_magnitude_16i
  3/134 Test   #4: qa_volk_16i_max_star_16i .............................   Passed    0.03 sec
        Start  15: qa_volk_16ic_s32f_deinterleave_32f_x2
  4/134 Test   #5: qa_volk_16i_max_star_horizontal_16i ..................   Passed    0.04 sec
        Start  16: qa_volk_16ic_s32f_deinterleave_real_32f
  5/134 Test   #6: qa_volk_16i_permute_and_scalar_add ...................   Passed    0.04 sec
        Start  17: qa_volk_16ic_s32f_magnitude_32f
  6/134 Test   #8: qa_volk_16i_x4_quad_max_star_16i .....................   Passed    0.04 sec
        Start  18: qa_volk_16ic_x2_dot_prod_16ic
  7/134 Test   #9: qa_volk_16i_x5_add_quad_16i_x4 .......................   Passed    0.05 sec
        Start  19: qa_volk_16ic_x2_multiply_16ic
  8/134 Test   #1: qa_volk_16i_32fc_dot_prod_32fc .......................   Passed    0.07 sec
        Start  20: qa_volk_16u_byteswap
  9/134 Test   #7: qa_volk_16i_s32f_convert_32f .........................   Passed    0.06 sec
        Start  21: qa_volk_16u_byteswappuppet_16u
 10/134 Test  #11: qa_volk_16ic_deinterleave_16i_x2 .....................   Passed    0.04 sec
        Start  22: qa_volk_32f_64f_add_64f
 11/134 Test  #12: qa_volk_16ic_deinterleave_real_16i ...................   Passed    0.04 sec
        Start  23: qa_volk_32f_64f_multiply_64f
 12/134 Test  #13: qa_volk_16ic_deinterleave_real_8i ....................   Passed    0.04 sec
        Start  24: qa_volk_32f_8u_polarbutterfly_32f
 13/134 Test  #10: qa_volk_16ic_convert_32fc ............................   Passed    0.07 sec
        Start  25: qa_volk_32f_8u_polarbutterflypuppet_32f
 14/134 Test  #14: qa_volk_16ic_magnitude_16i ...........................   Passed    0.05 sec
        Start  26: qa_volk_32f_accumulator_s32f
 15/134 Test  #15: qa_volk_16ic_s32f_deinterleave_32f_x2 ................   Passed    0.05 sec
        Start  27: qa_volk_32f_acos_32f
 16/134 Test  #16: qa_volk_16ic_s32f_deinterleave_real_32f ..............   Passed    0.05 sec
        Start  28: qa_volk_32f_asin_32f
 17/134 Test  #20: qa_volk_16u_byteswap .................................   Passed    0.02 sec
        Start  29: qa_volk_32f_atan_32f
 18/134 Test  #17: qa_volk_16ic_s32f_magnitude_32f ......................   Passed    0.06 sec
        Start  30: qa_volk_32f_binary_slicer_32i
 19/134 Test  #18: qa_volk_16ic_x2_dot_prod_16ic ........................   Passed    0.06 sec
        Start  31: qa_volk_32f_binary_slicer_8i
 20/134 Test  #19: qa_volk_16ic_x2_multiply_16ic ........................   Passed    0.06 sec
        Start  32: qa_volk_32f_convert_64f
 21/134 Test  #21: qa_volk_16u_byteswappuppet_16u .......................   Passed    0.05 sec
        Start  33: qa_volk_32f_cos_32f
 22/134 Test  #24: qa_volk_32f_8u_polarbutterfly_32f ....................   Passed    0.04 sec
        Start  34: qa_volk_32f_exp_32f
 23/134 Test  #22: qa_volk_32f_64f_add_64f ..............................   Passed    0.06 sec
        Start  35: qa_volk_32f_expfast_32f
 24/134 Test  #23: qa_volk_32f_64f_multiply_64f .........................   Passed    0.05 sec
        Start  36: qa_volk_32f_index_max_16u
 25/134 Test  #26: qa_volk_32f_accumulator_s32f .........................   Passed    0.05 sec
        Start  37: qa_volk_32f_index_max_32u
 26/134 Test  #27: qa_volk_32f_acos_32f .................................   Passed    0.06 sec
        Start  38: qa_volk_32f_invsqrt_32f
 27/134 Test  #28: qa_volk_32f_asin_32f .................................   Passed    0.05 sec
        Start  39: qa_volk_32f_log2_32f
 28/134 Test  #29: qa_volk_32f_atan_32f .................................   Passed    0.05 sec
[... 185 lines cut ...]
121/134 Test #121: qa_volk_8i_convert_16i ...............................   Passed    0.06 sec
        Start 133: qa_volk_8u_x3_encodepolarpuppet_8u
122/134 Test #122: qa_volk_8i_s32f_convert_32f ..........................   Passed    0.06 sec
        Start 134: qa_volk_8u_x4_conv_k7_r2_8u
123/134 Test #123: qa_volk_8ic_deinterleave_16i_x2 ......................   Passed    0.06 sec
124/134 Test #124: qa_volk_8ic_deinterleave_real_16i ....................   Passed    0.05 sec
125/134 Test #125: qa_volk_8ic_deinterleave_real_8i .....................   Passed    0.05 sec
126/134 Test #127: qa_volk_8ic_s32f_deinterleave_real_32f ...............   Passed    0.04 sec
127/134 Test #126: qa_volk_8ic_s32f_deinterleave_32f_x2 .................   Passed    0.04 sec
128/134 Test #131: qa_volk_8u_x2_encodeframepolar_8u ....................   Passed    0.02 sec
129/134 Test #128: qa_volk_8ic_x2_multiply_conjugate_16ic ...............   Passed    0.03 sec
130/134 Test #132: qa_volk_8u_x3_encodepolar_8u_x2 ......................   Passed    0.02 sec
131/134 Test #134: qa_volk_8u_x4_conv_k7_r2_8u ..........................   Passed    0.01 sec
132/134 Test #129: qa_volk_8ic_x2_s32f_multiply_conjugate_32fc ..........   Passed    0.04 sec
133/134 Test #133: qa_volk_8u_x3_encodepolarpuppet_8u ...................   Passed    0.03 sec
134/134 Test #130: qa_volk_8u_conv_k7_r2puppet_8u .......................   Passed    0.05 sec

99% tests passed, 1 tests failed out of 134

Total Test time (real) =   0.80 sec

The following tests FAILED:
	 61 - qa_volk_32f_x2_dot_prod_32f (Failed)
Errors while running CTest
```

marcusmueller, 2020-05-29:
ah, wait, @ZeroChaos-, maybe telling us about the exact CPU you're running on would help - also, can you run `ctest -VV -R x2_dot_prod` and tell us what implementation specifically fails?

ZeroChaos-, 2020-06-02:
I'm running this in the gentoo package manager so if I messed up somehow, sorry.  Here's what the output looks like
```
UpdateCTestConfiguration  from :/var/tmp/portage/sci-libs/volk-2.3.0/work/volk-2.3.0_build/DartConfiguration.tcl
UpdateCTestConfiguration  from :/var/tmp/portage/sci-libs/volk-2.3.0/work/volk-2.3.0_build/DartConfiguration.tcl
Test project /var/tmp/portage/sci-libs/volk-2.3.0/work/volk-2.3.0_build
Constructing a list of tests
Done constructing a list of tests
Updating test list for fixtures
Added 0 tests to meet fixture requirements
Checking test dependency graph...
Checking test dependency graph end
test 18
    Start 18: qa_volk_16ic_x2_dot_prod_16ic

18: Test command: /bin/sh "/var/tmp/portage/sci-libs/volk-2.3.0/work/volk-2.3.0_build/lib/volk_16ic_x2_dot_prod_16ic_test.sh" "/var/tmp/portage/sci-libs/volk-2.3.0/work/volk-2.3.0_build/lib"
18: Test timeout computed to be: 10000000
18: RUN_VOLK_TESTS: volk_16ic_x2_dot_prod_16ic(131071,1)
18: generic completed in 1.12208 ms
18: a_sse2 completed in 0.189971 ms
18: u_sse2 completed in 0.205457 ms
18: u_axv2 completed in 0.184869 ms
18: a_axv2 completed in 0.14008 ms
18: Best aligned arch: a_axv2
18: Best unaligned arch: u_axv2
1/4 Test #18: qa_volk_16ic_x2_dot_prod_16ic ....   Passed    0.03 sec
test 60
    Start 60: qa_volk_32f_x2_dot_prod_16i

60: Test command: /bin/sh "/var/tmp/portage/sci-libs/volk-2.3.0/work/volk-2.3.0_build/lib/volk_32f_x2_dot_prod_16i_test.sh" "/var/tmp/portage/sci-libs/volk-2.3.0/work/volk-2.3.0_build/lib"
60: Test timeout computed to be: 10000000
60: RUN_VOLK_TESTS: volk_32f_x2_dot_prod_16i(131071,1)
60: generic completed in 0.15099 ms
60: a_sse completed in 0.062778 ms
60: a_avx2_fma completed in 0.066278 ms
60: a_avx completed in 0.075649 ms
60: u_sse completed in 0.058729 ms
60: u_avx2_fma completed in 0.053631 ms
60: u_avx completed in 0.035849 ms
60: Best aligned arch: u_avx
60: Best unaligned arch: u_avx
2/4 Test #60: qa_volk_32f_x2_dot_prod_16i ......   Passed    0.03 sec
test 61
    Start 61: qa_volk_32f_x2_dot_prod_32f

61: Test command: /bin/sh "/var/tmp/portage/sci-libs/volk-2.3.0/work/volk-2.3.0_build/lib/volk_32f_x2_dot_prod_32f_test.sh" "/var/tmp/portage/sci-libs/volk-2.3.0/work/volk-2.3.0_build/lib"
61: Test timeout computed to be: 10000000
61: RUN_VOLK_TESTS: volk_32f_x2_dot_prod_32f(131071,1)
61: generic completed in 0.152256 ms
61: u_sse completed in 0.091711 ms
61: u_sse3 completed in 0.07584 ms
61: u_sse4_1 completed in 0.082822 ms
61: u_avx completed in 0.063481 ms
61: u_avx2_fma completed in 0.084034 ms
61: a_generic completed in 0.215894 ms
61: a_sse completed in 0.1544 ms
61: a_sse3 completed in 0.130655 ms
61: a_sse4_1 completed in 0.192361 ms
61: a_avx completed in 0.15293 ms
61: a_avx2_fma completed in 0.149334 ms
61: offset 0 in1: -209.176 in2: -144.413 tolerance was: 0.01
61: volk_32f_x2_dot_prod_32f: fail on arch u_avx2_fma
61: offset 0 in1: -209.176 in2: -144.413 tolerance was: 0.01
61: volk_32f_x2_dot_prod_32f: fail on arch a_avx2_fma
61: Best aligned arch: u_avx
61: Best unaligned arch: u_avx
3/4 Test #61: qa_volk_32f_x2_dot_prod_32f ......***Failed    0.04 sec
test 98
    Start 98: qa_volk_32fc_x2_dot_prod_32fc

98: Test command: /bin/sh "/var/tmp/portage/sci-libs/volk-2.3.0/work/volk-2.3.0_build/lib/volk_32fc_x2_dot_prod_32fc_test.sh" "/var/tmp/portage/sci-libs/volk-2.3.0/work/volk-2.3.0_build/lib"
98: Test timeout computed to be: 10000000
98: RUN_VOLK_TESTS: volk_32fc_x2_dot_prod_32fc(131071,1)
98: generic completed in 0.157809 ms
98: u_sse_64 completed in 0.13014 ms
98: u_sse3 completed in 0.184052 ms
98: u_sse4_1 completed in 0.148972 ms
98: u_avx completed in 0.108756 ms
98: u_avx_fma completed in 0.109926 ms
98: a_generic completed in 0.147653 ms
98: a_sse_64 completed in 0.139198 ms
98: a_sse3 completed in 0.13641 ms
98: a_sse4_1 completed in 0.147799 ms
98: a_avx completed in 0.113286 ms
98: a_avx_fma completed in 0.098955 ms
98: Best aligned arch: a_avx_fma
98: Best unaligned arch: u_avx
4/4 Test #98: qa_volk_32fc_x2_dot_prod_32fc ....   Passed    0.08 sec

The following tests passed:
	qa_volk_16ic_x2_dot_prod_16ic
	qa_volk_32f_x2_dot_prod_16i
	qa_volk_32fc_x2_dot_prod_32fc

75% tests passed, 1 tests failed out of 4

Total Test time (real) =   0.18 sec

The following tests FAILED:
	 61 - qa_volk_32f_x2_dot_prod_32f (Failed)
Errors while running CTest
```

Here's my cpu info
```
processor	: 0
vendor_id	: GenuineIntel
cpu family	: 6
model		: 158
model name	: Intel(R) Core(TM) i7-8750H CPU @ 2.20GHz
stepping	: 10
microcode	: 0xd2
cpu MHz		: 3399.999
cache size	: 9216 KB
physical id	: 0
siblings	: 12
core id		: 0
cpu cores	: 6
apicid		: 0
initial apicid	: 0
fpu		: yes
fpu_exception	: yes
cpuid level	: 22
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc art arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault epb invpcid_single pti ssbd ibrs ibpb stibp tpr_shadow vnmi flexpriority ept vpid ept_ad fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid mpx rdseed adx smap clflushopt intel_pt xsaveopt xsavec xgetbv1 xsaves dtherm ida arat pln pts hwp hwp_notify hwp_act_window hwp_epp md_clear flush_l1d
bugs		: cpu_meltdown spectre_v1 spectre_v2 spec_store_bypass l1tf mds swapgs itlb_multihit
bogomips	: 4399.99
clflush size	: 64
cache_alignment	: 64
address sizes	: 39 bits physical, 48 bits virtual
power management:
```

jdemel, 2020-06-02:
```
61: a_avx2_fma completed in 0.149334 ms
61: offset 0 in1: -209.176 in2: -144.413 tolerance was: 0.01
61: volk_32f_x2_dot_prod_32f: fail on arch u_avx2_fma
61: offset 0 in1: -209.176 in2: -144.413 tolerance was: 0.01
61: volk_32f_x2_dot_prod_32f: fail on arch a_avx2_fma
```
So now we see `avx2_fma` is broken. Well, at least on your CPU where it should work. So we need to dig deeper.
Just as a reference `ctest -V -R 32f_x2_dot_prod_32f` should be the minimal test.

Could you comment out these 2 lines and recompile and rerun your test?
https://github.com/gnuradio/volk/blob/e564299446a974ee48e30ffc36ee9d443633b9bc/kernels/volk/volk_32f_x2_dot_prod_32f.h#L419
https://github.com/gnuradio/volk/blob/e564299446a974ee48e30ffc36ee9d443633b9bc/kernels/volk/volk_32f_x2_dot_prod_32f.h#L828

I suppose they zero out half of the accumulated result. Otherwise the following lines would only need to sum up 4 values instead of 8.
 
We do have a test with Intel SDE to catch all kernels and the avx2_fma kernel is also executed in other tests. I wonder why we zero the upper half of this register and why this isn't an issue anywhere else.

The SDE test broke because it can't download the SDE anymore. We need to look into this.

ZeroChaos-, 2020-06-03:
removing the two instances of that line, from that file, causes all tests to pass.

jdemel, 2020-06-03:
I'm happy to hear that the tests pass now.

Still, we need to figure out why these tests pass on most systems. 

I had another look at these lines. They should not affect the kernel at all. 
https://github.com/gnuradio/volk/blob/e564299446a974ee48e30ffc36ee9d443633b9bc/kernels/volk/volk_32f_x2_dot_prod_32f.h#L416-L423
In L416, we declare a new array, then in L417, we store the computed result in this array. Thus, the call to `_mm256_zeroupper` in L419 should not affect our result at all because it does only zero out the upper half of our AVX register. Still, the test error suggests that this affects the result. 

The whole call to `_mm256_zeroupper` is confusing. It does affect the state of all your AVX registers. It's probably not local to the function it is called in. Yet, it is in there because otherwise we might see a performance penalty in case we mix SSE and AVX instructions. e.g. if we call a SSE kernel after an AVX kernel there might be a delay.

@ZeroChaos- which compiler do you use exactly? And which flags are set? Maybe the compiler does some optimization that affects the result here.

ZeroChaos-, 2020-06-03:
This enough info?
```
-- Check size of void*[8]
-- Check size of void*[8] - done
-- CPU width is 64 bits, Overruled arch 32
-- Available architectures: generic;64;3dnow;abm;popcount;mmx;fma;sse;sse2;orc;norc;sse3;ssse3;sse4_a;sse4_1;sse4_2;avx;avx2;avx512f;avx512cd
-- Available machines: generic_orc;sse2_64_mmx_orc;sse3_64_mmx_orc;ssse3_64_mmx_orc;sse4_a_64_mmx_orc;sse4_1_64_mmx_orc;sse4_2_64_mmx_orc;avx_64_mmx_orc;avx2_64_mmx_orc;avx512f_64_mmx_orc;avx512cd_64_mmx_orc
-- BUILD TYPE = RELEASE
-- Base cflags =  -Os -march=native -mtune=native -pipe -frecord-gcc-switches -fcx-limited-range -Wall
-- BUILD INFO ::: generic_orc ::: GNU :::  -Os -march=native -mtune=native -pipe -frecord-gcc-switches -fcx-limited-range -Wall 
-- BUILD INFO ::: sse2_64_mmx_orc ::: GNU :::  -Os -march=native -mtune=native -pipe -frecord-gcc-switches -fcx-limited-range -Wall -m64 -mmmx -msse -msse2
-- BUILD INFO ::: sse3_64_mmx_orc ::: GNU :::  -Os -march=native -mtune=native -pipe -frecord-gcc-switches -fcx-limited-range -Wall -m64 -mmmx -msse -msse2 -msse3
-- BUILD INFO ::: ssse3_64_mmx_orc ::: GNU :::  -Os -march=native -mtune=native -pipe -frecord-gcc-switches -fcx-limited-range -Wall -m64 -mmmx -msse -msse2 -msse3 -mssse3
-- BUILD INFO ::: sse4_a_64_mmx_orc ::: GNU :::  -Os -march=native -mtune=native -pipe -frecord-gcc-switches -fcx-limited-range -Wall -m64 -mmmx -msse -msse2 -msse3 -msse4a -mpopcnt
-- BUILD INFO ::: sse4_1_64_mmx_orc ::: GNU :::  -Os -march=native -mtune=native -pipe -frecord-gcc-switches -fcx-limited-range -Wall -m64 -mmmx -msse -msse2 -msse3 -mssse3 -msse4.1
-- BUILD INFO ::: sse4_2_64_mmx_orc ::: GNU :::  -Os -march=native -mtune=native -pipe -frecord-gcc-switches -fcx-limited-range -Wall -m64 -mmmx -msse -msse2 -msse3 -mssse3 -msse4.1 -msse4.2 -mpopcnt
-- BUILD INFO ::: avx_64_mmx_orc ::: GNU :::  -Os -march=native -mtune=native -pipe -frecord-gcc-switches -fcx-limited-range -Wall -m64 -mmmx -msse -msse2 -msse3 -mssse3 -msse4.1 -msse4.2 -mpopcnt -mavx
-- BUILD INFO ::: avx2_64_mmx_orc ::: GNU :::  -Os -march=native -mtune=native -pipe -frecord-gcc-switches -fcx-limited-range -Wall -m64 -mmmx -msse -msse2 -msse3 -mssse3 -msse4.1 -msse4.2 -mpopcnt -mavx -mfma -mavx2
-- BUILD INFO ::: avx512f_64_mmx_orc ::: GNU :::  -Os -march=native -mtune=native -pipe -frecord-gcc-switches -fcx-limited-range -Wall -m64 -mmmx -msse -msse2 -msse3 -mssse3 -msse4.1 -msse4.2 -mpopcnt -mavx -mfma -mavx2 -mavx512f
-- BUILD INFO ::: avx512cd_64_mmx_orc ::: GNU :::  -Os -march=native -mtune=native -pipe -frecord-gcc-switches -fcx-limited-range -Wall -m64 -mmmx -msse -msse2 -msse3 -mssse3 -msse4.1 -msse4.2 -mpopcnt -mavx -mfma -mavx2 -mavx512f -mavx512cd
-- Compiler Version: x86_64-pc-linux-gnu-gcc (Gentoo Hardened 9.3.0 p2) 9.3.0
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/x86_64-pc-linux-gnu-gcc
-- c flags: -Os -march=native -mtune=native -pipe -frecord-gcc-switches -fcx-limited-range -Wall;
-- asm flags: -Os -march=native -mtune=native -pipe -frecord-gcc-switches
-- c flags: -Os -march=native -mtune=native -pipe -frecord-gcc-switches -fcx-limited-range -Wall;
-- asm flags: -Os -march=native -mtune=native -pipe -frecord-gcc-switches
-- Loading version 2.3 into constants...
-- Using install prefix: /usr
-- <<< Gentoo configuration >>>
```

ZeroChaos-, 2020-06-03:
if that wasn't enough...
```
zero@theprophet ~ % gcc -dumpspecs
*asm:
%{m16|m32:--32}  %{m16|m32|mx32:;:--64}  %{mx32:--x32}  %{msse2avx:%{!mavx:-msse2avx}}

*asm_debug:
%{%:debug-level-gt(0):%{gstabs*:--gstabs}%{!gstabs*:%{g*:--gdwarf2}}} %{fdebug-prefix-map=*:--debug-prefix-map %*}

*asm_final:
%{gsplit-dwarf: 
       objcopy --extract-dwo 	 %{c:%{o*:%*}%{!o*:%b%O}}%{!c:%U%O} 	 %{c:%{o*:%:replace-extension(%{o*:%*} .dwo)}%{!o*:%b.dwo}}%{!c:%b.dwo} 
       objcopy --strip-dwo 	 %{c:%{o*:%*}%{!o*:%b%O}}%{!c:%U%O}     }

*asm_options:
%{-target-help:%:print-asm-header()} %{v} %{w:-W} %{I*}  %{gz|gz=zlib:--compress-debug-sections=zlib} %{gz=none:--compress-debug-sections=none} %{gz=zlib-gnu:--compress-debug-sections=zlib-gnu} %a %Y %{c:%W{o*}%{!o*:-o %w%b%O}}%{!c:-o %d%w%u%O}

*invoke_as:
%{!fwpa*:   %{fcompare-debug=*|fdump-final-insns=*:%:compare-debug-dump-opt()}   %{!S:-o %|.s |
 as %(asm_options) %m.s %A }  }

*cpp:
%{posix:-D_POSIX_SOURCE} %{pthread:-D_REENTRANT}

*cpp_options:
%(cpp_unique_options) %1 %{m*} %{std*&ansi&trigraphs} %{W*&pedantic*} %{w} %{f*} %{g*:%{%:debug-level-gt(0):%{g*} %{!fno-working-directory:-fworking-directory}}} %{O*} %{undef} %{save-temps*:-fpch-preprocess}

*cpp_debug_options:
%{d*}

*cpp_unique_options:
%{!Q:-quiet} %{nostdinc*} %{C} %{CC} %{v} %@{I*&F*} %{P} %I %{MD:-MD %{!o:%b.d}%{o*:%.d%*}} %{MMD:-MMD %{!o:%b.d}%{o*:%.d%*}} %{M} %{MM} %{MF*} %{MG} %{MP} %{MQ*} %{MT*} %{!E:%{!M:%{!MM:%{!MT:%{!MQ:%{MD|MMD:%{o*:-MQ %*}}}}}}} %{remap} %{g3|ggdb3|gstabs3|gxcoff3|gvms3:-dD} %{!iplugindir*:%{fplugin*:%:find-plugindir()}} %{H} %C %{D*&U*&A*} %{i*} %Z %i %{E|M|MM:%W{o*}}

*trad_capable_cpp:
cc1 -E %{traditional|traditional-cpp:-traditional-cpp}

*cc1:
%{!mandroid|tno-android-cc:%(cc1_cpu) %{profile:-p};:%(cc1_cpu) %{profile:-p} %{!mglibc:%{!muclibc:%{!mbionic: -mbionic}}} %{!fno-pic:%{!fno-PIC:%{!fpic:%{!fPIC: -fPIC}}}}}

*cc1_options:
%{pg:%{fomit-frame-pointer:%e-pg and -fomit-frame-pointer are incompatible}} %{!iplugindir*:%{fplugin*:%:find-plugindir()}} %1 %{!Q:-quiet} %{!dumpbase:-dumpbase %B} %{d*} %{m*} %{aux-info*} %{fcompare-debug-second:%:compare-debug-auxbase-opt(%b)}  %{!fcompare-debug-second:%{c|S:%{o*:-auxbase-strip %*}%{!o*:-auxbase %b}}}%{!c:%{!S:-auxbase %b}}  %{g*} %{O*} %{W*&pedantic*} %{w} %{std*&ansi&trigraphs} %{v:-version} %{pg:-p} %{p} %{f*} %{undef} %{Qn:-fno-ident} %{Qy:} %{-help:--help} %{-target-help:--target-help} %{-version:--version} %{-help=*:--help=%*} %{!fsyntax-only:%{S:%W{o*}%{!o*:-o %b.s}}} %{fsyntax-only:-o %j} %{-param*} %{nostdlib|nodefaultlibs|ffreestanding:-fno-stack-protector}  %{coverage:-fprofile-arcs -ftest-coverage} %{fprofile-arcs|fprofile-generate*|coverage:   %{!fprofile-update=single:     %{pthread:-fprofile-update=prefer-atomic}}}

*cc1plus:

*link_gcc_c_sequence:
%{static|static-pie:--start-group} %G %{!nolibc:%L}    %{static|static-pie:--end-group}%{!static:%{!static-pie:%G}}

*link_ssp:
%{fstack-protector|fstack-protector-all|fstack-protector-strong|fstack-protector-explicit:}

*endfile:
%{!mandroid|tno-android-ld:%{Ofast|ffast-math|funsafe-math-optimizations:crtfastmath.o%s}    %{mpc32:crtprec32.o%s}    %{mpc64:crtprec64.o%s}    %{mpc80:crtprec80.o%s} %{fvtable-verify=none:%s;      fvtable-verify=preinit:vtv_end_preinit.o%s;      fvtable-verify=std:vtv_end.o%s}    %{static:crtend.o%s;      shared|static-pie|!no-pie:crtendS.o%s;      :crtend.o%s} crtn.o%s ;:%{Ofast|ffast-math|funsafe-math-optimizations:crtfastmath.o%s}    %{mpc32:crtprec32.o%s}    %{mpc64:crtprec64.o%s}    %{mpc80:crtprec80.o%s} %{shared: crtend_so%O%s;: crtend_android%O%s}}

*link:
%{!static|static-pie:--eh-frame-hdr} %{!mandroid|tno-android-ld:%{m16|m32|mx32:;:-m elf_x86_64}                    %{m16|m32:-m elf_i386}                    %{mx32:-m elf32_x86_64}   %{shared:-shared}   %{!shared:     %{!static:       %{!static-pie: 	%{rdynamic:-export-dynamic} 	%{m16|m32:-dynamic-linker %{muclibc:/lib/ld-uClibc.so.0;:%{mbionic:/system/bin/linker;:%{mmusl:/lib/ld-musl-i386.so.1;:/lib/ld-linux.so.2}}}} 	%{m16|m32|mx32:;:-dynamic-linker %{muclibc:/lib/ld64-uClibc.so.0;:%{mbionic:/system/bin/linker64;:%{mmusl:/lib/ld-musl-x86_64.so.1;:/lib64/ld-linux-x86-64.so.2}}}} 	%{mx32:-dynamic-linker %{muclibc:/lib/ldx32-uClibc.so.0;:%{mbionic:/system/bin/linkerx32;:%{mmusl:/lib/ld-musl-x32.so.1;:/libx32/ld-linux-x32.so.2}}}}}}     %{static:-static} %{static-pie:-static -pie --no-dynamic-linker -z text}};:%{m16|m32|mx32:;:-m elf_x86_64}                    %{m16|m32:-m elf_i386}                    %{mx32:-m elf32_x86_64}   %{shared:-shared}   %{!shared:     %{!static:       %{!static-pie: 	%{rdynamic:-export-dynamic} 	%{m16|m32:-dynamic-linker %{muclibc:/lib/ld-uClibc.so.0;:%{mbionic:/system/bin/linker;:%{mmusl:/lib/ld-musl-i386.so.1;:/lib/ld-linux.so.2}}}} 	%{m16|m32|mx32:;:-dynamic-linker %{muclibc:/lib/ld64-uClibc.so.0;:%{mbionic:/system/bin/linker64;:%{mmusl:/lib/ld-musl-x86_64.so.1;:/lib64/ld-linux-x86-64.so.2}}}} 	%{mx32:-dynamic-linker %{muclibc:/lib/ldx32-uClibc.so.0;:%{mbionic:/system/bin/linkerx32;:%{mmusl:/lib/ld-musl-x32.so.1;:/libx32/ld-linux-x32.so.2}}}}}}     %{static:-static} %{static-pie:-static -pie --no-dynamic-linker -z text}} %{shared: -Bsymbolic}}

*lib:
%{!mandroid|tno-android-ld:%{pthread:-lpthread} %{shared:-lc}    %{!shared:%{profile:-lc_p}%{!profile:-lc}};:%{shared:-lc}    %{!shared:%{profile:-lc_p}%{!profile:-lc}} %{!static: -ldl}}

*link_gomp:

*libgcc:
[... cut ...] itizer.spec)%(link_libtsan)}    %{static:%ecannot specify -static with -fsanitize=thread}}    %{%:sanitize(undefined):%{static-libubsan:-Bstatic} -lubsan %{static-libubsan:-Bdynamic} %{static-libubsan|static:%:include(libsanitizer.spec)%(link_libubsan)}}    %{%:sanitize(leak): %{static-liblsan|static:%:include(libsanitizer.spec)%(link_liblsan)}}}}}     %{!nostdlib:%{!r:%{!nodefaultlibs:%(link_ssp) %(link_gcc_c_sequence)}}}    %{!nostdlib:%{!r:%{!nostartfiles:%E}}} %{T*}  
%(post_link) }}}}}}
```

jdemel, 2020-06-09:
This issue is tricky. Apparently @ZeroChaos- found a compiler set up that breaks this kernel. Thus, we'd probably be better off removing these two `zeroupper` calls.
On the other hand, this might have a severe performance impact on newer machines, especially Skylake and post-Skylake. I don't know about Ryzen yet.

Such a CPU will take a severe performance hit everytime SSE code is executed after AVX. Of course, we don't want this performance penalty and thus `zeroupper` is our housekeeping effort.

I'd like to find a way to keep our housekeeping efforts and fix your issue.

@michaelld what's your opinion on this?

rear1019, 2021-01-02:
I can not reproduce this with GCC 9.3 provided in Ubuntu 20.04 (`9.3.0-18ubuntu1`). This might be a miscompilation as noted in [1]. @ZeroChaos- Are you still able to reproduce the error by any chance?

[1] https://github.com/gnuradio/volk/issues/384#issuecomment-753447355

PR #434 (2021-01-07): Remove _mm256_zeroupper() calls
See commit message and discussion in #384 for details. TLDR: manual usage of `_mm256_zeroupper()` is not required and leads to miscompilation with GCC 10.2 and optimization level `-O3`.

This fixes #384.
This fixes #382.
closed 2021-01-07 by michaelld