gnuradio/volk #383: remove unneeded fatal error

ZeroChaos-, 2020-06-08:
What exactly is this fatal error fixing?
https://github.com/gnuradio/volk/blob/master/cmake/Modules/VolkBuildTypes.cmake#L66

In Gentoo we set a build type of "gentoo". This is specifically done to avoid the cmake build types from having a defined flag set or whatever, so we just get what the user requested without any additional cflags and what not.  It's not causing me a major heartache to override and set a non-gentoo build type, but if this fatal error is fixing nothing I'd love to see it removed to save me the override.

If this is actually fixing something, can we append "Gentoo" to the permitted list?

Thanks!

michaelld, 2020-06-08:
Yeah I get where you're coming from. MacPorts does the same thing, which I deal with inside MacPorts so that we use either "Release" or "Debug". We could change this error to instead just warn the user that this is the case, but let the build proceed. We could even try checking some of the variables that CMake will be looking for to see if any are set. I'm all for the user creating their own optimizations & telling CMake how to use them.

jdemel, 2020-06-14:
Does Gentoo require a special configuration?

I suggest to add the required configuration with a more specific name so others can use it on different distros without confusion. Maybe we can do the same for MacPorts?

I totally understand your requirement. Though, I also understand why this check exists. I'm sure it prevents unexpected behavior in case of typos. CMake can be such fun.

ZeroChaos-, 2020-06-23:
Gentoo specifically asks for a configuration which is unlikely to exist so it gets a blank configuration :-)  What we need is simply the "don't do anything special" config which so often doesn't exist by default.  You could make a Gentoo configuration type and basically leave it blank, which should be fine based on my understanding of the build system.

jdemel, 2020-06-25:
I see multiple options:

- We could add a `GENTOO` build type. 
- We could turn this error into a warning. 
- We could add another CMake switch to allow any build type.

There are probably more alternatives. I just don't know which one is better or worse.

Is `None` your config of choice?

ZeroChaos-, 2020-06-25:
Basically yes, "None" is my config of choice.  That said, apparently trust is low in Gentoo that "None" means "don't mess with anything" as a bunch of projects still add cflags and such.  Setting an intentionally non-existent build type was the solution.

I'm fine adding a Gentoo build type that does nothing, or removing the fatal check.  Honestly the issue title says it all, I think the fatal error is unneeded and removing it seems to make sense.  I'm fine with any solution, I'm just highlighting that I still believe the checked to be unnecessarily fatal.