gnuradio/volk #385: 1 - qa_volk_16i_32fc_dot_prod_32fc (Failed)

mkaesberger, 2020-06-13:
I am building volk from the master branch and the test _qa_volk_16i_32fc_dot_prod_32fc_ is failing in release and debug configuration with gcc 9.3.0:

```
$ ctest -VV -R 32fc_dot_prod_32fc
UpdateCTestConfiguration  from :/home/martin/gnuradio/volk/build/DartConfiguration.tcl
UpdateCTestConfiguration  from :/home/martin/gnuradio/volk/build/DartConfiguration.tcl
Test project /home/martin/gnuradio/volk/build
Constructing a list of tests
Done constructing a list of tests
Updating test list for fixtures
Added 0 tests to meet fixture requirements
Checking test dependency graph...
Checking test dependency graph end
test 1
    Start 1: qa_volk_16i_32fc_dot_prod_32fc

1: Test command: /bin/sh "/home/martin/gnuradio/volk/build/lib/volk_16i_32fc_dot_prod_32fc_test.sh" "/home/martin/gnuradio/volk/build/lib"
1: Test timeout computed to be: 10000000
1: RUN_VOLK_TESTS: volk_16i_32fc_dot_prod_32fc(131071,1)
1: generic completed in 0.685358 ms
1: u_sse completed in -nan ms
1: Segmentation fault
1/1 Test #1: qa_volk_16i_32fc_dot_prod_32fc ...***Failed    0.04 sec

0% tests passed, 1 tests failed out of 1

Total Test time (real) =   0.05 sec

The following tests FAILED:
	  1 - qa_volk_16i_32fc_dot_prod_32fc (Failed)
Errors while running CTest
```

cpu:

```
processor	: 0
vendor_id	: AuthenticAMD
cpu family	: 23
model		: 24
model name	: AMD Ryzen 5 3500U with Radeon Vega Mobile Gfx
stepping	: 1
microcode	: 0x8108102
cpu MHz		: 1740.088
cache size	: 512 KB
physical id	: 0
siblings	: 8
core id		: 0
cpu cores	: 4
apicid		: 0
initial apicid	: 0
fpu		: yes
fpu_exception	: yes
cpuid level	: 13
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ht syscall nx mmxext fxsr_opt pdpe1gb rdtscp lm constant_tsc rep_good nopl nonstop_tsc cpuid extd_apicid aperfmperf pni pclmulqdq monitor ssse3 fma cx16 sse4_1 sse4_2 movbe popcnt aes xsave avx f16c rdrand lahf_lm cmp_legacy svm extapic cr8_legacy abm sse4a misalignsse 3dnowprefetch osvw skinit wdt tce topoext perfctr_core perfctr_nb bpext perfctr_llc mwaitx cpb hw_pstate sme ssbd sev ibpb vmmcall fsgsbase bmi1 avx2 smep bmi2 rdseed adx smap clflushopt sha_ni xsaveopt xsavec xgetbv1 xsaves clzero irperf xsaveerptr arat npt lbrv svm_lock nrip_save tsc_scale vmcb_clean flushbyasid decodeassists pausefilter pfthreshold avic v_vmsave_vmload vgif overflow_recov succor smca
bugs		: sysret_ss_attrs null_seg spectre_v1 spectre_v2 spec_store_bypass
bogomips	: 4194.04
TLB size	: 2560 4K pages
clflush size	: 64
cache_alignment	: 64
address sizes	: 43 bits physical, 48 bits virtual
power management: ts ttp tm hwpstate eff_freq_ro [13] [14]
```

jdemel, 2020-06-13:
Could you rerun this test with GDB and debug symbols?
It should be sufficient to run `gdb` and then `run ctest -V -R 16i_32fc_dot_prod`. If the test segfaults, please run `bt` to see where this kernel segfaults and share the results here. Thanks!

mkaesberger, 2020-06-13:
The issue might be related to my libc, because I'm compiling volk against musl. Same issue with clang.

```
$ gdb --args /home/martin/gnuradio/volk/build/lib/volk_test_all volk_16i_32fc_dot_prod_32fc
GNU gdb (GDB) 9.2
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-alpine-linux-musl".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/martin/gnuradio/volk/build/lib/volk_test_all...
(gdb) r
Starting program: /home/martin/gnuradio/volk/build/lib/volk_test_all volk_16i_32fc_dot_prod_32fc
RUN_VOLK_TESTS: volk_16i_32fc_dot_prod_32fc(131071,1)
generic completed in 0.568163 ms
u_sse completed in -nan ms

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff7fb2fe8 in ?? () from /lib/ld-musl-x86_64.so.1
(gdb) bt
#0  0x00007ffff7fb2fe8 in ?? () from /lib/ld-musl-x86_64.so.1
#1  0x0000000000000000 in ?? ()
(gdb)
```

jdemel, 2020-06-14:
They way I read this is: The segfault happens in musl and VOLK just calls that function.

It would be nice to have that VOLK works with musl. I'm curious which VOLK line triggers the segfault. The GDB output suggests that the unaligned SSE kernel triggers the segfault. A `-nan` result may also indicate a problem with printing the results.

mkaesberger, 2020-06-15:
It is not that easy. If a segmentation fault occurs, the backtrace may become unreliable and might contain incorrect data.

I did continue to debug and found the printing in [https://github.com/gnuradio/volk/blob/e564299446a974ee48e30ffc36ee9d443633b9bc/lib/qa_utils.cc#L696](https://github.com/gnuradio/volk/blob/e564299446a974ee48e30ffc36ee9d443633b9bc/lib/qa_utils.cc#L696) triggering the segfault when printing the result from the AVX kernel. Further testing revealed the `_mm_cvtpi16_ps` intrinsic in [https://github.com/gnuradio/volk/blob/e564299446a974ee48e30ffc36ee9d443633b9bc/kernels/volk/volk_16i_32fc_dot_prod_32fc.h#L183-L186](https://github.com/gnuradio/volk/blob/e564299446a974ee48e30ffc36ee9d443633b9bc/kernels/volk/volk_16i_32fc_dot_prod_32fc.h#L183-L186) to interact with the segfault.

In the end the problem is rather simple: this intrinsic is using MMX instructions/registers, which unfortunately is not described in [Intels Intrinsics Guide](https://software.intel.com/sites/landingpage/IntrinsicsGuide/#techs=SSE&expand=133,158,4414,4929,4932,1726,3834,3853,6042,1726,1726,1726&text=_mm_cvtpi16_ps). MMX aliases the x87 FPU registers, which need to be reset/cleared using the `EMMS` instruction before being used by the FPU (refer to  [Intel 64 and IA-32 Architectures Software Developer's Manual](https://software.intel.com/sites/default/files/managed/39/c5/325462-sdm-vol-1-2abcd-3abcd.pdf) sections 9.4.8, 9.5, 9.6.2f for more details). With the invalid state of the FPU registers and [differences in handling floating-point exception](https://wiki.musl-libc.org/functional-differences-from-glibc.html#Floating-point-and-mathematical-library) the conversion to string is failing.

The solution to this problem is to clear the MMX state by calling `_mm_empty()` before using the FPU. I dumped the intermediate assembly and did not find any other kernel using those registers.

closed 2020-06-25 by mkaesberger