gnuradio/volk #386: Erro while installing gnuradio

ik8ozv, 2020-06-14:
Hello all,
I am having the error while I install the gnuradio.
When I give the command:

make test

I get at the end of the list the following error:

99% tests passed, 1 tests failed out of 365
 
Total Test time (real) = 160.52 sec
 
The following tests FAILED:
                25 - qa_volk_32f_8u_polarbutterflypuppet_32f (Failed)
Errors while running CTest
make: *** [Makefile:130: test] Error 8
 
 
Please any advice???
Thanks
Enzo

jdemel, 2020-06-14:
Which VOLK version do you use?
Which GNU Radio version do you use? 
What's your compiler? What's your system?

Could you run `ctest -V -R polarbutterflypuppet` in your VOLK build folder and report the result here?

It might be helpful to update VOLK to the latest version.

jdemel, 2020-07-21:
I assume this issue got resolved. I close it now. If this issue persists, please reopen it.

closed 2020-07-21 by jdemel