gnuradio/volk #387: Inconsistent project name in CMake

skoslowski, 2020-06-14:
I getting the following warning from CMake when consuming VOLK (e.g. in GNU Radio):
```
CMake Warning (dev) at /usr/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:272 (message):
  The package name passed to `find_package_handle_standard_args` (VOLK) does
  not match the name of the calling package (Volk).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
```
The (obvious) solution would be to rename those variables using `CMAKE_FIND_PACKAGE_NAME` instead of a hard-coded casing or, alternatively, use consistent casing. For backwards compatibly, I'd keep the current variables around until some future version TDB. 

First, however, the "right" casing of VOLK should be decided. Currently the cmake project name (and main target is "volk"), the export namespace and config module is "Volk" and the config variable prefix is "VOLK_". Ideally, these should all use the same casing.

jdemel, 2020-06-15:
I agree. 

Do you have a more comprehensive list of CMake lines that would need to be adjusted?
How did you trigger this warning? 

The preferred solution would probably be to rename all CMake variables to `VOLK` and `VOLK_`. Most of those `VOLK_` variables are not hardcoded but CMake generated. Thus, following this style would probably be the right way forward.
Though, the target name, for `target_link_libraries` is `Volk::volk` which is an inconsistency in itself. It should be `volk::volk`.

Do you have a suggestion how to fix this issue?

skoslowski, 2020-06-15:
I'd actually be willing to work on this myself, so I'll provide a list of changes in form of a PR =)
To trigger the warnings I simply configured a GR build on a fedora32 machine.

I strongly suggest to use the same casing for both targets and variables in the config module, see e.g. https://cmake.org/cmake/help/git-stage/module/FindPython.html

So its either VOLK, Volk or volk. A mix isn't really improving the current state.

In any case, the question of backwards compat needs to be decided. Break now or keep copy with the current names around? There isn't really a way to "warn on use", so all you can do is hope people read the release notes or have these the aliases default off.

drmpeg, 2020-06-15:
On Ubuntu, these issues show up if you use CMake version 3.17 or later.

jdemel, 2020-10-18:
@skoslowski did #403 already fix this issue or are there more changes required?

skoslowski, 2020-10-20:
It is definitely a step in the right direction. If `Volk` is now the "correct" casing, I suggest to rename the cmake project to that as well. The prefix for variables could remain `VOLK` for compatibility. Or have a set of duplicates for a while...

drmpeg, 2020-10-20:
Renaming the CMake project does essentially nothing. The install files are identical and even the build directory is identical except for one thing. You can name the project "mxyzptlk" and you still get the same output.

Here's the only thing that changes in the build directory.
```
diff -r ./build/CMakeCache.txt ../volk1/build/CMakeCache.txt
215c215
< CMAKE_PROJECT_NAME:STATIC=Volk
---
> CMAKE_PROJECT_NAME:STATIC=volk
354,359d353
< //Value Computed by CMake
< Volk_BINARY_DIR:STATIC=/home/ubuntu/xfer/volk/build
< 
< //Value Computed by CMake
< Volk_SOURCE_DIR:STATIC=/home/ubuntu/xfer/volk
< 
362a357,359
> //Value Computed by CMake
> volk_BINARY_DIR:STATIC=/home/ubuntu/xfer/volk/build
> 
364a362,364
> 
> //Value Computed by CMake
> volk_SOURCE_DIR:STATIC=/home/ubuntu/xfer/volk
```

skoslowski, 2020-10-21:
True, changing the project name has no real impact - in standalone alone mode. When used as a subproject the impact is also pretty low. So changing it shouldn't break stuff.

jdemel, 2020-10-21:
So far we settled on either `Volk` or `volk`. Are there any places in C/C++ code where we deviate from `volk`? Besides, it looks kind of weird to add the link target in CMake `Volk::volk`. 
@skoslowski would you be willing to make a quick search to find all the different spellings in code? And if they matter? That would be great to decide if we want to take further action.

drmpeg, 2020-10-21:
My 2 cents. Stop right here. The primary issue has been fixed. Anything else is just a rabbit hole for the sake of cosmetics. Also changing the link target from ```Volk::volk``` affects folks that build GNU Radio 3.8 with a newer version of Volk.

jdemel, 2020-10-21:
@drmpeg that's right.
@skoslowski I suggest you close this issue. Or did we miss anything?

closed 2020-10-21 by skoslowski