gnuradio/volk #395: volk_8u_x4_conv_k7_r2_8u kernel is failing for GR conv encoder examples

mormj, 2020-07-30:
See https://github.com/gnuradio/gnuradio/issues/3376

Narrowed the discrepancies in the GR examples to the specific volk kernel.  When setting to `GENERIC` the examples work fine.  I'm guessing one solution would be to remove the offending kernel here.

jdemel, 2020-07-30:
We have 3 kernels here: generix, avx2 and spiral. Avx2/spiral both yield a ~6x speed up and VOLK tests pass. We don't want to break API, thus we can't remove it entirely. Which kernel fails exactly? I can't reproduce this error.

jdemel, 2020-08-02:
By now, I could reproduce the error. One of the AVX2 kernels is broken. Unfortunately, the bug only manifests itself in this GR example flowgraph. I couldn't make it fail with VOLK directly. I guess, this does require some `volk_profile -n -R conv_k7 -v SOMELENGTH -i SOMEITERATIONS` to figure out which configuration fails eventually. Or maybe this is related to how the kernel is used in GR.

armin485, 2020-11-13:
Same Problem rises when working with any kind of POLAR encoding/decoding blocks. after changing to GENERIC, everything are solved.

jdemel, 2020-11-13:
From  [gnuradio#3376](https://github.com/gnuradio/gnuradio/issues/3376):
There's a workaround for volk_8u_x4_conv_k7_r2_8u`. Run `volk_profile`, then inspect `~/.volk/volk_config`. Find the line that starts with:
```
volk_8u_x4_conv_k7_r2_8u 
```
and make sure, it ends with spiral spiral (or generic generic or a mix thereof)
```
volk_8u_x4_conv_k7_r2_8u spiral spiral
```
The AVX kernel seems to be the reason for this bug. Unfortunately, it only manifests itself in GR. Thus, this bug is difficult to reproduce and track down. For some reason, the exact parameters to make this kernel fail are missing.

jdemel, 2020-11-13:
@armin485 may I ask you to do the following things:
- Open a new issue for the polar encoder and decoder kernels
- Describe how you make these fail
- Give us insight into your configuration. OS/distro, versions, etc.

That'll make it easier to resolve the issue. Currently, it would probably be forgotten aka overlooked.

Aang23, 2021-03-12:
I can confirm this still is an issue to this day.

The AVX kernel produces gargabe output in all tests I've done, except when processing a single encoded frame or similar.

Aang23, 2022-08-18:
@michaelld @jdemel sorry for the ping - but I'd say this issue can be closed now. The 2 fixes have now been in Volk for months : 
- The first issue, initially affecting GR being the broken AVX2 kernel port. This is now fixes as the port has been disabled, and the identically (if not better) performing SSE kernel does not have any of those issues.
- The second which was about the Generic implementation having extremely poor FEC performances. It's also been well tried in the real world by now, and I've not heard of any further issues since then.

jdemel, 2022-08-19:
I agree. @michaelld if you agree too, I suggest we close this issue.

closed 2022-08-19 by michaelld