gnuradio/volk #397: volk_profile & many qa_volk_* fail due to illegal instruction

isophase, 2020-07-31:
Hello, 
I'm pretty certain the issue described in the title is that the profiler/tester is running an instruction it thinks is available on my CPU but is not. Also fairly confident that the solution would just be selecting a different kernel, but I'm not familiar enough with the internals to know how best to do that, which kernel should be selected, or what the root cause might be (aside from the massive number of extensions that may or may not be available!).

It's a new 13" MacBook Pro, Intel Sunny Cove
[Output of cpuid](https://github.com/gnuradio/volk/files/5008917/cpuid.txt)
[Output of /proc/cpuinfo](https://github.com/gnuradio/volk/files/5008918/proc_cpuinfo.txt)
`(simple synth)  = Intel Core i*-10000 (Ice Lake-U/Y) [Sunny Cove] {Sunny Cove}, 10nm`

Full test output is attached below: 60/134 tests failed, all due to illegal instruction

Running GDB on a couple of the failing tests, the failures I saw were on the following instruction/function:
```
VPBROADCASTQ in volk_64upopcntpuppet_64u_generic
VRNDSCALESS in volk_16ic_magnitude_16i_a_avx2
VMOVUPS in volk_32f_64f_add_64f_generic and volk_32f_64f_multiply_64f
```

It said [somewhere on the 'net](https://www.felixcloutier.com/x86/vrndscaless) that VRNDSCALESS is an AVX-512F instruction, which cpuid says is supported but apparently causes an illegal instruction. Very puzzled.

The VPBROADCASTQ one is less surprising [the same resource](https://www.felixcloutier.com/x86/vpbroadcastb:vpbroadcastw:vpbroadcastd:vpbroadcastq) lists it as an AVX512VL (vector length) instruction which is not supported according to cpuid.

Assistance would be greatly appreciated.

Thanks,

Jeff Cassidy
Isophase Computing Ltd

Here's the deal:
Build setup:
Ubuntu 20.04
CMake 3.16.3
Ninja 1.10.0
G++ 9.3.0 (stock version with Ubuntu 20.04)
Docker on Mac, so it's inside a Hyperkit vm
Output from `cpuid` and `cat /proc/cpuinfo` attached

Docker commands to build: 

```
ARG ENABLE_TESTING=ON
ARG CMAKE_BUILD_TYPE=Release
RUN cmake -G Ninja -DCMAKE_BUILD_TYPE=$CMAKE_BUILD_TYPE -DENABLE_TESTING=$ENABLE_TESTING -DCMAKE_INSTALL_PREFIX=/opt/volk /src/volk
RUN cmake --build .
RUN cmake --build . test
```

Output from `ctest --output-on-failure`:
```
Test project /build/volk
        Start   1: qa_volk_16i_32fc_dot_prod_32fc
  1/134 Test   #1: qa_volk_16i_32fc_dot_prod_32fc .......................   Passed    0.03 sec
        Start   2: qa_volk_16i_branch_4_state_8
  2/134 Test   #2: qa_volk_16i_branch_4_state_8 .........................   Passed    0.01 sec
        Start   3: qa_volk_16i_convert_8i
  3/134 Test   #3: qa_volk_16i_convert_8i ...............................   Passed    0.01 sec
        Start   4: qa_volk_16i_max_star_16i
  4/134 Test   #4: qa_volk_16i_max_star_16i .............................   Passed    0.01 sec
        Start   5: qa_volk_16i_max_star_horizontal_16i
  5/134 Test   #5: qa_volk_16i_max_star_horizontal_16i ..................   Passed    0.01 sec
        Start   6: qa_volk_16i_permute_and_scalar_add
  6/134 Test   #6: qa_volk_16i_permute_and_scalar_add ...................   Passed    0.01 sec
        Start   7: qa_volk_16i_s32f_convert_32f
  7/134 Test   #7: qa_volk_16i_s32f_convert_32f .........................   Passed    0.02 sec
        Start   8: qa_volk_16i_x4_quad_max_star_16i
  8/134 Test   #8: qa_volk_16i_x4_quad_max_star_16i .....................   Passed    0.01 sec
        Start   9: qa_volk_16i_x5_add_quad_16i_x4
  9/134 Test   #9: qa_volk_16i_x5_add_quad_16i_x4 .......................   Passed    0.01 sec
        Start  10: qa_volk_16ic_convert_32fc
 10/134 Test  #10: qa_volk_16ic_convert_32fc ............................   Passed    0.02 sec
[... 524 lines cut ...]
	 80 - qa_volk_32fc_deinterleave_real_64f (Failed)
	 84 - qa_volk_32fc_magnitude_squared_32f (Failed)
	 87 - qa_volk_32fc_s32f_magnitude_16i (Failed)
	 92 - qa_volk_32fc_s32fc_multiply_32fc (Failed)
	 95 - qa_volk_32fc_x2_add_32fc (Failed)
	 96 - qa_volk_32fc_x2_conjugate_dot_prod_32fc (Failed)
	 97 - qa_volk_32fc_x2_divide_32fc (Failed)
	 98 - qa_volk_32fc_x2_dot_prod_32fc (Failed)
	 99 - qa_volk_32fc_x2_multiply_32fc (Failed)
	100 - qa_volk_32fc_x2_multiply_conjugate_32fc (Failed)
	101 - qa_volk_32fc_x2_s32f_square_dist_scalar_mult_32f (Failed)
	102 - qa_volk_32fc_x2_s32fc_multiply_conjugate_add_32fc (Failed)
	103 - qa_volk_32fc_x2_square_dist_32f (Failed)
	104 - qa_volk_32i_s32f_convert_32f (Failed)
	105 - qa_volk_32i_x2_and_32i (Failed)
	106 - qa_volk_32i_x2_or_32i (Failed)
	110 - qa_volk_32u_popcntpuppet_32u (Failed)
	112 - qa_volk_64f_convert_32f (Failed)
	113 - qa_volk_64f_x2_add_64f (Failed)
	114 - qa_volk_64f_x2_max_64f (Failed)
	115 - qa_volk_64f_x2_min_64f (Failed)
	116 - qa_volk_64f_x2_multiply_64f (Failed)
	120 - qa_volk_64u_popcntpuppet_64u (Failed)
	129 - qa_volk_8ic_x2_s32f_multiply_conjugate_32fc (Failed)
```

jdemel, 2020-08-02:
I assume this will be difficult to debug. So let's give it a try. 
Could you change the CMake build type to debug and rerun those tests in GDB? Unfortunately e.g. `VPBROADCASTQ` is used in a couple of different intrinsics according to [Intel Intrinsics Guide](https://software.intel.com/sites/landingpage/IntrinsicsGuide/#text=VPBROADCASTQ&expand=557). I hope to nail down the line of code that causes the error. Also, I hope to get more insight which version of the kernel fails exactly.

isophase, 2020-08-02:
Hi @jdemel , thanks for stepping up to help so quickly! Really appreciated. 
When I build in debug mode, only 20/134 tests fail so looks like there's an optimization that may be a factor. The instructions that fail also change with build type so you're right this may be a fun one.

I automated some data-gathering to help - script & output are attached. 
I ran CTest to figure out which cases fail, then asked CTest to list the failing tests. A python script turned that list into a gdb script to run all the failing tests, and for each query the instruction and get the backtrace like this:

[munge.py.txt (change extension)](https://github.com/gnuradio/volk/files/5012391/munge.py.txt)
[gdb_output_debug.txt](https://github.com/gnuradio/volk/files/5012392/gdb_output_debug.txt)
[gdb_output_release.txt](https://github.com/gnuradio/volk/files/5012400/gdb_output_release.txt)

```
ctest --rerun-failed -N | python3 munge.py > run_tests.gdb
gdb --batch -x run_tests.gdb > gdb_output.txt
```

Please let me know what else might be helpful and thanks again for your help!

Jeff

jdemel, 2020-08-03:
Thanks for putting together these debug scripts! As far as I can see you ran your tests with current master?
I just looked at the first few errors:
```cpp
*ft = (-2 * ((float)*target++)) + 1.0f;
```
It's in `volk_32f_8u_polarbutterflypuppet_32f`.
In `volk_32f_index_max_32u`, the error is on
```cpp
target[0] = (uint32_t)index;
```
The third in `volk_32f_s32f_calc_spectral_noise_floor_32f`:
```cpp
const float meanAmplitude = (sumMean / ((float)num_points)) + spectralExclusionValue;
```
Likely candidates for errors: `vcvttss2usi`, `vmovss`, `vmovups`. 
[vcvttss2usi](https://software.intel.com/sites/landingpage/IntrinsicsGuide/#text=_mm_cvtt_roundss_u32&expand=5236,1938) is apparently another AVX512F instruction, as the ones failing in your original post.
Your system reports AVX512... support but then fails to execute.

Since you're inside a VM, you might want to try compile VOLK on your host system. I know VMs are sometimes not the best at passing around SIMD instructions. If I recall correctly, older VirtualBox Versions wouldn't allow AVX2 for some reason. I hope your errors disappear on your host system.

isophase, 2020-08-13:
Sorry I dropped this for a bit as work/life got a little busy (this is a side project). 
volk_profile runs OK when I use the Homebrew install (3.8.1.0_1) so that's positive.
Perhaps looking at the Homebrew patches would be informative.

I'd prefer to run in a Docker container to match my normal multi-platform workflow (Windows, Mac, & Linux) but want to make progress so will go with the Homebrew install for now.

Native build of tag v3.8.1.0 fails with Clang for reasons that appear unrelated (output attached). I think I've seen similar where bind placeholders need a namespace qualifier in Clang not g++ but I don't have the time to resolve at the moment.

Compiler version output: [clang_version.txt](https://github.com/gnuradio/volk/files/5070182/clang_version.txt)
Errors: [errors.txt](https://github.com/gnuradio/volk/files/5070183/errors.txt)

I will follow up (attempt G++ build and/or try to fix the Clang build and run) as I can.
Thanks again for your help

jdemel, 2020-08-14:
@isophase I assume you want to file a bug report against Docker/Mac and ask them to fix the DockerMacVM issue then? Also, I assume as far as VOLK is concerned, did we resolve the issue?

> Native build of tag v3.8.1.0 fails with Clang for reasons that appear unrelated (output attached). I think I've seen similar where bind placeholders need a namespace qualifier in Clang not g++ but I [...]

This issue is about GNU Radio, not VOLK, correct? GR doesn't build with latest Boost. You might want to follow [GRissue3566](https://github.com/gnuradio/gnuradio/pull/3566) here.

jdemel, 2020-10-18:
@isophase I assume we could resolve your issue. Thanks for the discussion. I close this issue now. Feel free to re-open it in case this issue is not resolved.

closed 2020-10-18 by jdemel