gnuradio/volk #400: Undefined behaviour in  32f_s32f_convert_32i

marcusmueller, 2020-08-08:
clang complains, afaict correctly about implicitly rounding up `INT_MAX` to the next integer (logical, since a 32 bit float can't have more significant digits than a 32 bit int at full dynamic range):

```
../kernels/volk/volk_32f_s32f_convert_32i.h:384:21: warning: implicit conversion from 'int' to 'float' changes value from 2147483647 to 2147483648 [-Wimplicit-int-float-conversion]
    float max_val = INT_MAX;
          ~~~~~~~   ^~~~~~~
/usr/lib64/clang/10.0.0/include/limits.h:46:19: note: expanded from macro 'INT_MAX'
#define INT_MAX   __INT_MAX__
                  ^~~~~~~~~~~
<built-in>:37:21: note: expanded from here
#define __INT_MAX__ 2147483647
                    ^~~~~~~~~~
```

Further down, we then convert the `max_val`-bounded value, which is larger than `INT_MAX`, to a signed int.

That includes the overflow case. Signed integer overflow is UB.

We should probably, instead of clipping the float value to the max integer (which as shown above fails due to precision) and then converting that to int, simply compare `r >= max_val`, and then directly set the output value depending on that (to `INT_MAX` or the conversion result).

jdemel, 2020-10-18:
I can confirm this issue on my machine.
```
warning: implicit conversion from 'int' to 'float' changes value from 2147483647 to 2147483648 [-Wimplicit-int-float-conversion]
```
I'm curious why this value is changed to `INT_MAX + 1`. Besides, how should this be fixed? I found quite a few project that discuss this issue but none discuss the one we have.

michael-roe, 2022-05-27:
Ok, so this is meant to be a float to int conversion where large (negative or positive) values saturate at MAXINT or MININT. But....

- what about NaN?
- Do we want the rounding to be according to the current rounding mode (which is probably round to nearest tie even) or is it ok to always round to nearest tie even?

michael-roe, 2022-05-27:
For what it's worth, the behaviour of MIPS cvt.w.s is delightfully dependent on ISA version and mode flags, but will probably do what you want with a sufficiently recent version of the ISA and fcsr.nan2008=true. In that case, Nan gets rounded to zero.

N.B. I am not suggesting using using assembly language here. I am hoping that the compiler is clever enough to spot that the explicit check for out of range can be optimised out, because the underlying machine instruction does the check for you.

marcusmueller, 2022-05-27:
> what about NaN?

since (unsigned) ints don't "know" NaN, I'd be OK with defining that to be undefined behaviour, in the name of throughput. If you have a stream of floats with some being NaNs (and not just +- infinities), then you kind of either have a problem you need to fix elsewhere, or don't care about these exact values. But that's just my 2 cents!

> Do we want the rounding to be according to the current rounding mode (which is probably round to nearest tie even) or is it ok to always round to nearest tie even?

*tears of joy* yes, this is something we need to talk about. By the way, for the whole project. SIMD unit rounding modes are sometimes decoupled from these used in "generic" C code, so we really should be settling on something.

Tie to even is the preferred mode for most people, I guess, and as far as I can tell any processor supports it. So, that seems to be a good standard to define as mandatory when not specified otherwise in the kernel documentation. (I wouldn't generally say we *need* every kernel to be strict on that – many, if not most, are approximations, and rounding errors are significantly lower than their approximative error. I *would*, however, encourage kernel authors to explicitly state what kind of precision they guarantee. We have arithmetic functions, which we know to be less precise than e.g. libc versions, yet we nowhere state how good or bad they are; the kernel tests just sometimes use comically large acceptable errors with no explanation!)

michael-roe, 2022-05-27:
In gnuradio's float_to_int_impl.c, I see this comment:

   // Disable the Volk for now. There is a problem for large 32-bit ints that
    // are not properly represented by the precisions of a single float, which
    // can cause wrapping from large, positive numbers to negative.
    // In gri_float_to_int, the value is first promoted to a 64-bit
    // value, clipped, then converted to a float.

It seems that gnuradio is not using volk_32f_s32f_convert_32i because of this bug.

I think changing a > to a >= would fix the generic case, at least.

jdemel, 2024-01-12:
#575 should've fixed this issue. I assume we just forgot to close this issue back then.

I'm closing this issue now. Feel free to re-open if the issue is unresolved.

closed 2024-01-12 by jdemel