gnuradio/volk #401: macOS volk_profile reports many `no architectures to test` and `generic`

citizenrich, 2020-08-23:
Hi folks. I'm getting a lot of `no architectures to test` and `generic` in the `volk_profile` report generated. This is on macOS (AMD64 arch) built from source for volk tag 2.3.0. This is strange. I can do more debugging this week, but curious of folks have seen this before. Shall I explicitly declare arch?

Output: https://pastebin.com/f7sEXrRD

cc: https://github.com/Homebrew/homebrew-core/issues/59921

jdemel, 2020-08-24:
@michaelld knows more about the details here. Generally, we hope to address this issue with #393 . It'd be great if you could test this PR and find and remove the switches that disable arch specific builds.

At the moment, as long as your output looks like our macOS CI build, you got a standard Mac install.
This should boil down to a missing `xgetbv` function. Currently, we rely on it to detect architectures on runtime.

citizenrich, 2020-08-25:
I tried the PR. Yep, an issue with macs and xgetbv. I see a solution for this, but tbh I don't entirely understand the solution:
https://github.com/google/cpu_features/pull/69

The solution is explained a bit better here: https://github.com/asmjit/asmjit/issues/78

Sorry if this isn't much help.

jdemel, 2020-08-25:
Thanks for your link. That kind of incremental improvement was my motivation to integrate cpu_features into VOLK. I don't want to reinvent the wheel but benefit from a working solution.

citizenrich, 2020-08-25:
Thanks. It seems the numpy folks had to address this as well:
https://github.com/numpy/numpy/pull/16297

jdemel, 2020-10-18:
@citizenrich, would you be willing to have a look at #393 to figure out what to change in our CMake to fix your issue? When I started to work on #393 I had your issue in mind.

michaelld, 2020-10-18:
Sorry I'm late to the party here! Yes, I see this as well. Both Intel and ARM64. This is causing minor issues with some work I'm doing, which uses Volk and requires a kernel to be usable ... "generic" is find for some but I'd much prefer real kernels. Not sure if this is related to the CPU features discussion or not ... which I also need to chime in on. Sigh; not enough time in the day!

jdemel, 2020-10-18:
I dug deeper into this. The issue is related. Let me summarize my findings so far. I don't have MacOS available thus it is kind of tricky.

- GH Actions CI reports `COMPILER_NAME=AppleClang`. in `gen/archs.xml` we only define flags for `clang`. Thus, `gen/volk_compile_utils.py` skips arch checks for `appleclang` because it assumes this compiler is unsupported.
- If we assume `clang==appleclang`, we could probably just overwrite this option. Is it correct to assume that AppleClang is just some kind of "old stable" clang compiler?
- The exact relation with #393 is: On AppleClang, we miss `xgetbv`. We don't need it anymore with `cpu_features`. Thus, we are able to really check for archs during runtime.

citizenrich, 2020-10-18:
Thanks for support. I'll test out the PR.

jdemel, 2020-10-21:
@citizenrich #393 enables all kernels in our macos GH CI now. Of course, it would be great if you could test it. If #393 solves your issue, I'd be happy to link it as a solution here.

closed 2020-10-23 by michaelld

citizenrich, 2020-10-23:
Apologies for the delay. Yep, I'm looking at all tests passing using main branch. Such volk goodness! Thanks @jdemel and @michaelld

michaelld, 2020-10-23:
> Apologies for the delay. Yep, I'm looking at all tests passing using main branch. Such volk goodness! Thanks @jdemel and @michaelld

You're welcome ... I'll take the minor credit of merging the PR ... it was all @jdemel 's work really!

PR #393 (2020-10-23): cpu_features: Switch to cpu_features
We used our own CPU feature detection methods. These were not totally
reliable and difficult to implement across compilers and OSes. We switch
to `cpu_features` as a subproject and build upon this expertise.

So far this is a work-in-progress. Known issues:
   - <s>code needs clean-up</s>
   - <s>remove unnecessary checks</s>
   - <s>make sure all NEON versions are still detected correctly</s> - CI indicates correct detection.
   - <s>incomplete `VOLK_CPU` feature checks:</s>
	- <s>`has_64`: check 64bit CPU.</s> We compile for one or the other. Not for both [... cut]