gnuradio/volk #402: volk_profile fails with "libvolk.so.2.3: cannot open shared object file: no such file or directory"

VictorWiedemann, 2020-09-08:
I am using the following steps that are also found on the volk README:

_Building on most x86 (32-bit and 64-bit) platforms
$ mkdir build
$ cd build
$ cmake ..
$ make
$ make test
$ sudo make install

# volk_profile will profile your system so that the best kernel is used
$ volk_profile_

I am trying to get the volk_profile to run on on my VM. When I try to run volk_profile, I get the following error:

**dev@dev:~/Documents/gitRepo/volk/build$** volk_profile
volk_profile: error while loading shared libraries: libvolk.so.2.3: cannot open shared object file: No such file or directory

When I go directly into usr/local/bin and run ./volk_profile it gives me the same error as above.

When I run the volk_profile within the gitRepo/volk/build/apps and run ./volk_profile from there it does work. I am trying to get it to run volk_profile properly from usr/local/bin.

jdemel, 2020-09-08:
This sounds a lot like the shared library is not on your search path.
First, double check for your system installed `volk_profile` where it is located with `which volk_profile`.
Second, check with `ldd` which shared libraries are found for your local and your installed volk_profile. What do they link against?
Third, what kind of VM do you use?

VictorWiedemann, 2020-09-08:
1. This is the returned value from "which volk_profile":
/usr/local/bin/volk_profile

from within usr/local/bin, here is the output from ldd volk_profile:
 
	linux-vdso.so.1 (0x00007fff7e7dc000)
	libvolk.so.2.3 => not found
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f46d5969000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f46d55cb000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f46d53b3000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f46d4fc2000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f46d5f3f000)

I am using Oracle VM VirtualBox running Ubuntu 18.04

VictorWiedemann, 2020-09-08:
So jdemel gave me an idea. Since ldd showed the missing link, I found this link and accepted answer: https://unix.stackexchange.com/questions/279397/ldd-does-not-find-path-how-to-add?newreg=e0abfb67edc3474a8d069621970b4063

To quote this post in-case of, future broken link, the following worked for me:

> 
> if your libraries are not on standard path then either you need to add them to the path or add non-standard path to LD_LIBRARY_PATH
> 
> `export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:<Your_non-Standard_path>`
> 
> Once you done any one of above things then you need to update the dynamic linker run-time binding by executing below command:
> 
> `sudo ldconfig`

closed 2020-09-08 by VictorWiedemann