gnuradio/volk #410: cpu_features build issue -> failure w/ pybombs

cclamb, 2020-10-23:
Hey guys! So cpu-features is a submodule but when I install gnuradio via pybombs the git checkout doesn't execute recursively so the cpu-features submodule isn't checked out. As a result the pybombs build fails. To recreate:

pybombs auto-config
pybombs recipes add-defaults
pybombs prefix init ~/gnuradio -R gnuradio-default

This seemed like the right place to report. If not, could you let me know who I should notify? Thanks!

michaelld, 2020-10-23:
There should be a Volk recipe used by PyBOMBS. Pretty sure we don't maintain that. It does indeed need to be updated before the next Volk release. Try adding an issue to PyBOMBS GitHub noting this change; hopefully the new maintainer(s) will know what to do. Leaving this open for moment as a reminder.

jdemel, 2020-10-23:
Thanks for the report. This needs to be fixed in [gr-recipes](https://github.com/gnuradio/gr-recipes).

PyBOMBS installs the VOLK `master` branch by default:
https://github.com/gnuradio/gr-recipes/blob/2df89e6cd25e9ee78cfe4f1c9d45e2a0107c66b3/libvolk.lwr#L26
Just add a new line with `gitargs: --recursive` and further pybombs installations will run through smoothly.

jdemel, 2020-10-23:
You may also try https://github.com/gnuradio/gr-recipes/pull/238 that fixes this issue.

jdemel, 2020-10-25:
The required fix got merged into pybombs. You might want to update your gr-recipes clone. I close this issue now. Feel free to reopen it if the issue persists.

closed 2020-10-25 by jdemel