gnuradio/volk #412: Lastest Volk fails as submodule to GNU Radio 3.8.2

drmpeg, 2020-11-01:
When trying to build GNU Radio 3.8.2 with the latest Volk, the build fails when trying to link ```list_cpu_features```. It appears that the build for ```list_cpu_features``` is different when Volk is a submodule than when it's built stand-alone.

The most obvious difference is that the library libcpu_features gets build as a static library (libcpu_features.a) for stand-alone and as a shared library for submodule (libcpu_features.so).

Steps to reproduce:
```
git clone https://github.com/gnuradio/gnuradio.git
cd gnuradio
git checkout maint-3.8
git submodule update --init
cd volk
git checkout 24c7c84
git submodule update --init
mkdir build
cd build
cmake -DCMAKE_INSTALL_PREFIX=/opt/gnuradio-3.8.2 ../
make
```

Here's the verbose dump for Volk as a submodule.
```
re@beavis:~/xfer/gnuradio/build$ make VERBOSE=1
/usr/local/bin/cmake -S/home/re/xfer/gnuradio -B/home/re/xfer/gnuradio/build --check-build-system CMakeFiles/Makefile.cmake 0
/usr/local/bin/cmake -E cmake_progress_start /home/re/xfer/gnuradio/build/CMakeFiles /home/re/xfer/gnuradio/build//CMakeFiles/progress.marks
make  -f CMakeFiles/Makefile2 all
make[1]: Entering directory '/home/re/xfer/gnuradio/build'
make  -f volk/cpu_features/CMakeFiles/unix_based_hardware_detection.dir/build.make volk/cpu_features/CMakeFiles/unix_based_hardware_detection.dir/depend
make[2]: Entering directory '/home/re/xfer/gnuradio/build'
cd /home/re/xfer/gnuradio/build && /usr/local/bin/cmake -E cmake_depends "Unix Makefiles" /home/re/xfer/gnuradio /home/re/xfer/gnuradio/volk/cpu_features /home/re/xfer/gnuradio/build /home/re/xfer/gnuradio/build/volk/cpu_features /home/re/xfer/gnuradio/build/volk/cpu_features/CMakeFiles/unix_based_hardware_detection.dir/DependInfo.cmake --color=
Dependee "/home/re/xfer/gnuradio/build/volk/cpu_features/CMakeFiles/unix_based_hardware_detection.dir/DependInfo.cmake" is newer than depender "/home/re/xfer/gnuradio/build/volk/cpu_features/CMakeFiles/unix_based_hardware_detection.dir/depend.internal".
Dependee "/home/re/xfer/gnuradio/build/volk/cpu_features/CMakeFiles/CMakeDirectoryInformation.cmake" is newer than depender "/home/re/xfer/gnuradio/build/volk/cpu_features/CMakeFiles/unix_based_hardware_detection.dir/depend.internal".
Scanning dependencies of target unix_based_hardware_detection
make[2]: Leaving directory '/home/re/xfer/gnuradio/build'
make  -f volk/cpu_features/CMakeFiles/unix_based_hardware_detection.dir/build.make volk/cpu_features/CMakeFiles/unix_based_hardware_detection.dir/build
make[2]: Entering directory '/home/re/xfer/gnuradio/build'
[  1%] Building C object volk/cpu_features/CMakeFiles/unix_based_hardware_detection.dir/src/hwcaps.c.o
cd /home/re/xfer/gnuradio/build/volk/cpu_features && /usr/bin/cc -DGR_PERFORMANCE_COUNTERS -DHAVE_DLFCN_H -DHAVE_STRONG_GETAUXVAL -DSTACK_LINE_READER_BUFFER_SIZE=1024 -D_GLIBCXX_USE_CXX11_ABI=1 -I/home/re/xfer/gnuradio/volk/cpu_features/include -I/home/re/xfer/gnuradio/volk/cpu_features/include/internal -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -fcx-limited-range -Wall -O3 -DNDEBUG -fPIC -std=c99 -o CMakeFiles/unix_based_hardware_detection.dir/src/hwcaps.c.o -c /home/re/xfer/gnuradio/volk/cpu_features/src/hwcaps.c
make[2]: Leaving directory '/home/re/xfer/gnuradio/build'
[  1%] Built target unix_based_hardware_detection
make  -f volk/cpu_features/CMakeFiles/utils.dir/build.make volk/cpu_features/CMakeFiles/utils.dir/depend
make[2]: Entering directory '/home/re/xfer/gnuradio/build'
cd /home/re/xfer/gnuradio/build && /usr/local/bin/cmake -E cmake_depends "Unix Makefiles" /home/re/xfer/gnuradio /home/re/xfer/gnuradio/volk/cpu_features /home/re/xfer/gnuradio/build /home/re/xfer/gnuradio/build/volk/cpu_features /home/re/xfer/gnuradio/build/volk/cpu_features/CMakeFiles/utils.dir/DependInfo.cmake --color=
Dependee "/home/re/xfer/gnuradio/build/volk/cpu_features/CMakeFiles/utils.dir/DependInfo.cmake" is newer than depender "/home/re/xfer/gnuradio/build/volk/cpu_features/CMakeFiles/utils.dir/depend.internal".
Dependee "/home/re/xfer/gnuradio/build/volk/cpu_features/CMakeFiles/CMakeDirectoryInformation.cmake" is newer than depender "/home/re/xfer/gnuradio/build/volk/cpu_features/CMakeFiles/utils.dir/depend.internal".
Scanning dependencies of target utils
make[2]: Leaving directory '/home/re/xfer/gnuradio/build'
make  -f volk/cpu_features/CMakeFiles/utils.dir/build.make volk/cpu_features/CMakeFiles/utils.dir/build
make[2]: Entering directory '/home/re/xfer/gnuradio/build'
[  1%] Building C object volk/cpu_features/CMakeFiles/utils.dir/src/filesystem.c.o
cd /home/re/xfer/gnuradio/build/volk/cpu_features && /usr/bin/cc -DGR_PERFORMANCE_COUNTERS -DSTACK_LINE_READER_BUFFER_SIZE=1024 -D_GLIBCXX_USE_CXX11_ABI=1 -I/home/re/xfer/gnuradio/volk/cpu_features/include -I/home/re/xfer/gnuradio/volk/cpu_features/include/internal -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -fcx-limited-range -Wall -O3 -DNDEBUG -fPIC -std=c99 -o CMakeFiles/utils.dir/src/filesystem.c.o -c /home/re/xfer/gnuradio/volk/cpu_features/src/filesystem.c
[  1%] Building C object volk/cpu_features/CMakeFiles/utils.dir/src/stack_line_reader.c.o
cd /home/re/xfer/gnuradio/build/volk/cpu_features && /usr/bin/cc -DGR_PERFORMANCE_COUNTERS -DSTACK_LINE_READER_BUFFER_SIZE=1024 -D_GLIBCXX_USE_CXX11_ABI=1 -I/home/re/xfer/gnuradio/volk/cpu_features/include -I/home/re/xfer/gnuradio/volk/cpu_features/include/internal -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -fcx-limited-range -Wall -O3 -DNDEBUG -fPIC -std=c99 -o CMakeFiles/utils.dir/src/stack_line_reader.c.o -c /home/re/xfer/gnuradio/volk/cpu_features/src/stack_line_reader.c
[  1%] Building C object volk/cpu_features/CMakeFiles/utils.dir/src/string_view.c.o
cd /home/re/xfer/gnuradio/build/volk/cpu_features && /usr/bin/cc -DGR_PERFORMANCE_COUNTERS -DSTACK_LINE_READER_BUFFER_SIZE=1024 -D_GLIBCXX_USE_CXX11_ABI=1 -I/home/re/xfer/gnuradio/volk/cpu_features/include -I/home/re/xfer/gnuradio/volk/cpu_features/include/internal -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -fcx-limited-range -Wall -O3 -DNDEBUG -fPIC -std=c99 -o CMakeFiles/utils.dir/src/string_view.c.o -c /home/re/xfer/gnuradio/volk/cpu_features/src/string_view.c
make[2]: Leaving directory '/home/re/xfer/gnuradio/build'
[  1%] Built target utils
make  -f volk/cpu_features/CMakeFiles/cpu_features.dir/build.make volk/cpu_features/CMakeFiles/cpu_features.dir/depend
make[2]: Entering directory '/home/re/xfer/gnuradio/build'
cd /home/re/xfer/gnuradio/build && /usr/local/bin/cmake -E cmake_depends "Unix Makefiles" /home/re/xfer/gnuradio /home/re/xfer/gnuradio/volk/cpu_features /home/re/xfer/gnuradio/build /home/re/xfer/gnuradio/build/volk/cpu_features /home/re/xfer/gnuradio/build/volk/cpu_features/CMakeFiles/cpu_features.dir/DependInfo.cmake --color=
Dependee "/home/re/xfer/gnuradio/build/volk/cpu_features/CMakeFiles/cpu_features.dir/DependInfo.cmake" is newer than depender "/home/re/xfer/gnuradio/build/volk/cpu_features/CMakeFiles/cpu_features.dir/depend.internal".
Dependee "/home/re/xfer/gnuradio/build/volk/cpu_features/CMakeFiles/CMakeDirectoryInformation.cmake" is newer than depender "/home/re/xfer/gnuradio/build/volk/cpu_features/CMakeFiles/cpu_features.dir/depend.internal".
Scanning dependencies of target cpu_features
make[2]: Leaving directory '/home/re/xfer/gnuradio/build'
make  -f volk/cpu_features/CMakeFiles/cpu_features.dir/build.make volk/cpu_features/CMakeFiles/cpu_features.dir/build
make[2]: Entering directory '/home/re/xfer/gnuradio/build'
[  1%] Building C object volk/cpu_features/CMakeFiles/cpu_features.dir/src/cpuinfo_x86.c.o
cd /home/re/xfer/gnuradio/build/volk/cpu_features && /usr/bin/cc -DGR_PERFORMANCE_COUNTERS -DSTACK_LINE_READER_BUFFER_SIZE=1024 -D_GLIBCXX_USE_CXX11_ABI=1 -Dcpu_features_EXPORTS -I/home/re/xfer/gnuradio/volk/cpu_features/include -I/home/re/xfer/gnuradio/volk/cpu_features/include/internal -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -fcx-limited-range -Wall -O3 -DNDEBUG -fPIC -std=c99 -o CMakeFiles/cpu_features.dir/src/cpuinfo_x86.c.o -c /home/re/xfer/gnuradio/volk/cpu_features/src/cpuinfo_x86.c
[  1%] Linking C shared library libcpu_features.so
cd /home/re/xfer/gnuradio/build/volk/cpu_features && /usr/local/bin/cmake -E cmake_link_script CMakeFiles/cpu_features.dir/link.txt --verbose=1
[... cut ...]
/home/re/xfer/volk/cpu_features/src/utils/list_cpu_features.c:343:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
[  7%] Linking C executable list_cpu_features
cd /home/re/xfer/volk/build/cpu_features && /usr/local/bin/cmake -E cmake_link_script CMakeFiles/list_cpu_features.dir/link.txt --verbose=1
/usr/bin/cc  -fcx-limited-range -Wall -O3 -DNDEBUG -rdynamic CMakeFiles/list_cpu_features.dir/src/utils/list_cpu_features.c.o -o list_cpu_features  libcpu_features.a -ldl 
make[2]: Leaving directory '/home/re/xfer/volk/build'
[  7%] Built target list_cpu_features
```

jdemel, 2020-11-07:
I wonder how to properly solve this issue. Should we force a static `cpu_features` build? Or go down another road? I'd like to avoid a `cpu_features.so` because it should be kept private in `volk`. We could add:

```cmake
set(BUILD_SHARED_LIBS OFF CACHE BOOL
    "Build cpu_features static lib for internal linkage only."
    FORCE)
```
before we do: 
```cmake
add_subdirectory(cpu_features)
```
would that solve your issue?

drmpeg, 2020-11-08:
I'm afraid your suggested fix didn't work. It causes the following CMake errors:
```
CMake Error: install(EXPORT "gnuradio-runtime-export" ...) includes target "gnuradio-runtime" which requires target "gnuradio-pmt" that is not in any export set.
CMake Error: install(EXPORT "gnuradio-filter-export" ...) includes target "gnuradio-filter" which requires target "gnuradio-fft" that is not in any export set.
CMake Error: install(EXPORT "gnuradio-filter-export" ...) includes target "gnuradio-filter" which requires target "gnuradio-blocks" that is not in any export set.
```

BTW, I don't consider this a critical bug. It's easily worked around by using an older commit. I just stumbled across it while building GNU Radio 3.8.2 for arm7. Volk 2.0 doesn't build for arm7, so I always have to check out something newer. Usually, I just use v2.3.0, but this time I tried the latest commit.

drmpeg, 2020-11-08:
Looks like this works.
```
set(BUILD_SHARED_LIBS_SAVED "${BUILD_SHARED_LIBS}")
set(BUILD_SHARED_LIBS OFF)
add_subdirectory(cpu_features)
set(BUILD_SHARED_LIBS "${BUILD_SHARED_LIBS_SAVED}")
```
Let me test it more and I'll submit a PR.

jdemel, 2020-11-08:
A PR would be great. Our switch to `cpu_features` is quite invasive I'd like to make sure it works as flawlessly as possible before we tag a new release.

closed 2020-11-09 by michaelld