gnuradio/volk #414: Problem with POLAR encoding/decoding blocks

armin485, 2020-11-13:
It seems newer version of volk got issue with POLAR encoding/decoding blocks as well.
if you open examples in gr-fec/examples/polar_code_example.grc & fecapi_polar_decoders.grc you can observe that source info bits and decoded bits are not same.
it seems the bug comes from the same point because #395 and #3376 in gnuradio git have same issues.
workaround as other issues with same effect:
1. run volk_profile
2. change volk_8u_x3_encode_polar_8u_x2 in ~/.volk/volk_config to generic generic
3. restart gnuradio

jdemel, 2020-11-28:
@armin485 in VOLK convolutional and polar functions are completely separate. I'd assume it's only a coincidence that they manifest in the kind of issue in GR. 
What was  ` volk_8u_x3_encode_polar_8u_x2 generic generic` before you updated it? Could you test this with other kernel options as well?

argilo, 2023-10-12:
I noticed this as well: https://github.com/gnuradio/volk/issues/356#issuecomment-590150158

In fact, if `VOLK_GENERIC=1` is removed from GNU Radio's test suite, then the `qa_polar_decoder_sc_systematic` test begins failing because the AVX2 polar implementation produces incorrect results when the block size is 16.

argilo, 2023-10-12:
Fix: #642

PR #642 (2023-10-22): Add length checks to volk_8u_x2_encodeframepolar_8u
Fixes #414.

The various `volk_8u_x2_encodeframepolar_8u` implementations have warnings about the input size:

https://github.com/gnuradio/volk/blob/a26a1b808ddd75baeebe660e324f23c36dec6416/kernels/volk/volk_8u_x2_encodeframepolar_8u.h#L157-L158

https://github.com/gnuradio/volk/blob/a26a1b808ddd75baeebe660e324f23c36dec6416/kernels/volk/volk_8u_x2_encodeframepolar_8u.h#L430-L431

https://github.com/gnuradio/volk/blob/a26a1b808ddd75baeebe660e324f23c36dec6416/kernels/volk/volk_8u_x2_encodeframepolar_8u.h#L692-L693

https://github.com/gnuradio/v [... cut]
closed 2023-10-22 by jdemel