gnuradio/volk #415: volk 2.4.0 build issue

chenrui333, 2020-11-23:
👋 trying to build the latest release, but run into some build issue. The error log is as below:

<details>
<summary>build failure</summary>

```
==> cmake . -DCMAKE_C_FLAGS_RELEASE=-DNDEBUG -DCMAKE_CXX_FLAGS_RELEASE=-DNDEBUG -DCMAKE_INSTALL_PREFIX=/usr/local/Cellar/volk/2.4.0 -DCMAKE_BUILD_TYPE=Release -DCMAKE_FIND_FRAMEWORK=LAST -DCMAKE_VERBOSE_MAKEFILE=ON -Wno-dev -DCMAKE_OSX_SYSROOT=/Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk -DPYTHON_EXECUTABLE=/usr/local/Cellar/volk/2.4.0/libexec/venv/bin/python -DENABLE_TESTING=OFF
-- The C compiler identification is AppleClang 12.0.0.12000032
-- The CXX compiler identification is AppleClang 12.0.0.12000032
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/local/Homebrew/Library/Homebrew/shims/mac/super/clang - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/local/Homebrew/Library/Homebrew/shims/mac/super/clang++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test HAVE_CX_LIMITED_RANGE
-- Performing Test HAVE_CX_LIMITED_RANGE - Failed
-- Performing Test HAVE_C_LIMITED_RANGE
-- Performing Test HAVE_C_LIMITED_RANGE - Failed
-- Build type set to Release.
-- Found Git: /usr/local/Homebrew/Library/Homebrew/shims/mac/super/git (found version "2.24.3 (Apple Git-128)") 
CMake Error at CMakeLists.txt:123 (add_subdirectory):
  The source directory

    /tmp/volk-20201122-37806-wbegp7/volk-2.4.0/cpu_features

  does not contain a CMakeLists.txt file.

-- 
-- Python checking for python >= 3.4
-- Python checking for python >= 3.4 - found
-- 
-- Python checking for mako >= 0.4.2
-- Python checking for mako >= 0.4.2 - found
-- Looking for C++ include filesystem
-- Looking for C++ include filesystem - found
-- Performing Test CXX_FILESYSTEM_NO_LINK_NEEDED
-- Performing Test have_msse4a - Success
-- Performing Test have_msse4_1
-- Performing Test have_msse4_1 - Success
-- Performing Test have_msse4_2
-- Performing Test have_msse4_2 - Success
-- Performing Test have_mavx
-- Performing Test have_mavx - Success
-- Performing Test have_mavx2
-- Performing Test have_mavx2 - Success
-- Performing Test have_mavx512f
-- Performing Test have_mavx512f - Success
-- Performing Test have_mavx512cd
-- Performing Test have_mavx512cd - Success
-- Performing Test neon_compile_result
-- Performing Test neon_compile_result - Failed
-- Compiler doesn't support NEON, Overruled arch neon
-- Compiler doesn't support NEON, Overruled arch neonv7
-- Compiler doesn't support NEON, Overruled arch neonv8
-- Check size of void*[8]
-- Check size of void*[8] - done
-- CPU width is 64 bits, Overruled arch 32
-- Available architectures: generic;softfp;hardfp;64;popcount;mmx;fma;sse;sse2;orc;norc;sse3;ssse3;sse4_a;sse4_1;sse4_2;avx;avx2;avx512f;avx512cd
-- Available machines: generic_orc;sse2_64_mmx_orc;sse3_64_mmx_orc;ssse3_64_mmx_orc;sse4_a_64_mmx_orc;sse4_1_64_mmx_orc;sse4_2_64_mmx_orc;avx_64_mmx_orc;avx2_64_mmx_orc;avx512f_64_mmx_orc;avx512cd_64_mmx_orc
-- BUILD TYPE = RELEASE
-- Base cflags = -DNDEBUG  -Wall
-- BUILD INFO ::: generic_orc ::: Clang ::: -DNDEBUG  -Wall 
-- BUILD INFO ::: sse2_64_mmx_orc ::: Clang ::: -DNDEBUG  -Wall -m64 -mmmx -msse -msse2
-- BUILD INFO ::: sse3_64_mmx_orc ::: Clang ::: -DNDEBUG  -Wall -m64 -mmmx -msse -msse2 -msse3
-- BUILD INFO ::: ssse3_64_mmx_orc ::: Clang ::: -DNDEBUG  -Wall -m64 -mmmx -msse -msse2 -msse3 -mssse3
-- BUILD INFO ::: sse4_a_64_mmx_orc ::: Clang ::: -DNDEBUG  -Wall -m64 -mmmx -msse -msse2 -msse3 -msse4a -mpopcnt
-- BUILD INFO ::: sse4_1_64_mmx_orc ::: Clang ::: -DNDEBUG  -Wall -m64 -mmmx -msse -msse2 -msse3 -mssse3 -msse4.1
-- BUILD INFO ::: sse4_2_64_mmx_orc ::: Clang ::: -DNDEBUG  -Wall -m64 -mmmx -msse -msse2 -msse3 -mssse3 -msse4.1 -msse4.2 -mpopcnt
-- BUILD INFO ::: avx_64_mmx_orc ::: Clang ::: -DNDEBUG  -Wall -m64 -mmmx -msse -msse2 -msse3 -mssse3 -msse4.1 -msse4.2 -mpopcnt -mavx
-- BUILD INFO ::: avx2_64_mmx_orc ::: Clang ::: -DNDEBUG  -Wall -m64 -mmmx -msse -msse2 -msse3 -mssse3 -msse4.1 -msse4.2 -mpopcnt -mavx -mfma -mavx2
-- BUILD INFO ::: avx512f_64_mmx_orc ::: Clang ::: -DNDEBUG  -Wall -m64 -mmmx -msse -msse2 -msse3 -mssse3 -msse4.1 -msse4.2 -mpopcnt -mavx -mfma -mavx2 -mavx512f
-- BUILD INFO ::: avx512cd_64_mmx_orc ::: Clang ::: -DNDEBUG  -Wall -m64 -mmmx -msse -msse2 -msse3 -mssse3 -msse4.1 -msse4.2 -mpopcnt -mavx -mfma -mavx2 -mavx512f -mavx512cd
-- Compiler Version: Apple clang version 12.0.0 (clang-1200.0.32.27)
Target: x86_64-apple-darwin19.6.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
-- The ASM compiler identification is Clang
-- Found assembler: /usr/local/Homebrew/Library/Homebrew/shims/mac/super/clang
-- c flags:  -Wall;
-- asm flags: 
-- c flags:  -Wall;
-- asm flags: 
-- Loading version 2.4 into constants...
-- Use `posix_memalign` for aligned malloc!
-- Using install prefix: /usr/local/Cellar/volk/2.4.0
-- Configuring incomplete, errors occurred!
See also "/tmp/volk-20201122-37806-wbegp7/volk-2.4.0/CMakeFiles/CMakeOutput.log".
See also "/tmp/volk-20201122-37806-wbegp7/volk-2.4.0/CMakeFiles/CMakeError.log".
```

</details>

Full build log is here, https://github.com/Homebrew/homebrew-core/runs/1439499434
relates to https://github.com/Homebrew/homebrew-core/pull/65462

jdemel, 2020-11-23:
I assume you cloned VOLK without `recursive`? CMake can't find `cpu_features` which is a submodule. Does that solve your issue?

balister, 2020-11-23:
Try running git submodule update --init

On 11/22/20 7:01 PM, chenrui wrote:
> 👋 trying to build the latest release, but run into some build issue. The error log is as below:
>
>
> <details>
> <summary>build failure</summary>
>
> ```
> ==> cmake . -DCMAKE_C_FLAGS_RELEASE=-DNDEBUG [...]

chenrui333, 2020-11-23:
let me give a shot.

chenrui333, 2020-11-23:
That works for me, thanks!

closed 2020-11-23 by chenrui333

jonchang, 2020-11-24:
Note that this is actually an issue with volk's distribution tarballs, i.e. https://github.com/gnuradio/volk/releases/download/v2.4.0/volk-2.4.0.tar.gz and https://github.com/gnuradio/volk/releases/download/v2.4.0/volk-2.4.0.tar.xz, which have an empty `cpu_features` directory. So whatever process that was used to ship these tarballs lacks the submodule checkout. We've worked around this issue in Homebrew by using a git checkout but this is something that should be addressed regardless.

```console
% curl -sL https://github.com/gnuradio/volk/releases/download/v2.4.0/volk-2.4.0.tar.gz | tar tz | grep cpu_features     
volk-2.4.0/cpu_features/
```

tnn2, 2020-12-14:
Ran in to the same problem while trying to update volk in NetBSD pkgsrc. Please reopen the issue and upload fixed release tarballs. Having a dependency on git defeats the purpose of release tarballs. cc @jdemel @chenrui333

jdemel, 2020-12-17:
Hi all!

Initially, we thought it would be a good idea to just have a bugfix release with fixed tarballs. As part of this release process, I updated the v2.4.0 tarballs as well. They include the cpu_features subdirectory now too. I hope that works.
Sorry for the inconvenience. Of course, I'd like to learn and improve the release process.
Also, I'd like to get to know package maintainers. Feel free to get in touch. If you have recommendations etc. let me know too.